probe-rs/probe-rs #1793: Unable to use probe-rs with an fe310g002 microcontroller

romancardenas, 2023-10-04:
**Describe the bug**
I am not able to flash my binary using `probe-rs` over a fe310g002 RISC-V target. However, GDB works fine.

**To Reproduce**
Steps to reproduce the behavior:

1. Just try to run a binary with `probe-rs run --chip fe310-g002 path/to/binary`

**Expected behavior**
I expected my binary to be successfully flashed and run on my board. However, I get this output:
```zsh
$ probe-rs run --chip fe310-g002 path/to/binary
DEBUG probe_rs::architecture::riscv: Before requesting halt, the Dmcontrol register value was: Dmcontrol { .0: 1, hartreset: false, hasel: false, hartsello: 0, hartselhi: 0, ndmreset: false, dmactive: true }
     Erasing sectors ⠁ [00:00:00] [####]      0 B/     0 B @      0 B/s (eta 0s )
 Programming pages   ⠁ [00:00:00] [####]      0 B/     0 B @      0 B/s (eta 0s )DEBUG probe_rs::architecture::riscv: Before requesting halt, the Dmcontrol register value was: Dmcontrol { .0: 1, hartreset: false, hasel: false, hartsello: 0, ha
     Erasing sectors ⠁ [00:00:00] [####]      0 B/     0 B @      0 B/s (eta 0s )Error: The flashing procedure failed for 'path/to/binary'.

Caused by:
    0: Something during the interaction with the core went wrong
    1: A RISCV specific error occurred.
    2: The target does not support halt after reset.
```

**Desktop:**
macOS Moterey

Tiwalun, 2023-10-04:
I think this is fixed by https://github.com/probe-rs/probe-rs/pull/1785. Can you please try the current master version of probe-rs, and check if the issue still happens?

romancardenas, 2023-10-04:
It did the work. I now get the following output:
```zsh
DEBUG probe_rs::architecture::riscv: Before requesting halt, the Dmcontrol register value was: Dmcontrol { .0: 1, hartreset: false, hasel: false, hartsello: 0, hartselhi: 0, ndmreset: false, dmactive: true }
     Erasing sectors ⠁ [00:00:00] [#################################]      0 B/     0 B @      0 B/s (eta 0s )
 Programming pages   ⠁ [00:00:00] [#################################]      0 B/     0 B @      0 B/s (eta 0s )DEBUG probe_rs::architecture::riscv: Before requesting halt, the Dmcontrol register value was: Dmcontrol { .0:
     Erasing sectors ✔ [00:00:04] [#############################################################] 36.00 KiB/36.00 KiB @ 8.42 KiB/s (eta 0s )
 Programming pages   ✔ [00:01:08] [#############################################################] 36.00 KiB/36.00 KiB @    542 B/s (eta 0s )    Finished in 74.473s
ERROR probe_rs::cmd::run: Failed to enable_vector_catch: NotImplemented("vector catch")
ERROR probe_rs::cmd::run: Failed to attach to RTT continuing...    
```

Not sure about the vector catch error, maybe it does not apply to RISC-V targets. On the other hand, I haven't yet integrated RTT in my binary, I'll do it now and see if it works properly.

Thank you so much!

romancardenas, 2023-10-04:
As a side note, it took 74 seconds to flash. Is it normal or is there something that could be improved somehow?

Yatekii, 2023-11-05:
RISC-V is really unoptimized. There is several ways to flash RISC-V and some are more elaborate but faster and vice versa. The implementation in probe-rs right now is the easier one but not so fast afaik.

yusefkarim, 2023-11-13:
I am also running into this problem. My code is [here](https://github.com/yusefkarim/e310x-playground). Initially I was using probe-rs `v0.21.1`,  defmt `v0.3`, and defmt-rtt `v0.4`. Then I tried, the latest probe-rs (off of master branch commit `cb51fd41`). I also tried ditching `defmt` and `defmt-rtt`, then opting for `rtt-target` (`v0.2`) or `jlink_rtt` (`v0.2`) instead. Both `defmt` and `rtt-target` resulted in no output when running flash or trying to attach after flashing. However, for some reason, `jlink_rtt` did print out my message when trying to attach after flashing (but then immediately crashed). I've provided some output below. Hope this helps :crossed_fingers: .

```sh
# probe-rs -V
probe-rs 0.21.1
```

<details><summary>Error when running probe-rs run with any RTT crate</summary>
<p>

```sh
# cargo run --bin blinky

    Finished dev [optimized + debuginfo] target(s) in 0.02s
     Running `probe-rs run --chip fe310-g002 .target/riscv32imac-unknown-none-elf/debug/blinky`
DEBUG probe_rs::architecture::riscv: Before requesting halt, the Dmcontrol register value was: Dmcontrol { .0: 1, hartreset: false, hasel: false, hartsello: 0, hartselhi: 0, ndmreset: false, dmactive: true }
     Erasing sectors ⠁ [00:00:00] [########################################################################################################################]      0 B/     0 B @      0 B/s (eta 0s )
 Programming pages   ⠁ [00:00:00] [########################################################################################################################]      0 B/     0 B @      0 B/s (eta 0s )DEBUG probe_rs::architecture::riscv: Before requesting halt, the Dmcontrol register value was: Dmcontrol { .0: 1, hartreset: false, hasel: false, hartsello: 0, hartselhi: 0, ndmreset: false, dmacti
     Erasing sectors ✔ [00:00:01] [########################################################################################################################] 8.00 KiB/8.00 KiB @ 5.05 KiB/s (eta 0s )
 Programming pages   ✔ [00:00:09] [########################################################################################################################] 4.25 KiB/4.25 KiB @    438 B/s (eta 0s )    Finished in 15.57s
ERROR probe_rs::cmd::run: Failed to enable_vector_catch: NotImplemented("vector catch")
ERROR probe_rs::cmd::run: Failed to attach to RTT continuing...    
    Finished dev [optimized + debuginfo] target(s) in 0.02s
     Running `probe-rs run --chip fe310-g002 .target/riscv32imac-unknown-none-elf/debug/blinky`
DEBUG probe_rs::architecture::riscv: Before requesting halt, the Dmcontrol register value was: Dmcontrol { .0: 1, hartreset: false, hasel: false, hartsello: 0, hartselhi: 0, ndmreset: false, dmactive: true }
     Erasing sectors ⠁ [00:00:00] [########################################################################################################################]      0 B/     0 B @      0 B/s (eta 0s )
 Programming pages   ⠁ [00:00:00] [########################################################################################################################]      0 B/     0 B @      0 B/s (eta 0s )DEBUG probe_rs::architecture::riscv: Before requesting halt, the Dmcontrol register value was: Dmcontrol { .0: 1, hartreset: false, hasel: false, hartsello: 0, hartselhi: 0, ndmreset: false, dmacti
     Erasing sectors ✔ [00:00:01] [########################################################################################################################] 8.00 KiB/8.00 KiB @ 5.05 KiB/s (eta 0s )
 Programming pages   ✔ [00:00:09] [########################################################################################################################] 4.25 KiB/4.25 KiB @    438 B/s (eta 0s )    Finished in 15.57s
ERROR probe_rs::cmd::run: Failed to enable_vector_catch: NotImplemented("vector catch")
ERROR probe_rs::cmd::run: Failed to attach to RTT continuing...    
```
</p>
</details> 

<details><summary>Error when running probe-rs attach against code flashed using the defmt and defmt-rtt crates</summary>
<p>

```sh
DEBUG probe_rs::architecture::riscv: Before requesting halt, the Dmcontrol register value was: Dmcontrol { .0: 1, hartreset: false, hasel: false, hartsello: 0, hartselhi: 0, ndmreset: false, dmactive: true }
Frame 0: __risc_v_rt__main @ 0x20000206
       /home/cuyler/Documents/e310x-playground/without_hal/src/bin/blinky.rs
Frame 1: start_rust @ 0x20000108
       /home/cuyler/.cargo/registry/src/index.crates.io-6f17d22bba15001f/riscv-rt-0.11.0/src/lib.rs:424:5
Error: CPU halted unexpectedly.
```

</p>
</details> 

<details><summary>Error when running probe-rs attach against code flashed using the rtt-target crate</summary>
<p>

```sh
# probe-rs attach --chip fe310-g002 .target/riscv32imac-unknown-none-elf/debug/blinky

DEBUG probe_rs::architecture::riscv: Before requesting halt, the Dmcontrol register value was: Dmcontrol { .0: 1, hartreset: false, hasel: false, hartsello: 0, hartselhi: 0, ndmreset: false, dmactive: true }
Frame 0: spec_next<i32> @ 0x200002b8 inline
       /rustc/cc66ad468955717ab92600c770da8c1601a4ff33/library/core/src/iter/range.rs:621:12
Frame 1: next<i32> @ 0x00000000200002b8 inline
       /rustc/cc66ad468955717ab92600c770da8c1601a4ff33/library/core/src/iter/range.rs:712:9
Frame 2: __risc_v_rt__main @ 0x00000000200002b8
       /home/cuyler/Documents/e310x-playground/without_hal/src/bin/blinky.rs:30:22
Frame 3: start_rust @ 0x20000108
[... cut ...]
DEBUG probe_rs::architecture::riscv: Before requesting halt, the Dmcontrol register value was: Dmcontrol { .0: 1, hartreset: false, hasel: false, hartsello: 0, hartselhi: 0, ndmreset: false, dmactive: true }
Hello, RISC-V!
Frame 0: __risc_v_rt__main @ 0x2000022c
       /home/cuyler/Documents/e310x-playground/without_hal/src/bin/blinky.rs:34:21
Frame 1: start_rust @ 0x20000108
       /home/cuyler/.cargo/registry/src/index.crates.io-6f17d22bba15001f/riscv-rt-0.11.0/src/lib.rs:424:5
Error: CPU halted unexpectedly.
```
</p>
</details>

Yatekii, 2023-11-17:
So your target crashing is something bad in your firmware most likely. We don't halt your CPU usually. Also, the `rtt-target` crate you are using is severely outdated. It could hit the recent miscompilation issue in `cortex-m`. I have not seen`jlink-rtt` in use since many years - I would not use it.

From the stacktraces you can see that you iterate over some range and my assumption is that it panicks. I cannot tell why without source code.

yusefkarim, 2023-11-17:
The program I am running is a simply blinky: [blinky.rs](https://github.com/yusefkarim/e310x-playground/blob/main/without_hal/src/bin/blinky.rs). I am using rtt-target `v0.4`, I had a typo saying `v0.2` which I have now fixed in the original comment, sorry!

Regarding something bad in firmware...I took another look this morning and can't find any problem. The program just blinks an on-board LED. I updated the code slightly since my last comment and here is the sequence of things that happen now:

I run `probe-rs run --chip fe310-g002 .target/riscv32imac-unknown-none-elf/debug/blinky` and see:
```sh
    Finished dev [optimized + debuginfo] target(s) in 0.12s
     Running `probe-rs run --chip fe310-g002 .target/riscv32imac-unknown-none-elf/debug/blinky`
DEBUG probe_rs::architecture::riscv: Before requesting halt, the Dmcontrol register value was: Dmcontrol { .0: 1, hartreset: false, hasel: false, hartsello: 0, hartselhi: 0, ndmreset: false, dmactive: true }
     Erasing sectors ⠁ [00:00:00] [########################################################################################################################]      0 B/     0 B @      0 B/s (eta 0s )
 Programming pages   ⠁ [00:00:00] [########################################################################################################################]      0 B/     0 B @      0 B/s (eta 0s )DEBUG probe_rs::architecture::riscv: Before requesting halt, the Dmcontrol register value was: Dmcontrol { .0: 1, hartreset: false, hasel: false, hartsello: 0, hartselhi: 0, ndmreset: false, dmacti
     Erasing sectors ✔ [00:00:02] [########################################################################################################################] 8.00 KiB/8.00 KiB @ 3.66 KiB/s (eta 0s )
 Programming pages   ✔ [00:00:18] [########################################################################################################################] 6.50 KiB/6.50 KiB @    350 B/s (eta 0s )    Finished in 23.53s
ERROR probe_rs::cmd::run: Failed to enable_vector_catch: NotImplemented("vector catch")
ERROR probe_rs::cmd::run: Failed to attach to RTT continuing...   
```

The LED at this point is blinking consistently without a problem. I then CTRL-C out of the run prompt, reset the CPU (noticing that the LED blinks fine), then run: `probe-rs attach --chip fe310-g002 .target/riscv32imac-unknown-none-elf/debug/blinky` and see:

```sh
DEBUG probe_rs::architecture::riscv: Before requesting halt, the Dmcontrol register value was: Dmcontrol { .0: 1, hartreset: false, hasel: false, hartsello: 0, hartselhi: 0, ndmreset: false, dmactive: true }
Hello, RISC-V!
Toggling user LED
Toggling user LED
Toggling user LED
Toggling user LED
Toggling user LED
Toggling user LED
Toggling user LED
Toggling user LED
Toggling user LED
Toggling user LED
Toggling user LED
Toggling user LED
Toggling user LED
Toggling user LED
Toggling user LED
Toggling user LED
Toggling user LED
Toggling user LED
Toggling user LED
Toggling user LED
Toggling user LED
Toggling user LED
Toggling user LED
Toggling user LED
Toggling user LED
Toggling user LED
Toggling user LED
Toggling user LED
Toggling user LED
Toggling user LED
Toggling user LED
Toggling user LED
Toggling user LED
Toggling user LED
Toggling user LED
Toggling user LED
Toggling user LED
Toggling user LED
Toggling user LED
Toggling user LED
Toggling user LED
Toggling user LED
Toggling user LED
Toggling user LED
Toggling user LED
Toggling user LED
Toggling user LED
Toggling user LED
Toggling user LED
Toggling user LED
Toggling user LED
Toggling user LED
Toggling user LED
Toggling user LED
Toggling user LED
Toggling user LED
Frame 0: __risc_v_rt__main @ 0x20000292
       /home/cuyler/Documents/e310x-playground/without_hal/src/bin/blinky.rs
Frame 1: start_rust @ 0x20000108
       /home/cuyler/.cargo/registry/src/index.crates.io-6f17d22bba15001f/riscv-rt-0.11.0/src/lib.rs:424:5
Error: CPU halted unexpectedly.
```
At this point (i.e., only right after I run attach) the LED stops blinking.

There are two things confusing me now:

1. Why do I not see anything except that error message when I execute `probe-rs run`?
2. Why does the CPU halt AFTER I execute `probe-rs attach`?
  - Note: I am using `rtt_target::rtt_init_print!(NoBlockSkip);` and I've also tried with `BlockIfFull` (similar results)

It's been awhile since I've done any embedded-related things, so I apologize if I am doing anything silly. Please let me know if there is anything I can help with or clarify.