probe-rs/probe-rs #1802: Cannot connect to LPC55S69 chip

spcan, 2023-10-06:
**Describe the bug**

Any attempt to flash to LPC55S69 device with `cargo-flash` results in one of two errors:

Without `--connect-under-reset`:

```
        WARN probe_rs::config::target > Using custom sequence for LPC55S16/26/28/66/69
       Error Connecting to the chip was unsuccessful.

  Caused by:  
          0: An ARM specific error occurred.
          1: Error using access port
          2: Failed to read register DRW at address 0x0000000c
          3: An error occurred in the communication with an access port or debug port.
          4: Target device did not respond to request.

        Hint A hard reset during attaching might help. This will reset the entire chip. Run with `--connect-under-reset` to enable this feature.
```

With `--connect-under-reset`:

```
       Error Connecting to the chip was unsuccessful.

  Caused by:  
          0: An ARM specific error occurred.
          1: An error occurred in the communication with an access port or debug port.
          2: Target device did not respond to request.
```

Using probe-rs from code also fails with the same error, as well as any other `probe-rs` tool.

**To Reproduce**
Steps to reproduce the behavior:

1. Connect an LPC22S69 to a debugger
2. Attempt to connect to it (or flash it, or debug it) with any `probe-rs` tool (VSCode, flash, embed, run, cli, with custom code, etc...)

**Expected behavior**
Being able to flash and/or debug the device

**Stacktrace**
N/A

**Desktop (please complete the following information):**

- Linux Ubuntu

**Additional context**
Using non-EVK board (OKDO E1)

Yatekii, 2023-10-06:
Considering the device does not respond at all, are you sure its all wired ip correctly?

spcan, 2023-10-10:
probe-rs is only connected to the on-board debugger theough USB. This debugger is a CMSIS-DAP debugger using an LPC11U35

It seems like somehow the SWD debugger interface got disabled at some point, because previously I was able to fully debug and access it. Is there a recovery process in probe-rs for this? Or does the LPC55S69 debug sequence include a way to unlock the device?

sefidel, 2023-10-15:
I'm having the same issue with LPCXpresso55S69 development board.

Trace log: https://gist.github.com/sefidel/c7221448e0a923d5bbf0eb5b269428bd

Tiwalun, 2023-10-15:
Please try the fix in #1823, that should fix this as long as you don't need access to the second core.

closed 2023-10-15 by Yatekii

spcan, 2023-10-17:
This fixes `probe-rs debug` as it can now attach and debug the target correctly, but flashing does still not work

Both `probe-rs run` and `cargo flash` fail with the error

```
        WARN probe_rs::config::target > Using custom sequence for LPC55S16/26/28/66/69
     Erasing sectors ⠁ [00:00:00] [#################################################################################################################]      0 B/     0 B @      0 B/s (eta 0s )

Error Trying to write flash, but no suitable (default) flash loader algorithm is linked to the given target: LPC55S69JBD100 .

  Caused by:  
          0: Trying to write flash, but no suitable (default) flash loader algorithm is linked to the given target: LPC55S69JBD100 .
```

PR #1823 (2023-10-15): fix: Disable second core on LPC55S69
There is an error when trying to connect to the second core,
which causes probe-rs to abort.

There is probably some setup required so that the second core works.
After that setup is implemented in probe-rs, the second core
could be reenabled.

This should fix #1802.