probe-rs/probe-rs #1807: crash: error while resolving the location for variable

andresovela, 2023-10-10:
**Describe the bug**
`probe-rs` crashes with the following error when stepping through my program.

```
Debug Adapter terminated unexpectedly with an error: Other(Error executing request.

Caused by:
    0: Error parsing debug data
    1: Type mismatch when evaluating expression)

ERROR probe_rs::debug::unit_info: Encounted an unexpected  Named("send_stop"). Please report this as a bug
```

The repo I'm debugging is [here](https://github.com/andresovela/embassy-poc/tree/e8e8a3485af257137b013fd03d6e928ccc2483f7).

I'm setting up a breakpoint in `src/bsp/i2c/mod.rs:52`, from there I step into until I get to `write_internal` in `embassy_stm32/src/i2c/v2.rs:374`. There `probe-rs` crashes.

Note that I'm using the probe-rs vscode extension.

**Expected behavior**
The debugger shouldn't crash.

**Desktop:**

- Fedora 34

noppej, 2023-10-10:
@andresovela Can you please upload a copy of your binary?

andresovela, 2023-10-10:
I had to zip it because GitHub was complaining about the file format.
[ecospeaker.zip](https://github.com/probe-rs/probe-rs/files/12857780/ecospeaker.zip)

noppej, 2023-10-10:
Is this message (from above) "ERROR probe_rs::debug::unit_info: Encounted an unexpected  Named("send_stop"). Please report this as a bug" truncated in some way?

andresovela, 2023-10-10:
I don't think so, that's what's shown in the vscode popup

noppej, 2023-10-10:
The crash is caused by build optimization settings. The current implementation only works for non optimized code. The debugger should not crash, and the error message should be improved.

PR #1810 (2023-10-10): Fix crash and misleading error while resolving the location for variable in optimized binary.
Fixes #1807