probe-rs/probe-rs #1816: Connect `probe-rs run` standard input to RTT down channel

ia0, 2023-10-12:
**Describe the solution you'd like**
I would like to use `probe-rs run` as a terminal to the device using the RTT down channel.

**Describe alternatives you've considered**
I considered using something else than RTT and would understand if that's what you would suggest.

Yatekii, 2023-10-12:
You could use `cargo-embed` for this :) probe-run cannot really do this because afaik cargo runners cannot take input :)

ia0, 2023-10-12:
> probe-run cannot really do this because afaik cargo runners cannot take input

I don't use `probe-run` but `probe-rs run`, but maybe that's what you meant. And I don't use a cargo runner, I call `probe-rs run` directly on the command line. So reading from standard input should not be a problem.

> You could use cargo-embed for this

I don't think this works for me. I'm building the binary myself (because I need precise control using `cargo rustc`). I just want something to flash and connect to the defmt debug output. And now I'd like it also to connect to the debug input.

Also I don't like how `cargo-embed` can only be configured by a file. My configuration is dynamic so building a command line on the fly matches much better my workflow. If I want to specify all possible combinations of configurations using `cargo-embed` "configs" it would not be maintenable and the file would be huge. It would also restrain to anything that can be anticipated only.

Yatekii, 2023-11-05:
Sorry for the late response! Yeah, I meant `probe-rs run`, was typing this up in a hurry on my phone while travelling ...

I am happy to have this functionality in `probe-rs run`. How it would look exactly, I am not sure. Maybe we can have a flag that enables it. Or maybe cargo actually supports this natively. Would be worth a try!