probe-rs/probe-rs #1820: ESP32-C6 using the JTAG interface with embed

blue-freedom-technologies, 2023-10-13:
### Discussed in https://github.com/probe-rs/probe-rs/discussions/1817

<div type='discussions-op-text'>

<sup>Originally posted by **blue-freedom-technologies** October 12, 2023</sup>
Since the ESP32-C6 have a [JTAG interface](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c6/api-guides/jtag-debugging/index.html) for debugging , anyone knows how to setup this with embed ?</div>

blue-freedom-technologies, 2023-10-20:
Ok,closing the issue. Looks like probe-rs does not support ESP32 chips.

Yatekii, 2023-10-20:
That is incorrect. It supports ESP (with RISC-V) chips. Could you try rewording your question? It is not clear to me what the problem is.

blue-freedom-technologies, 2023-11-05:
Closing the issue due the lack of responses from the contributors..

closed 2023-11-05 by blue-freedom-technologies

Yatekii, 2023-11-05:
You did not even read the basic error message and try and browse the config flags and see what could help.