probe-rs/probe-rs #1827: Cannot flash STM32WL55JC

kolegs, 2023-10-16:
**Describe the bug**
Hi,

I am trying to flash stm32 using the `cargo flash` tool, but for some reason it cannot connect to the chip and I am not sure what is the problem here. I can flash the board no problem using stm32cubeprogrammer. I got NUCLEO-WL55JC2 board.

**To Reproduce**
Steps to reproduce the behavior:

1. Connect the board
2. Try running `cargo flash --relase --chip stm32wl55jcix --connect-under-reset --log=DEBUG`
3. See error in log

**Expected behavior**
Expect board to be flashed

**Stacktrace**
```
   Flashing /home/adejewski/Projects/smart-postbox/smart-postbox/target/thumbv7em-none-eabi/release/smart-postbox                                                                                                                                                                                                                                                                                                                             
       DEBUG probe_rs::config::registry > Searching registry for chip with name stm32wl55jcix                                                                                                                                                                                                                                                                                                                                                  
       DEBUG probe_rs::config::registry > Exact match for chip name: STM32WL55JCIx                                                                                                                                                                                                                                                                                                                                                             
        WARN probe_rs::config::target   > Using custom sequence for ARMv7 STM32WL55JCIx                                                                                                                                                                                                                                                                                                                                                        
        INFO tracing::span              > list_all;                                                                                                                                                                                                                                                                                                                                                                                            
        INFO tracing::span              > list_cmsisdap_devices;                                                                                                                                                                                                                                                                                                                                                                               
       DEBUG probe_rs::probe::cmsisdap::tools > Searching for CMSIS-DAP probes using libusb                                                                                                                                                                                                                                                                                                                                                    
       DEBUG probe_rs::probe::cmsisdap::tools > Found 0 CMSIS-DAP probes using libusb, searching HID                                                                                                                                                                                                                                                                                                                                           
       DEBUG probe_rs::probe::cmsisdap::tools > Found 0 CMSIS-DAP probes total                                                                                                                                                                                                                                                                                                                                                                 
        INFO tracing::span                    > list_stlink_devices;                                                                                                                                                                                                                                                                                                                                                                           
        INFO tracing::span                    > list_jlink_devices;                                                                                                                                                                                                                                                                                                                                                                            
       DEBUG jaylink                          > libusb 1.0.26.11724                                                                                                                                                                                                                                                                                                                                                                            
       DEBUG jaylink                          > libusb has capability API: true                                                                                                                                                                                                                                                                                                                                                                
       DEBUG jaylink                          > libusb has HID access: true                                                                                                                                                                                                                                                                                                                                                                    
       DEBUG jaylink                          > libusb has hotplug support: true                                                                                                                                                                                                                                                                                                                                                               
       DEBUG jaylink                          > libusb can detach kernel driver: true                                                                                                                                                                                                                                                                                                                                                          
        INFO tracing::span                    > list_espjtag_devices;                                                                                                                                                                                                                                                                                                                                                                          
        INFO tracing::span                    > open;                                                                                                                                                                                                                                                                                                                                                                                          
       DEBUG probe_rs::probe::cmsisdap::tools > Attempting to open 0483:374e in CMSIS-DAP v1 mode                                                                                                            [... cut ...]
        "ST-Link V3",
    ),
    speed: Some(
        "12000",
    ),
    release: "0.18.0",
    commit: "crates.io",
}
       Error Connecting to the chip was unsuccessful.

  Caused by:  
          0: An ARM specific error occured.
          1: The debug probe encountered an error.
          2: An error specific to a probe type occurred
          3: Command failed with status SwdApFault
```
**Desktop (please complete the following information):**

- Linux (Ubuntu,)

Any hints, what might be wrong?

Yatekii, 2023-10-17:
Your version seems to be outdated. Mind upgrading and reporting back? :)

kolegs, 2023-10-17:
Thanks for pointing it out, you are right, I installed cargo-flash instead of probe-rs, now I am on the latest version:
```
cargo flash --version
cargo-flash 0.21.1
git commit: crates.io
```

Unfortunately it didn't help, still the same issue. I attach the log below
```
    Flashing /home/adejewski/Projects/smart-postbox/smart-postbox/target/thumbv7em-none-eabi/release/smart-postbox                                                                                                 
       DEBUG probe_rs::config::registry > Searching registry for chip with name stm32wl55jcix                                                                                                                      
       DEBUG probe_rs::config::registry > Exact match for chip name: STM32WL55JCIx                                                                                                                                 
        WARN probe_rs::config::target   > Using custom sequence for ARMv7 STM32WL55JCIx                                                                                                                            
        INFO tracing::span              > list_all;                                                                                                                                                                
        INFO tracing::span              > list_cmsisdap_devices;                                                                                                                                                   
       DEBUG probe_rs::probe::cmsisdap::tools > Searching for CMSIS-DAP probes using libusb                                                                                                                        
       DEBUG probe_rs::probe::cmsisdap::tools > Found 0 CMSIS-DAP probes using libusb, searching HID                                                                                                               
       DEBUG probe_rs::probe::cmsisdap::tools > Found 0 CMSIS-DAP probes total                                                                                                                                     
        INFO tracing::span                    > list_stlink_devices;                                                                                                                                               
        INFO tracing::span                    > list_jlink_devices;                                                                                                                                                
       DEBUG jaylink                          > libusb 1.0.26.11724                                                                                                                                                
       DEBUG jaylink                          > libusb has capability API: true                                                                                                                                    
       DEBUG jaylink                          > libusb has HID access: true                                                                                                                                        
       DEBUG jaylink                          > libusb has hotplug support: true                                                                                                                                   
       DEBUG jaylink                          > libusb can detach kernel driver: true                                                                                                                              
        INFO tracing::span                    > list_espjtag_devices;                                                                                                                                              
        INFO tracing::span                    > open;                                                                                                                                                              
       DEBUG probe_rs::probe::cmsisdap::tools > Attempting to open 0483:374e in CMSIS-DAP v1 mode                                                                                                                  
       DEBUG probe_rs::probe::stlink::usb_interface > Acquired libusb context.                                                                                                                                     
       DEBUG probe_rs::probe::stlink::usb_interface > Aquired handle for probe                                                                                                                                     
       DEBUG probe_rs::probe::stlink::usb_interface > Active config descriptor: ConfigDescriptor { bLength: 9, bDescriptorType: 2, wTotalLength: 128, bNumInterfaces: 4, bConfigurationValue: 1, iConfiguration: 4,
 bmAttributes: 128, bMaxPower: 250, extra: [] }                                                                                                                                                                    
       DEBUG probe_rs::probe::stlink::usb_interface > Device descriptor: DeviceDescriptor { bLength: 18, bDescriptorType: 1, bcdUSB: 512, bDeviceClass: 239, bDeviceSubClass: 2, bDeviceProtocol: 1, bMaxPacketSize
[... cut ...]
        INFO tracing::span                                        > get_last_rw_status;
        WARN probe_rs::probe::stlink                              > send_jtag_command 242 failed: SwdApWdataError
       DEBUG probe_rs::probe::stlink                              > Current device mode: Jtag
       Error Connecting to the chip was unsuccessful.

  Caused by:  
          0: An ARM specific error occurred.
          1: The debug probe encountered an error.
          2: An error specific to a probe type occurred
          3: Command failed with status SwdApWdataError
```

kolegs, 2023-10-26:
hmm, it works for me also from the openocd command, but cargo fails :(

Yatekii, 2023-11-05:
Hmm this should be performing a proper HW reset tho. I think the error might be somewhere else ... it looks like it has trouble unlocking the co-processor. This might need some custom sequence to work :)

kolegs, 2023-11-06:
@Yatekii any hints how to deal with this problem? I can try to debug it but not sure where to start

Yatekii, 2023-11-07:
So, there is a trait called [`DebugSequence`](https://github.com/probe-rs/probe-rs/blob/master/probe-rs/src/architecture/arm/sequences/mod.rs#L410). That needs to be implemented for a handler struct that can be registered during startup (here: https://github.com/probe-rs/probe-rs/blob/master/probe-rs/src/config/target.rs#L112).

In there you have multiple hooks to set up your chip. How exactly you need to set it up is defined in the chips reference manual. What could simplify figuring what you need to do is having a look at the .pdsc file in the CMSIS-Pack for the chip, which often describes the sequence in terrible XML syntax, but it's a start :)

mgr9525, 2024-12-06:
My STM32WLE5CCUx also encountered the same issue: my version is: `probe-rs 0.24.0`

```
probe-rs run --chip STM32WLE5CCUx --protocol swd --connect-under-reset target/thumbv7m-none-eabi/release/bin.hex
```

```
WARN probe_rs::probe::stlink: send_jtag_command 242 failed: SwdApFault
Error: Connecting to the chip was unsuccessful.

Caused by:
    0: An ARM specific error occurred.
    1: The debug probe encountered an error.
    2: An error which is specific to the debug probe in use occurred.
    3: Command failed with status SwdApFault
```

But I can burn hex code with Keil.