probe-rs/probe-rs #1829: Unable to flash LPC55S69 because no suitable flash loader is found.

Tiwalun, 2023-10-18:
Both  `probe-rs run` and `cargo flash` fail with the error

```
        WARN probe_rs::config::target > Using custom sequence for LPC55S16/26/28/66/69
     Erasing sectors ⠁ [00:00:00] [#################################################################################################################]      0 B/     0 B @      0 B/s (eta 0s )

Error Trying to write flash, but no suitable (default) flash loader algorithm is linked to the given target: LPC55S69JBD100 .

  Caused by:  
          0: Trying to write flash, but no suitable (default) flash loader algorithm is linked to the given target: LPC55S69JBD100 .
```

_Originally posted by @spcan in https://github.com/probe-rs/probe-rs/issues/1802#issuecomment-1766272594_

Tiwalun, 2023-10-18:
The flash seems to be bigger than the flash algorithm:

https://github.com/probe-rs/probe-rs/blob/ea0fe1a9eb3e38b53806f1cae3c05b62c3771e55/probe-rs/targets/LPC55S69.yaml#L44-L51

but the algorithm only goes to 0x98000

https://github.com/probe-rs/probe-rs/blob/ea0fe1a9eb3e38b53806f1cae3c05b62c3771e55/probe-rs/targets/LPC55S69.yaml#L282-L286

Regenerating the target description with the newest pack might solve this, or just editing the file so that the two ranges match.

Tiwalun, 2023-10-18:
The target file is just cursed, the core names don't match as well. The memories claim they belong to core `main`, but the core is actually named `cm33_core0` :facepalm:

9names, 2023-10-19:
Fwiw, I hit this issue when hand-writing a target file. If we're keeping current behaviour, I would have found it helpful if there were a warning that the flash algo would not match any region (although I understand it's not going to help many people)

mciantyre, 2023-10-20:
#1193 tracks my experience with this error on another NXP chip. My workaround was to edit the target files.

If anyone wants to explore this issue without an LPC55, steps 1 and 2 of #1193's reproduction steps can demonstrate the issue on the main branch.

MabezDev, 2023-10-20:
Hitting this in the esp32 targets too. Just as I was adding H2 support which was super confusing 😭.

PR #1832 (2023-11-05): Fix: Target description for LPC55S69
- fix: Core names in LPC55S69 target description
       
       Also update target-gen to handle memory descriptions with missing cores
       better.

- fix: Region for LPC55S69 flash algorithm
- Fix debug sequence for LPC55S69

These changes should fix #1829.
closed 2023-11-05 by Yatekii