probe-rs/probe-rs #1846: No connection to lpc55s69 with jlink (on lpcxpresso)

daringer, 2023-10-29:
**Describe the bug**
Any `probe-rs` calls towards the MCU fail. SWD seems to end in an endless recursion.

**To Reproduce**
Steps to reproduce the behavior:

1. Connect jlink via SWD to lpc55s69
2. Run `probe-rs info`

**Expected behavior**
The SWD connection works

**Stacktrace / Log**
```
INFO probe_rs: Writing log to "foo.log"
DEBUG list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Searching for CMSIS-DAP probes using libusb
DEBUG list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Found 0 CMSIS-DAP probes using libusb, searching HID
DEBUG list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Found 0 CMSIS-DAP probes total
DEBUG open: probe_rs::probe::cmsisdap::tools: Attempting to open 1366:1024 in CMSIS-DAP v1 mode
DEBUG open: probe_rs::probe::stlink::usb_interface: Acquired libusb context.
Probing target via SWD

DEBUG probe_rs::probe::jlink: Attaching to J-Link
DEBUG probe_rs::probe::jlink: Attaching with protocol 'SWD'
 INFO probe_rs::probe::jlink: J-Link: S/N: 726567073
DEBUG probe_rs::probe::jlink: J-Link: Capabilities: GetHwVersion | WriteDcc | AdaptiveClocking | ReadConfig | WriteConfig | WriteMem | ReadMem | SpeedInfo | GetMaxBlockSize | GetHwInfo | SetKsPower | ResetStopTimed | MeasureRtckReact | SelectIf | RwMemArm79 | GetCounters | ReadDcc | GetCpuCaps | ExecCpuCmd | Swo | WriteDccEx | Register | Indicators | TestNetSpeed | GetCapsEx | HwJtagWrite
 INFO probe_rs::probe::jlink: J-Link: Firmware version: J-Link LPCXpresso V2 compiled Aug 25 2023 12:01:15
 INFO probe_rs::probe::jlink: J-Link: Hardware version: 109.0.0
 INFO probe_rs::probe::jlink: J-Link: Target voltage: 3.30 V
DEBUG probe_rs::probe::jlink: Attached succesfully
DEBUG debug_port_setup: probe_rs::probe::arm_jtag: Performing 1 transfers (0 additional transfers)
DEBUG debug_port_setup: probe_rs::probe::arm_jtag: Transfer result 0: Err(WaitResponse)
DEBUG debug_port_setup: probe_rs::probe::arm_jtag: DAP WAIT, (read), retries remaining 1000.
DEBUG debug_port_setup: probe_rs::probe::arm_jtag: Performing 1 transfers (0 additional transfers)
DEBUG debug_port_setup: probe_rs::probe::arm_jtag: Transfer result 0: Err(WaitResponse)
DEBUG debug_port_setup: probe_rs::probe::arm_jtag: DAP WAIT, (write), retries remaining 1000.

.... repeats for maaany lines ...

thread 'main' has overflowed its stack
fatal runtime error: stack overflow
```

full `--log-file` output:  [log.txt](https://github.com/probe-rs/probe-rs/files/13196968/log.txt) (remove 100k repeating lines in the middle)

**Desktop (please complete the following information):**
- Arch Linux

**Additional context**
* Generally the debugger + MCU works fine with my other toolchain 
* Tested with the most recent disabled-other-core fix https://github.com/probe-rs/probe-rs/pull/1823 - same behavior
* Also tested with: https://github.com/probe-rs/probe-rs/pull/1832 - same behavior
* My debugger probe is a LPCXpresso55S69 flashed with jlink firmware and it is connected to an external device (lpc55s69) via SWD

Tiwalun, 2023-10-30:
As a workaround, I would suggest to use the CMSIS-DAP firmware on the board, if you can. CMSIS-DAP is generally better supported, and it should work for the LPC55S69.

daringer, 2023-10-30:
OK thanks for the hint, will try that next

9names, 2023-11-26:
I've tested out LPC55S69 (using mcu-link as the target) with a JLink EDU Mini as the debugger.
If you instruct `probe-rs info` to use the SWD protocol it no longer fails:
```
probe-rs info --protocol swd
Probing target via SWD

ARM Chip:
Debug Port: Version 2, Designer: NXP (Philips), Part: 0x0, Revision: 0x0
├── 0 MemoryAP
│   └── ROM Table (Class 1)
│       ├── ROM Table (Class 1)
│       │   ├── Cortex-M33 SCS  (Coresight Component)
│       │   ├── Cortex-M33 DWT  (Coresight Component)
│       │   ├── Cortex-M33 BPU  (Coresight Component)
│       │   └── Cortex-M33 ITM  (Coresight Component)
│       └── Cortex-M33 TPIU (Coresight Component)
├── 1 MemoryAP
│   └── Error during access: Error using access port
└── 2 Unknown AP (Designer: NXP (Philips), Class: Undefined, Type: 0x0, Variant: 0x0, Revision: 0x0)
```
`probe-rs run` and `probe-rs erase` also work with the JLink and this chip - but with those tools I always specify `--protocol swd` since `probe-rs`, so that is possibly why it works for me and not for OP.
LPC55S69 does quite frequently get into a state where I have to `probe-rs erase` for it to be successfully programmed again.
There's also some weirdness with other tools (embed, debugger) with this chip but I'm not sure if they're related - I get them when attaching to LPC55S69 with a CMSIS-DAP debugger as well.