probe-rs/probe-rs #1852: Adding support for JH7110 SoCs

rmsyn, 2023-11-01:
**Is your feature request related to a problem? Please describe.**

I'm currently working on support for the JH7110 family of SoCs from StarFive.

**Describe the solution you'd like**

I'm  having trouble connecting to the board over JTAG using a `CMSIS-Pack` derived target file: https://github.com/rmsyn/probe-rs/blob/target/jh7110/probe-rs/targets/JH7110_Series.yaml

The target file was generated using `target-gen` from an ELF binary compiled from the [`starfive-flashloader`](https://github.com/rmsyn/starfive-flashloader). Some manual editing for names, and the `Ram` memory range.

I'm using the [`Sipeed USB-JTAG/TTL debugger`](https://www.seeedstudio.com/Sipeed-USB-JTAG-TTL-RISC-V-Debugger-ST-Link-V2-STM8-STM32-Simulator-p-2910.html).

Are there other debuggers that are more well-supported by `probe-rs`? Are there guides for adding support for a debugger?

The chip in the Sipeed debugger is an FTDI FT2232D.

**Describe alternatives you've considered**

I've also tried `openocd` using the `ftdi/openocd-usb.cfg` script, and a config script for the JH7110:

```
reset_config trst_only

jtag newtap e24 cpu -irlen 5 -expected-id 0x07110cfd
jtag newtap u74 cpu -irlen 5 -expected-id 0x07110cfd

#target create e24.cpu0 riscv -chain-position e24.cpu -coreid 0
target create u74.cpu0 riscv -chain-position u74.cpu -coreid 0 -rtos hwthread
target create u74.cpu1 riscv -chain-position u74.cpu -coreid 1
target create u74.cpu2 riscv -chain-position u74.cpu -coreid 2
target create u74.cpu3 riscv -chain-position u74.cpu -coreid 3
target create u74.cpu4 riscv -chain-position u74.cpu -coreid 4
target smp u74.cpu0 u74.cpu1 u74.cpu2 u74.cpu3 u74.cpu4

adapter speed 4000
```

These configs result in errors about the JTAG not being setup properly.

rmsyn, 2023-11-02:
Update: I was able to connect using `openocd` with the following during startup:

```
$ openocd -f ./openocd-usb.cfg -c "transport select jtag" -f ./openocd-jh7110.cfg

Open On-Chip Debugger 0.12.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
jtag
Info : Hardware thread awareness created
adapter speed: 4000 kHz

Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 4000 kHz
Info : JTAG tap: e24.cpu tap/device found: 0x07110cfd (mfg: 0x67e (Guangdong StarFive Technology Co), part: 0x7110, ver: 0x0)
Info : JTAG tap: u74.cpu tap/device found: 0x07110cfd (mfg: 0x67e (Guangdong StarFive Technology Co), part: 0x7110, ver: 0x0)
Info : datacount=2 progbufsize=16
Info : Disabling abstract command reads from CSRs.
Info : Core 0 made part of halt group 1.
Info : Examined RISC-V core; found 5 harts
Info :  hart 0: XLEN=64, misa=0x8000000000901107
Info : datacount=2 progbufsize=16
Info : Disabling abstract command reads from CSRs.
Info : Core 1 made part of halt group 1.
Info : Examined RISC-V core; found 5 harts
Info :  hart 1: XLEN=64, misa=0x800000000094112f
Info : datacount=2 progbufsize=16
Info : Disabling abstract command reads from CSRs.
Info : Core 2 made part of halt group 1.
Info : Examined RISC-V core; found 5 harts
Info :  hart 2: XLEN=64, misa=0x800000000094112f
Info : datacount=2 progbufsize=16
Info : Disabling abstract command reads from CSRs.
Info : Core 3 made part of halt group 1.
Info : Examined RISC-V core; found 5 harts
Info :  hart 3: XLEN=64, misa=0x800000000094112f
Info : datacount=2 progbufsize=16
Info : Disabling abstract command reads from CSRs.
Info : Core 4 made part of halt group 1.
Info : Examined RISC-V core; found 5 harts
Info :  hart 4: XLEN=64, misa=0x800000000094112f
Info : starting gdb server for u74.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
Info : accepting 'gdb' connection on tcp/3333
```

However, when I try to start `probe-rs list` or `probe-rs info` during startup, it reports no detected probes.

Yatekii, 2023-11-02:
Hi 👋 

Thanks a lot for working on this!

I am not sure what kind of probe the Sipeed debugger is exactly. Looks like all the Chinese ST-Link v2 but seemingly isn't.
Supported JTAG probes are JLink, CMSIS-DAP based probes and FTDI.

To add a new probe, you could follow the code that implements it for `CmsisDap` :) It should be rather straightforward :) The JTAG mechanic should be abstracted out by now :)

rmsyn, 2023-11-02:
> I am not sure what kind of probe the Sipeed debugger is exactly. Looks like all the Chinese ST-Link v2 but seemingly isn't.
Supported JTAG probes are JLink, CMSIS-DAP based probes and FTDI.

It does look like that, but taking off the case, the chip inside is an FTDI FT2232D 2306.

> To add a new probe, you could follow the code that implements it for CmsisDap :) It should be rather straightforward :) The JTAG mechanic should be abstracted out by now :)

I'm not familiar with CMSIS-DAP, so I'll brush up on the protocol. Wish me luck :)

> If you need any more help, also feel free to drop by in our Matrix channel :)

I'm sure I will, I'll create a matrix account and join the channel.

Yatekii, 2023-11-03:
The instructions are base64 encoded program code. It grows with the size of your flashloader that you are using :)

rmsyn, 2023-11-03:
Right, I think the PAC I wrote (generated?) is causing the binary size to blow up. I had to resort to building a custom SVD based on the DTS and various spec sheets...

Going to look at the `qspi` portion of the crate first to see if I can bring the binary size down.

Currently, it is sitting at 2.3M, while `hifive-flashloader` is closer to 56K :scream:

9names, 2023-11-04:
@rmsyn 
The size of your elf file is *not* the size of your binary.
llvm-size tells me your flash loader is around 240KB, which is large for a flashloader but not impossibly large.
Adding a few more of the regular rust size-optimisation flags to Cargo.toml brings that  down to 11KB which sounds about right for a flash algo:
```toml
[profile.release]
codegen-units = 1
debug = 2
debug-assertions = false
incremental = false
lto = 'fat'
opt-level = 's'
overflow-checks = false
```
I also notice you're trying to use PIC, that's not advised - LLVM doesn't have working PIC for either arm or riscv. Probe-rs (and all the other algo tools tbh) will always load your flash-loader at a location of your choosing, you should have a fixed address in the linker script for your flash algo.

rmsyn, 2023-11-04:
@9names Yes! This is exactly it, tried the `lto = true` profile option, and instantly dropped the binary size significantly.

I'll also add the other options you listed to drop the binary size further. So many thanks to both of you <3

rmsyn, 2023-11-04:
@Yatekii @9names looks like the DTM version is `0.13` for the `U74` in `JH7110` SoCs.

I'm going to start working on generalizing the DTM support module so that it's possible to support multiple versions. This should hopefully help with future portability for additional protocol additions.

9names, 2023-11-04:
dropping this here in case you haven't seen it https://github.com/probe-rs/probe-rs/issues/665