probe-rs/probe-rs #1863: Do not print backtrace on brakepoint

ia0, 2023-11-06:
**Is your feature request related to a problem? Please describe.**
I believe `probe-run` used to not print a backtrace on breakpoints. This was convenient to have some form of successful non-verbose exit.

**Describe the solution you'd like**
Provide a flag to disable backtrace on breakpoints.

**Describe alternatives you've considered**
I'm not aware of other mechanisms to exit `probe-rs run` without backtrace from the firmware.

**Additional context**
Right now, I get backtrace like this:
```
DEBUG probe_rs::debug::unit_info: Found DIE, now checking for inlined functions: name=None
Frame 0: __bkpt @ 0x00013214 inline
       ./asm/lib.rs:48:1
Frame 1: __bkpt @ 0x0000000000013214
       ./asm/lib.rs:51:17
Frame 2: bkpt @ 0x0000be38 inline
       [...]/cortex-m-0.7.7/src/asm.rs:15:5
```

Yatekii, 2023-11-06:
This should be supported since https://github.com/probe-rs/probe-rs/pull/1755. It supports the semihosting standard now :)

ia0, 2023-11-06:
Thanks for the quick answer! I believe I'm already using a version with that change. So I suspect it's because I'm not doing a semi-hosting breakpoint. I'll take a look tomorrow.

ia0, 2023-11-07:
Ok, I did some testing and I still observe some behavior difference between `probe-rs` and `probe-run`.

TLDR: `probe-rs` expects an RTT (or maybe retries for too long) while `probe-run` [looks at the elf](https://github.com/knurling-rs/probe-run/blob/00c446531c912d31fde9b830667cbc36eb487d7c/src/main.rs#L339-L344) to decide whether it should connect to the RTT. I'll create a separate bug for that.

Here's the details of what I experimented (one can see the `RTT logs not available; blocking until the device halts` behavior of `probe-run` when I use the "release" feature).

My end goal is to get nice (no backtrace and appropriate exit code) test success and test failure outputs, where a test exit is a special function I control. I would like to implement this test exit function such that the output is nice (as defined above).

### My old `probe-run` setup

I implement my test exit function like that:

```rust
    fn exit(success: bool) -> ! {
        if success {
            loop {
                cortex_m::asm::bkpt();
            }
        } else {
            #[cfg(feature = "debug")]
            panic_probe::hard_fault();
            #[cfg(feature = "release")]
            panic!();
        }
    }
```

I explicitly call `panic_probe::hard_fault()` instead of `panic!()` with the "debug" feature to bypass the panic message. Ideally I would like to also bypass the backtrace but that's not the case:

```
% probe-run --log-format=[...] --chip=[...] [elf]
(HOST) INFO  flashing program (45 pages / 180.00 KiB)
(HOST) INFO  success!
(HOST) INFO  stack measurement was not set up
────────────────────────────────────────────────────────────────────────────────
[... debugging output during the test ...]
0.308765: 42 != 13 at src/lib.rs:39:5
────────────────────────────────────────────────────────────────────────────────
stack backtrace:
  [... lots of output ...]
(HOST) ERROR the program panicked
6%
```

With the "release" feature, I use the `panic-abort` crate, compile with `panic_immediate_abort`, and disable backtraces:

```
% probe-run --log-format=[...] --chip=[...] --backtrace=never [elf]
(HOST) INFO  flashing program (47 pages / 188.00 KiB)
(HOST) INFO  success!
RTT logs not available; blocking until the device halts..
────────────────────────────────────────────────────────────────────────────────
────────────────────────────────────────────────────────────────────────────────
(HOST) INFO  program has used at least 1.57/64.00 KiB (2.5%) of stack space
(HOST) ERROR the program panicked
6%
```

In both cases I get a nice error code of 6 from `probe-run`.

A successful test with the "debug" feature looks like:

```
% probe-run --log-format=[...] --chip=[...] [elf]
(HOST) INFO  flashing program (45 pages / 180.00 KiB)
(HOST) INFO  success!
(HOST) INFO  stack measurement was not set up
────────────────────────────────────────────────────────────────────────────────
[... debugging output during the test ...]
────────────────────────────────────────────────────────────────────────────────
(HOST) INFO  device halted without error
%
```

With the "release" feature it looks like:

```
% probe-run --log-format=[...] --chip=[...] --backtrace=never [elf]
(HOST) INFO  flashing program (47 pages / 188.00 KiB)
(HOST) INFO  success!
RTT logs not available; blocking until the device halts..
────────────────────────────────────────────────────────────────────────────────
────────────────────────────────────────────────────────────────────────────────
(HOST) INFO  program has used at least 1.57/64.00 KiB (2.5%) of stack space
(HOST) INFO  device halted without error
%
```

In both cases I get a success code of 0 from `probe-run`, which I can rely on in shell scripts (I can also rely on `probe-run` to always terminate because all tests eventually call the test exit function).

### My new `probe-rs run` setup

I tried changing my test exit function to:

```rust
    fn exit(success: bool) -> ! {
        use cortex_m_semihosting::debug;
        let status = if success { debug::EXIT_SUCCESS } else { debug::EXIT_FAILURE };
        debug::exit(status);
        unreachable!();
    }
```

I get the following good behavior for a failing test with "debug" feature (no backtrace, good exit code):

```
% probe-rs run --chip=[...] [elf]
     Erasing sectors [...]
 Programming pages [...]
    Finished [...]
[... debugging output during the test ...]
0.315078: 42 != 13 at src/lib.rs:39:5
Error: Semihosting indicates exit with failure code: 0x020023 (131107)
1%
```

However, `probe-rs` runs into some timeout (instead of exiting as soon as the semihosting syscall is received) with the "release" feature:

```
% probe-rs run --chip=[...] [elf]
     Erasing sectors [...]
 Programming pages [...]
    Finished [...]
[... I have to wait a long time here: about 10 seconds for an almost instantaneous test ...]
ERROR probe_rs::cmd::run: Failed to attach to RTT continuing...    
Error: Semihosting indicates exit with failure code: 0x020023 (131107)
1%
```

I get a good behavior for a successful test with the "debug" feature (error code of 0):

```
% probe-rs run --chip=[...] [elf]
     Erasing sectors [...]
 Programming pages [...]
    Finished [...]
[... debugging output during the test ...]
%
```

And with the "release" feature, I get the timeout behavior again:

```
% probe-rs run --chip=[...] [elf]
     Erasing sectors [...]
 Programming pages [...]
    Finished [...]
[... I have to wait a long time here: about 10 seconds for an almost instantaneous test ...]
ERROR probe_rs::cmd::run: Failed to attach to RTT continuing...    
%
```

The exit code is correct though (0).

closed 2023-11-07 by ia0