probe-rs/probe-rs #1866: Allow verification through the flash loader CMSIS-Pack `Verfiy` interface

MabezDev, 2023-11-07:
See: https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/algorithmFunc.html#Verify.

This is useful with chips that don't have flash mapped to an address space at boot (any esp32 chip, and more advanced processors). 

Verification is currently achieved by reading flash through the debugger post flash of the entire application using mapped address spaces. Using `Verify` would allow per sector verification during flash, provided the flash loader understands how to verify the flash.

Yatekii, 2023-11-07:
I am not sure I follow what the advantages of this are. If the flash can be mapped arbitrarily, how does the verify function learn this and use it properly?

MabezDev, 2023-11-07:
Because the `Verify` function is part of the flash loader, which will know how to talk to the flash without any mapping. For example, there is no flash mapping at all in the current esp32* flash loaders, just raw writes to the raw address in flash (I.e 0 to FLASH_SIZE).

Yatekii, 2023-11-07:
Ah, yeah that makes sense :) The problem I see with this is this is that we need a loader for every flash possible. But that's something we need a solution for anyways.

MabezDev, 2023-11-07:
> Ah, yeah that makes sense :) The problem I see with this is this is that we need a loader for every flash possible. But that's something we need a solution for anyways.

`Verify` is optional at least, so we can check if the flash loader has it and use that (then hope to god the flash loader understands the flash chip), if its not present we can just default to the current behaviour of trying to verify flash through the mapped address space.

Yatekii, 2023-11-07:
Sounds like a plan to me :)

bugadani, 2024-07-31:
The best part about Verify? We can send don compressed data, making it about 3x as fast as reading back flash! :D

PR #2722 (2024-08-11): Verify command and pre-verify option
Closes #2470
Closes #2686
Closes #1866

The algo is pretty stupid, we could pre-verify per flash block, but per-flashalgo is at least safe even if the flash erase granularity info we have is incorrect.

Open question: should pre-verify be on by default? Obviously it has the potential of doubling the flashing time, but I expect it to be very rare for only the last bytes of a firmware to differ.

Improvement possibilities:

Progress indication. We currently don't give any, and it looks like probe-rs hangs for a few seconds before/after flashing. This isn't id [... cut]
closed 2024-08-11 by Yatekii