probe-rs/probe-rs #1867: Unable to read peripheral memory with GDB

kossnikita, 2023-11-08:
**Describe the bug**
Peripheral reading does not work with gdb server probe-rs. While the rest of the debug generally works fine.

**To Reproduce**
Steps to reproduce the behavior:

1. run server `probe-rs gdb`
2. connect with gdb
3. run `-data-read-memory-bytes <any peripheral>`

**Expected behavior**
Reading peripherals returns correct status data.

**Desktop**

- Windows 10

**Additional context**
See [issue on cortex-debug](https://github.com/Marus/cortex-debug/issues/952)

Yatekii, 2023-11-08:
GDB is a piece of work wrt memory sections because of soft/hard breakpoint management. So we only map out RAM and Flash, but no peripherals, which then prompts GDB to say "hey, that memory does not exist". There is no good way to solve this, but I am happy to accept any solutions :)

bergfi, 2024-07-28:
Hi, I'm running into the same issue. Using the GDB command
```
set mem inaccessible-by-default off
```
I can actually then read the peripherals, but this doesn't seem like a proper way of dealing with this.

What would the steps be to add support for this to the GDB server? I gather it's no easy task.