probe-rs/probe-rs #1881: Add support for Microchip SAM L10/L11

olanod, 2023-11-14:
I'm working on implementing a HAL for the [SAML11](https://ww1.microchip.com/downloads/en/DeviceDoc/SAM-L10L11-Family-DataSheet-DS60001513F.pdf) using the [Xplained Pro evaluation kit](https://ww1.microchip.com/downloads/en/DeviceDoc/SAM-L10-L11-Xplained-Pro-User-Guide-70005359C.pdf) and it would be great to have support from probe-rs. I'm relatively new to the Rust embedded ecosystem and don't know what is needed to add support but happy to help in any way I can.

Yatekii, 2023-11-14:
You can generate it with [target-gen](https://github.com/probe-rs/probe-rs/tree/master/target-gen) :)

olanod, 2023-11-15:
The CMSIS-Pack for the SAML11 didn't include a flash algorithm so `target-gen` didn't work for me. As I'm not qualified to write a flash loader I'll leave it open for anyone who might want to volunteer. :)  
It seems [`edbg`](https://github.com/ataradov/edbg/blob/70bf056c807af9f0f0e7f2f0e68b2717b8cc3ebf/target_mchp_cm23.c#L109) supports this chip so I'll give that a try in the meantime.

bugadani, 2024-05-02:
I have an L10, I can probably manage that. Probably not the L11's fancy additions, though.