probe-rs/probe-rs #1891: Flashing an STM32 target with JLink probe on Linux (Arch) fails unless `--disable-double-buffering` is used

jcard0na, 2023-11-18:
**Describe the bug**

Flashing fails with different errors, sometimes a timeout, sometimes a LockedUp status:

```
    Finished release [optimized + debuginfo] target(s) in 0.06s
     Running `probe-rs run --chip STM32L052C8Tx target/thumbv6m-none-eabi/release/lightnote-rtic`
     Erasing sectors ✔ [00:00:08] [#######################################] 15.00 KiB/15.00 KiB @ 1.78 KiB/s (eta 0s )
 Programming pages   ✔ [00:00:02] [######----------------------------------] 2.00 KiB/15.00 KiB @ 6.21 KiB/s (eta 0s )Error: The flashing procedure failed for 'target/thumbv6m-none-eabi/release/lightnote-rtic'.

Caused by:
    0: The page write of the page at address 0x08000400 failed.
    1: Something during the interaction with the core went wrong
    2: A timeout occurred.
```
I noticed that the address where the error occurs varies from execution to execution. I've observed 0x08000400, 0x08000c00, 0x08001c00
```
    Finished release [optimized + debuginfo] target(s) in 0.07s
     Running `probe-rs run --chip STM32L052C8Tx target/thumbv6m-none-eabi/release/lightnote-rtic`
     Erasing sectors ✔ [00:00:09] [#######################] 20.00 KiB/20.00 KiB @ 2.05 KiB/s (eta 0s )
 Programming pages   ✔ [00:00:01] [##########-------------] 8.00 KiB/20.00 KiB @ 15.93 KiB/s (eta 0s )Error: The flashing procedure failed for 'target/thumbv6m-none-eabi/release/lightnote-rtic'.

Caused by:
    0: The page write of the page at address 0x08001c00 failed.
    1: The core entered an unexpected status: LockedUp.
```

Flashing succeed if `double-buffering` is disabled:
```
    Finished release [optimized + debuginfo] target(s) in 0.10s
     Running `probe-rs run --chip STM32L052C8Tx --disable-double-buffering target/thumbv6m-none-eabi/release/lightnote-rtic`
     Erasing sectors ✔ [00:00:09] [#######################] 20.00 KiB/20.00 KiB @ 2.17 KiB/s (eta 0s )
 Programming pages   ✔ [00:00:04] [#######################] 20.00 KiB/20.00 KiB @ 4.32 KiB/s (eta 0s )    Finished in 13.888s
Hello, world!
```

**To Reproduce**
Steps to reproduce the behavior:

1. `probe-rs run --chip STM32L052C8Tx <file>`

**Expected behavior**
Flashes with no error

**Desktop (please complete the following information):**

- Linux (Arch)

**Additional context**
Target: STM32L052C8
Probe: JLink Base
Host: Arch Linux (Intel i5/Thinkpad)
This is a J-Link Base that works very reliably with Jlink tools as well as with the workaround mentioned above.

Nikita240, 2024-04-02:
Same problem when flashing an `ATSAMD11D14AU` with a J-Link Ultra+ V6. 

However, I also have a J-Link EDU edition, and that one does NOT need `--disable-double-buffering` to work.

Nikita240, 2024-04-02:
Speaking of which, is there a way to enable `--disable-double-buffering` for `cargo embed`?