probe-rs/probe-rs #1892: Please support more platforms for the pre-built binaries

thejpster, 2023-11-18:
**Is your feature request related to a problem? Please describe.**

Installing `probe-run` from source on a lower-power Arm machine (like a Raspberry Pi) takes a really long time to build.

**Describe the solution you'd like**

Please add pre-built 'Aarch64' 64-bit Arm Linux binaries to the release area.

As your CI is free, perhaps you could also consider adding RISC-V 'rv64gc' Linux binaries too.

Yatekii, 2023-11-18:
Good idea, happy to set this up! But I think GH CI does not offer ARM runners, so this requires a bit more setup. Especially for RISC-V.

thejpster, 2023-11-19:
I think you can cross compile using cross-rs, which gives you a container with a suitable sysroot.

Yatekii, 2023-11-19:
Yup that's correct and what I'll do, but it's not as simple as adding it to the existing system matrix and installing 2 dependencies most likely :D We'll see how it goes later today :)

Tiwalun, 2023-11-21:
The easiest solution for this would be  if cargo-dist adds support for cross-compiling: https://github.com/axodotdev/cargo-dist/issues/74

There is already a working `cross.toml` for ARM in the repo, adding support for RISCV there would be a good first step.

liamkinne, 2024-01-01:
It looks like GitHub have a private beta for ARM runners: https://resources.github.com/devops/accelerate-your-cicd-with-arm-and-gpu-runners-in-github-actions/

elpiel, 2024-01-07:
I've successfully used Cross to target RP Pi 3B+, however, the downside was always openssl version. Since the docker image for cross uses != version of openssl compared to Raspberry OS (the latest vawd on Debian bookworm) I haven't had the time to work out a custom built openssl in a custom docker image.

I would love to have these pre-buils binaries too!

I've never been successful to compile it directly on the RP Pi 3B+ over SSH.
I think last time I was running out of memory (OOM) error

thejpster, 2024-01-07:
You can just grab an Arm Debian install, unpack it to a temp folder, and point your Arm Linux C compiler at it. Probably works with Raspberry Pi OS too, although I don't know if you can get that as a tarball or just an SD Card image.

elpiel, 2024-01-08:
@thejpster there are some options indeed:
https://www.raspberrypi.com/software/operating-systems/

But I would really prefer to be able to build it with Cross or just probe-rs providing binaries.
Would make it much easier to build CI/CD pipeline with a RP Pi for everyone then

burrbull, 2024-01-08:
Maybe https://github.com/probe-rs/probe-rs/pull/1842 could simplify support.

thejpster, 2024-01-08:
Certainly! But I think probe-rs still uses openssl for .... uploading logs or something?

elpiel, 2024-01-08:
> Certainly! But I think probe-rs still uses openssl for .... uploading logs or something?

I guess the easiest way would be to run `cargo tree` on the branch with the `cli` feature enabled.

akauppi, 2024-12-21:
FYI, followed the [instructions](https://probe.rs/docs/library/crosscompiling/) today and got a `probe-rs` setup on Raspberry Pi 3B (Raspbian) running with "one evening" of efforts. Documented [here](https://github.com/akauppi/mp/tree/main/cross).

My twist on this is using Multipass VM's (and Docker therein). The "itch" is improving flashing times to ESP32 MCUs from 2min down to 6s.