probe-rs/probe-rs #1910: Unable to detect probe for RP2040

SteveCookTU, 2023-11-27:
**Describe the bug**
After connecting RP Pico W through USB, the CLI tool is unable to find the device using any command.

**To Reproduce**
1. Connect RP Pico W
2. Run `probe-rs list`

**Expected behavior**
A CMSIS-DAP probe should be found.

**Log File**
```
{"timestamp":"2023-11-27T04:35:22.310739Z","level":"INFO","fields":{"message":"Writing log to \"log.txt\""},"target":"probe_rs","filename":"C:\\Users\\stevu\\.cargo\\registry\\src\\index.crates.io-6f17d22bba15001f\\probe-rs-0.21.1\\src\\bin\\probe-rs\\main.rs","line_number":292}
{"timestamp":"2023-11-27T04:35:22.314823Z","level":"DEBUG","fields":{"message":"Searching registry for chip with name RP2040"},"target":"probe_rs::config::registry","filename":"C:\\Users\\stevu\\.cargo\\registry\\src\\index.crates.io-6f17d22bba15001f\\probe-rs-0.21.1\\src\\config\\registry.rs","line_number":172}
{"timestamp":"2023-11-27T04:35:22.314887Z","level":"DEBUG","fields":{"message":"Exact match for chip name: RP2040"},"target":"probe_rs::config::registry","filename":"C:\\Users\\stevu\\.cargo\\registry\\src\\index.crates.io-6f17d22bba15001f\\probe-rs-0.21.1\\src\\config\\registry.rs","line_number":183}
{"timestamp":"2023-11-27T04:35:22.315223Z","level":"DEBUG","fields":{"message":"Partial match for chip name: RP2040_SELFDEBUG"},"target":"probe_rs::config::registry","filename":"C:\\Users\\stevu\\.cargo\\registry\\src\\index.crates.io-6f17d22bba15001f\\probe-rs-0.21.1\\src\\config\\registry.rs","line_number":186}
{"timestamp":"2023-11-27T04:35:22.315389Z","level":"INFO","fields":{"message":"new"},"target":"probe_rs::probe","filename":"C:\\Users\\stevu\\.cargo\\registry\\src\\index.crates.io-6f17d22bba15001f\\probe-rs-0.21.1\\src\\probe.rs","line_number":245,"span":{"name":"list_all"},"spans":[]}
{"timestamp":"2023-11-27T04:35:22.315416Z","level":"INFO","fields":{"message":"enter"},"target":"probe_rs::probe","filename":"C:\\Users\\stevu\\.cargo\\registry\\src\\index.crates.io-6f17d22bba15001f\\probe-rs-0.21.1\\src\\probe.rs","line_number":245,"span":{"name":"list_all"},"spans":[{"name":"list_all"}]}
{"timestamp":"2023-11-27T04:35:22.315437Z","level":"INFO","fields":{"message":"new"},"target":"probe_rs::probe::cmsisdap::tools","filename":"C:\\Users\\stevu\\.cargo\\registry\\src\\index.crates.io-6f17d22bba15001f\\probe-rs-0.21.1\\src\\probe\\cmsisdap\\tools.rs","line_number":15,"span":{"name":"list_cmsisdap_devices"},"spans":[{"name":"list_all"}]}
{"timestamp":"2023-11-27T04:35:22.315443Z","level":"INFO","fields":{"message":"enter"},"target":"probe_rs::probe::cmsisdap::tools","filename":"C:\\Users\\stevu\\.cargo\\registry\\src\\index.crates.io-6f17d22bba15001f\\probe-rs-0.21.1\\src\\probe\\cmsisdap\\tools.rs","line_number":15,"span":{"name":"list_cmsisdap_devices"},"spans":[{"name":"list_all"},{"name":"list_cmsisdap_devices"}]}
{"timestamp":"2023-11-27T04:35:22.315446Z","level":"DEBUG","fields":{"message":"Searching for CMSIS-DAP probes using libusb"},"target":"probe_rs::probe::cmsisdap::tools","filename":"C:\\Users\\stevu\\.cargo\\registry\\src\\index.crates.io-6f17d22bba15001f\\probe-rs-0.21.1\\src\\probe\\cmsisdap\\tools.rs","line_number":17,"span":{"name":"list_cmsisdap_devices"},"spans":[{"name":"list_all"},{"name":"list_cmsisdap_devices"}]}
{"timestamp":"2023-11-27T04:35:22.358301Z","level":"TRACE","fields":{"message":"Could not read string for interface 0, skipping"},"target":"probe_rs::probe::cmsisdap::tools","filename":"C:\\Users\\stevu\\.cargo\\registry\\src\\index.crates.io-6f17d22bba15001f\\probe-rs-0.21.1\\src\\probe\\cmsisdap\\tools.rs","line_number":81,"span":{"name":"list_cmsisdap_devices"},"spans":[{"name":"list_all"},{"name":"list_cmsisdap_devices"}]}
{"timestamp":"2023-11-27T04:35:22.358326Z","level":"TRACE","fields":{"message":"Could not read string for interface 1, skipping"},"target":"probe_rs::probe::cmsisdap::tools","filename":"C:\\Users\\stevu\\.cargo\\registry\\src\\index.crates.io-6f17d22bba15001f\\probe-rs-0.21.1\\src\\probe\\cmsisdap\\tools.rs","line_number":81,"span":{"name":"list_cmsisdap_devices"},"spans":[{"name":"list_all"},{"name":"list_cmsisdap_devices"}]}
{"timestamp":"2023-11-27T04:35:22.358942Z","level":"TRACE","fields":{"message":"Could not read string for interface 0, skipping"},"target":"probe_rs::probe::cmsisdap::tools","filename":"C:\\Users\\stevu\\.cargo\\registry\\src\\index.crates.io-6f17d22bba15001f\\probe-rs-0.21.1\\src\\probe\\cmsisdap\\tools.rs","line_number":81,"span":{"name":"list_cmsisdap_devices"},"spans":[{"name":"list_all"},{"name":"list_cmsisdap_devices"}]}
{"timestamp":"2023-11-27T04:35:22.358973Z","level":"TRACE","fields":{"message":"Could not read string for interface 1, skipping"},"target":"probe_rs::probe::cmsisdap::tools","filename":"C:\\Users\\stevu\\.cargo\\registry\\src\\index.crates.io-6f17d22bba15001f\\probe-rs-0.21.1\\src\\probe\\cmsisdap\\tools.rs","line_number":81,"span":{"name":"list_cmsisdap_devices"},"spans":[{"name":"list_all"},{"name":"list_cmsisdap_devices"}]}
{"timestamp":"2023-11-27T04:35:22.359569Z","level":"TRACE","fields":{"message":"Could not read string for interface 0, skipping"},"target":"probe_rs::probe::cmsisdap::tools","filename":"C:\\Users\\stevu\\.cargo\\registry\\src\\index.crates.io-6f17d22bba15001f\\probe-rs-0.21.1\\src\\probe\\cmsisdap\\tools.rs","line_number":81,"span":{"name":"list_cmsisdap_devices"},"spans":[{"name":"list_all"},{"name":"list_cmsisdap_devices"}]}
{"timestamp":"2023-11-27T04:35:22.421036Z","level":"TRACE","fields":{"message":"Could not read string for interface 0, skipping"},"target":"probe_rs::probe::cmsisdap::tools","filename":"C:\\Users\\stevu\\.cargo\\registry\\src\\index.crates.io-6f17d22bba15001f\\probe-rs-0.21.1\\src\\probe\\cmsisdap\\tools.rs","line_number":81,"span":{"name":"list_cmsisdap_devices"},"spans":[{"name":"list_all"},{"name":"list_cmsisdap_devices"}]}
{"timestamp":"2023-11-27T04:35:22.421832Z","level":"TRACE","fields":{"message":"Could not read string for interface 0, skipping"},"target":"probe_rs::probe::cmsisdap::tools","filename":"C:\\Users\\stevu\\.cargo\\registry\\src\\index.crates.io-6f17d22bba15001f\\probe-rs-0.21.1\\src\\probe\\cmsisdap\\tools.rs","line_number":81,"span":{"name":"list_cmsisdap_devices"},"spans":[{"name":"list_all"},{"name":"list_cmsisdap_devices"}]}
{"timestamp":"2023-11-27T04:35:22.421838Z","level":"TRACE","fields":{"message":"Could not read string for interface 1, skipping"},"target":"probe_rs::probe::cmsisdap::tools","filename":"C:\\Users\\stevu\\.cargo\\registry\\src\\index.crates.io-6f17d22bba15001f\\probe-rs-0.21.1\\src\\probe\\cmsisdap\\tools.rs","line_number":81,"span":{"name":"list_cmsisdap_devices"},"spans":[{"name":"list_all"},{"name":"list_cmsisdap_devices"}]}
{"timestamp":"2023-11-27T04:35:22.422053Z","level":"DEBUG","fields":{"message":"Found 0 CMSIS-DAP probes using libusb, searching HID"},"target":"probe_rs::probe::cmsisdap::tools","filename":"C:\\Users\\stevu\\.cargo\\registry\\src\\index.crates.io-6f17d22bba15001f\\probe-rs-0.21.1\\src\\probe\\cmsisdap\\tools.rs","line_number":26,"span":{"name":"list_cmsisdap_devices"},"spans":[{"name":"list_all"},{"name":"list_cmsisdap_devices"}]}
{"timestamp":"2023-11-27T04:35:22.431759Z","level":"DEBUG","fields":{"message":"Found 0 CMSIS-DAP probes total"},"target":"probe_rs::probe::cmsisdap::tools","filename":"C:\\Users\\stevu\\.cargo\\registry\\src\\index.crates.io-6f17d22bba15001f\\probe-rs-0.21.1\\src\\probe\\cmsisdap\\tools.rs","line_number":47,"span":{"name":"list_cmsisdap_devices"},"spans":[{"name":"list_all"},{"name":"list_cmsisdap_devices"}]}
{"timestamp":"2023-11-27T04:35:22.431778Z","level":"INFO","fields":{"message":"exit"},"target":"probe_rs::probe::cmsisdap::tools","filename":"C:\\Users\\stevu\\.cargo\\registry\\src\\index.crates.io-6f17d22bba15001f\\probe-rs-0.21.1\\src\\probe\\cmsisdap\\tools.rs","line_number":15,"span":{"name":"list_cmsisdap_devices"},"spans":[{"name":"list_all"}]}
{"timestamp":"2023-11-27T04:35:22.431785Z","level":"INFO","fields":{"message":"close","time.busy":"116ms","time.idle":"12.7µs"},"target":"probe_rs::probe::cmsisdap::tools","filename":"C:\\Users\\stevu\\.cargo\\registry\\src\\index.crates.io-6f17d22bba15001f\\probe-rs-0.21.1\\src\\probe\\cmsisdap\\tools.rs","line_number":15,"span":{"name":"list_cmsisdap_devices"},"spans":[{"name":"list_all"}]}
{"timestamp":"2023-11-27T04:35:22.431814Z","level":"INFO","fields":{"message":"new"},"target":"probe_rs::probe::stlink::tools","filename":"C:\\Users\\stevu\\.cargo\\registry\\src\\index.crates.io-6f17d22bba15001f\\p [... cut ...]
**Desktop (please complete the following information):**

- Windows 11

**Additional context**
After encounter the issue, I used Zadig to install a WinUSB driver for the device which had no change. The device is listed for `Interface 1`. Other issues noted this may be related to an outdated `libusb` but `rusb` builds the library if none is found. I am unsure if the installation docs are accurate to how `libusb` is built for windows since installing through `vcpkg` may not be the only step. `pkg-config` seems to also be required to find the library.

SteveCookTU, 2023-11-27:
Discovered issue on system unrelated to the library or pico.

closed 2023-11-27 by SteveCookTU

eskorzon, 2023-11-29:
What WAS the issue? I am running into the same problem.

SteveCookTU, 2023-11-29:
> What WAS the issue? I am running into the same problem.

I am using a second pico as my debug probe and turns out the USB port on my PC case that I used to flash the probe has some issues. It didn't flash properly so it was not running correctly to be detected as a CMSIS-DAP device