probe-rs/probe-rs #1911: Runner in VSCode behaves strangely

Eplankton, 2023-11-27:
when I use the cargo flash in the terminal, no problem:
```
> cargo flash --chip STM32F429ZITx
   Compiling rust-stm32f4-disco-blinky v0.2.0 (D:\code\EIDE\rust-stm32f4-disco-blinky-main)
    Finished dev [unoptimized + debuginfo] target(s) in 0.29s
    Flashing D:\code\EIDE\rust-stm32f4-disco-blinky-main\target\thumbv7em-none-eabihf\debug\rust-stm32f4-disco-blinky
        WARN probe_rs::config::target > Using custom sequence for ARMv7 STM32F429ZITx
     Erasing sectors ✔ [00:00:00] [################################################################################################################] 16.00 KiB/16.00 KiB @ 38.11 KiB/s (eta 0s ) Programming pages   ✔ [00:00:00] [##################################################################################################################] 6.00 KiB/6.00 KiB @ 31.47 KiB/s (eta 0s )    Finished in 0.63s
```
However, when click the `Run` in the editor:
[image]

it failed:
```
    Finished dev [unoptimized + debuginfo] target(s) in 0.07s
     Running `cargo flash --chip STM32F429ZITx target\thumbv7em-none-eabihf\debug\rust-stm32f4-disco-blinky`
error: unexpected argument 'target\thumbv7em-none-eabihf\debug\rust-stm32f4-disco-blinky' found

Usage: cargo.exe build [OPTIONS]

For more information, try '--help'.
       Error Failed to build the cargo project.

  Caused by:  
          0: Failed to run cargo build: exit code = Some(1).

error: process didn't exit successfully: `cargo flash --chip STM32F429ZITx target\thumbv7em-none-eabihf\debug\rust-stm32f4-disco-blinky` (exit code: 1)
```

My `.cargo/config.toml`
```
[target.thumbv7em-none-eabihf]

[target.'cfg(all(target_arch = "arm", target_os = "none"))']
# make `cargo run` start a GDB session
runner = "cargo flash --chip STM32F429ZITx"

rustflags = [
  # This is needed if your flash or ram addresses are not aligned to 0x10000 in memory.x
  # See https://github.com/rust-embedded/cortex-m-quickstart/pull/95
  "-C", "link-arg=--nmagic",

  # LLD (shipped with the Rust toolchain) is used as the default linker
  "-C", "link-arg=-Tlink.x",
]

[build]
target = "thumbv7em-none-eabihf" # Cortex-M4F and Cortex-M7F (with FPU)
```

noppej, 2023-11-27:
Can you share your `launch.json` and `tasks.json` please?

Yatekii, 2023-11-27:
`cargo-flash` is not a runner. What you want is `runner = "probe-rs run --chip STM32F429ZITx"` in the config :)

Eplankton, 2023-11-28:
> `cargo-flash` is not a runner. What you want is `runner = "probe-rs run --chip STM32F429ZITx"` in the config :)

Great, it works, and also I want to ask about the `Debug` button on the right side of `Run`, how to config it? `My launch.json` is:
```
{
    "version": "0.2.0",
    "configurations": [
        {
            "cwd": "${workspaceRoot}",
            "type": "cortex-debug",
            "request": "launch",
            "name": "OpenOCD",
            "servertype": "openocd",
            "executable": "./target/thumbv7em-none-eabihf/debug/rust-stm32f4-disco-blinky",
            "runToEntryPoint": "main",
            "svdFile": "STM32F429x.svd",
            "liveWatch": {
                "enabled": true,
                "samplesPerSecond": 4
            },
            "device": "STM32F429ZITx",
            "configFiles": [
                "interface/stlink.cfg",
                "target/stm32f4x.cfg"
            ]
        },
        {
            "cwd": "${workspaceRoot}",
            "type": "probe-rs-debug",
            "request": "launch",
            "name": "probe_rs debug",
            //!MODIFY
            "chip": "STM32F429ZITx",
            "coreConfigs": [
                {
                    //!MODIFY
                    "programBinary": "./target/thumbv7em-none-eabihf/debug/rust-stm32f4-disco-blinky"
                }
            ]
        }
    ]
}
```

Tiwalun, 2023-11-29:
I think this is provided by rust-analyzer  (or another plugin)  so you might have more success asking there.