probe-rs/probe-rs #1939: Strange RTT issues on ESP32C3 when using embassy

dzamlo, 2023-12-08:
**Describe the bug**
When running a simple project that use embassy on an ESP32C3 (a ESP32-C3-DevKit-RUST-1 board to be exact), I get strange error when trying to debug using RTT. 

I get different issues:
- Messages not showing
- the following error: `ERROR probe_rs::cmd::run: Failed to attach to RTT continuing...`
- error similat to: `Error reading from RTT: Control block corrupted: write pointer is 2976726101 while buffer size is 1024 for "up" channel 0 (defmt) `
- Messages from past session showing at the start

The code on the chip is running because the led blink as expected.

If I try to remove embassy, I see no issue.

I have no issue when working with an STM32 board (a nucleo board with an ST-LINK)

**To Reproduce**
Steps to reproduce the behavior:

1. `git clone https://gitlab.com/dzamlo/minimal-example-rtt-issue.git`
2. `cd minimal-example-rtt-issue`
3. `DEFMT_LOG=trace cargo run --release`

**Expected behavior**
None of the issue mentionned above happen, the message outputed in the loop show on the screen in real time.

**Stacktrace**
If applicable, add a stacktrace to help explain your problem.

**Desktop (please complete the following information):**

- Linux (Arch)

**Additional context**

```
$ probe-rs --version
probe-rs 0.21.1
git commit: a6705249
```
(Installed with `cargo install probe-rs --features cli --git https://github.com/probe-rs/probe-rs.git`)

dzamlo, 2023-12-09:
I have the same issue with the rtt backend of [esp-println](https://github.com/esp-rs/esp-println)

dzamlo, 2023-12-09:
If I remove the `.await`, using `hal::Delay` to wait instead of `Timer`, but let everything else the same (still an async function, embassy still initialized), it seems to works as expected.

If I use `.await` for something else than a timer, like waiting for a falling edge on a gpio, I get the issues.

dzamlo, 2023-12-10:
After further investigation, it is the call to `core::arch::asm!("wfi");` that cause the issue, so nothing to do with async in itself. I've made a minimal reproduction: https://gitlab.com/dzamlo/minimal-example-rtt-issue/-/tree/wfi-example?ref_type=heads

I get the same issues with an esp32-c6.

dzamlo, 2023-12-10:
After further investigation, it is the call to core::arch::asm!("wfi"); that cause the issue, so nothing to do with async in itself. I've made a minimal reproduction: https://gitlab.com/dzamlo/minimal-example-rtt-issue/-/tree/wfi-example?ref_type=heads

I get the same issues with an esp32-c6.

dzamlo, 2023-12-10:
I can reproduce the issue on a STM32L476 (on a nucleo board with a ST-LINK/V2-1): https://gitlab.com/dzamlo/minimal-example-rtt-issue-arm

In addition to RTT issues, I sometime cannot even flash the board and get one the following error messages:

```
Error: Connecting to the chip was unsuccessful.

Caused by:
    0: An error with the usage of the probe occurred
    1: An error specific to a probe type occurred
    2: Command failed with status SwdDpWait
```

```
Error: Connecting to the chip was unsuccessful.

Caused by:
    0: An ARM specific error occurred.
    1: The debug probe encountered an error.
    2: An error specific to a probe type occurred
    3: Command failed with status JtagNoDeviceConnected
```

The solution to recover is to connect BOOT0 to VCC and unplug, replug the board.

Yatekii, 2023-12-10:
It is know that WFI can cause this dependent on what debug bits you set because the chips disable the debug unit. But we can become more resilient I guess.

Duplicate of #350.

dzamlo, 2023-12-12:
Is there any way to detect this situation ? It would be very helpful to show a link to a page that explain the situtation in this case, instead of just having the logs not working.

Yatekii, 2024-02-01:
It would be nice to at least display a hint I guess, but it's not easy to really know.

urbaniak, 2024-06-01:
got similar issues (using esp-wifi, probe-rs run, defmt-rtt), some messages are missing, sometimes they are "late" by few seconds (logging button press) and `Control block corrupted` error as in first comment

I can spend some time debugging that, but need some hints where to start looking as I'm quite new to embedded

AntBlo, 2024-07-05:
I'm hitting this issue as well.

Did some digging and found this:

> Another thing to note is that the USB device is dependent on both the PLL for the 48 MHz USB PHY clock, as
> well as APB_CLK. Specifically, an APB_CLK of 40 MHz or more is required for proper USB compliant operation,
> although the USB device will still function with most hosts with an APB_CLK as low as 10 MHz. Behaviour shown when this happens is dependent on the host USB hardware and drivers, and can include the device
> being unresponsive and it disappearing when first accessed.
> 
> More specifically, the APB_CLK will be affected by clock gating the USB Serial/JTAG Controller, which may
> happen in Light Sleep. Additionally, the USB serial/JTAG Controller (as well as the attached RISC-V CPU) will be
> entirely powered down in Deep Sleep mode. **If a device needs to be debugged in either of these two modes,**
> **it may be preferable to use an external JTAG debugger and serial interface instead.**

https://www.espressif.com/sites/default/files/documentation/esp32-c3_technical_reference_manual_en.pdf#30.4%20Recommended%20Operation

Gonna try an ESP-Prog instead of the built-in JTAG controller.

lure23, 2024-07-29:
Also having this - can help to test possible fixes (but not versed enough in the Rust ecosystem - yet - to make them).

- C3: DevKitC-02 have it
- C3: DevKitRust1 have it
- C6: DevKit-M01 works great

One addition: on C3's, and `probe run --chip=esp32c3` (which is under the `cargo run`s) it takes *ages* from the command to actually start erase+programming. On C6, that's instantanious. Don't know if that matters; just mentioning since no-one had brought that up, yet.

>Note: My usage is over USB/IP which may (or may not) bring its own headaches.

denysvitali, 2025-09-10:
I still have this issue. Is there anything I can do from my side to debug this further and help towards the issue resolution?

bugadani, 2026-01-14:
I believe this issue is why esp-idf has [this workaround](https://github.com/espressif/esp-idf/blob/b9a308a47ca4128d018495662b009a7c461b6780/components/esp_hw_support/cpu.c#L57-L60). probe-rs reads 0 from the device as the RTT read pointer, and thinks it has changed, even when it's only the device that can't return memory properly. This can be worked around in esp-rtos, but trying to force the relevant bit to 1 in probe-rs seems ineffective.

cadeo111, 2026-02-08:
> I believe this issue is why esp-idf has [this workaround](https://github.com/espressif/esp-idf/blob/b9a308a47ca4128d018495662b009a7c461b6780/components/esp_hw_support/cpu.c#L57-L60). probe-rs reads 0 from the device as the RTT read pointer, and thinks it has changed, even when it's only the device that can't return memory properly. This can be worked around in esp-rtos or esp-hal, or in application code by adding this after `esp_hal::init`:
> 
> unsafe {
>     let current = core::ptr::read_volatile(0x600c0008 as *mut u32);
>     core::ptr::write_volatile(0x600c0008 as *mut u32, current | 1<<3);
> }

for other people who find this before the fix is in place in esp-hal:

when debugging over usb with the ESP32-C3-DevKit-RUST-1 this work around didn't work for me  but downgrading probe-rs to 0.30 worked