probe-rs/probe-rs #1948: rtt upstream not working

jhgorse, 2023-12-10:
**Describe the bug**
rtt target (c/c++ SEGGER based implementation) is not receiving the upstream input from the 
`probe-rs run` attachment. rrt downstream is working fine.

Hardware is STM32F4 family. J-Link usb probe via SWD.

**To Reproduce**
Steps to reproduce the behavior:

1. Segger RTT implementation of RTT ~ 7.92o
2. int ch = SEGGER_RTT_WaitKey();
3. never returns

**Expected behavior**
text input after probe-rs run flashes and attaches goes to the upstream rtt buffer and SEGGER_RTT_WaitKey() returns.

**Desktop (please complete the following information):**

- macOS Ventura 13.6.1 (22G313) M2

**Additional context**
I thought this worked when I started testing and then it stopped. Not sure why.

jhgorse, 2023-12-10:
Possibly related https://github.com/probe-rs/probe-rs/discussions/1237

Yatekii, 2023-12-10:
`probe-rs run` does not support down channels, only up channels.

This has never worked and is pending impl :)

Duplicate of #1816.

closed 2023-12-10 by Yatekii