probe-rs/probe-rs #1953: Allow embedding metadata in elf's to pick the right probe (similar to teleprobe-meta).

t-moe, 2023-12-12:
Teleprobe supports embedding metadata in the elf, so that the host tool can pick it up and choose for instance the probe automatically.

With the coming addition of `probe-rs test` (via embedded-test), it would be nice to have something similar in probe-rs.

Although we could add this functionality to all subcommands, not only `probe-rs test`.

Yatekii, 2024-02-01:
Why is that useful? You want an ELF to be flashable with any probe, no?

t-moe, 2024-02-01:
The main idea are HIL Tests, where you have multiple probes connected to a computer and you want to run the different test (suites) on the different targets. 
When using teleprobe you can do exactly that: embedd some metadata into the test and teleprobe will then choose the right probe for you.

But yeah, it might make sense to only add this for the `probe-rs test` subcommand and not to all subcommands in general...