probe-rs/probe-rs #1954: compile error on arm regarding `view_bits` in `probe-rs/src/probe/arm_jtag.rs`

t-moe, 2023-12-12:
**Describe the bug**
The current master branch does not compile on armv7

`rustc --version --verbose`

> rustc 1.73.0 (cc66ad468 2023-10-03)
> binary: rustc
> commit-hash: cc66ad468955717ab92600c770da8c1601a4ff33
> commit-date: 2023-10-03
> host: armv7-unknown-linux-gnueabihf
> release: 1.73.0
> LLVM version: 17.0.2
> 

**To Reproduce**
Steps to reproduce the behavior:

1. Clone the repos master branch
2. `cargo install --path . -- features cli --bin probe-rs`
3. Gives you:

> error[E0599]: the method `view_bits` exists for type `u64`, but its trait bounds were not satisfied
>     --> probe-rs/src/probe/arm_jtag.rs:1309:25
>      |
> 1309 |         let bits = bits.view_bits::<Lsb0>();
>      |                         ^^^^^^^^^ method cannot be called on `u64` due to unsatisfied trait bounds
>      |
>      = note: the following trait bounds were not satisfied:
>              `u64: bitvec::store::BitStore`
>              which is required by `u64: BitView`
> 
> warning: unused import: `view::BitView`
>  --> probe-rs/src/probe/arm_jtag.rs:4:26
>   |
> 4 | use bitvec::{prelude::*, view::BitView};
>   |                          ^^^^^^^^^^^^^
>   |
>   = note: `#[warn(unused_imports)]` on by default
> 
> For more information about this error, try `rustc --explain E0599`.
> warning: `probe-rs` (lib) generated 1 warning
> error: could not compile `probe-rs` (lib) due to previous error; 1 warning emitted
> 

**Additional context**
I'm on a raspberry pi here.

Yatekii, 2023-12-12:
I think they did not implement this because it is a 32bit platform :)

t-moe, 2023-12-12:
Yes, you're correct.

Currently, this is the only line which prevents me from building it on the raspberry.

Question is, whether we want to support 32-bit builds in general (and should add a test for it?), or whether that is out of scope.

Yatekii, 2023-12-12:
I am happy supporting it. But it might be tricky when we assume `usize==u64` (I am not sure we do, but I could imagine).
If it's a simple fix, sure :)

PR #2239 (2024-02-26): Remove view_bits call from u64
If this was the only offender, this PR now closes #1954

cc @t-moe
closed 2024-02-26 by Yatekii