probe-rs/probe-rs #1957: Cortex-M55 and Cortex-M85 target-gen support

oscarmeisal, 2023-12-14:
**Describe the solution you'd like**
I would like to generate target YAML files using target-gen with a CMSIS pack for a Cortex-M85 chip.

**Additional context**
Running target-gen yields this error:
```
user@user-laptop:~$ target-gen pack ~/Downloads/Renesas.RA_DFP.5.0.1.pack ~/Downloads/test/
Error: Failed to process file /home/user/Downloads/Renesas.RA_DFP.5.0.1.pack.

Caused by:
    Core 'CortexM85' is not yet supported for target generation.
```

Yatekii, 2023-12-14:
What arch is that? I have never encountered an M85.

oscarmeisal, 2023-12-14:
It's an ARM processor, specifically the ARMv8.1-M architecture (similar to Cortex-M55, which i think is ARMv8). Until october 2023 there were no chips on the market with the cortex-m85, however Renesas just released the RA8M1 MCU which features this core.

Also, its architecture is already supported by rustc (target thumbv8.main-none-eabihf, same as Cortex-M55)

Tiwalun, 2023-12-14:
I think these chips also use a newer version of the debug interface (ARM ADI v6), so there might be more work required to support them. All the chips so far have been using ARM ADI v5.

oscarmeisal, 2023-12-14:
Yes - referring to the Cortex-M85 technical reference manual, this core implements ARM ADI v6.

PR #1959 (2023-12-14): target-gen: Support Cortex-M55 and Cortex-M85
Fix #1957
closed 2023-12-14 by Yatekii