probe-rs/probe-rs #1970: Picoprobe not detected

DagBjorndal, 2023-12-18:
**Describe the bug**
probe-rs 0.21.1 does not detect picoprobe under Raspberry Pi OS on Raspberry Pi 5.

**To Reproduce**
Steps to reproduce the behavior:

1. Connect picoprobe to USB on the Raspberry Pi 5.
2. Verify connection in a terminal window: <b>lsusb</b> produces a line like <i>Bus 001 Device 002: ID 2e8a:000c Raspberry Pi Picoprobe (CMSIS-DAP)</i> among other lines.
3. Ask probe-rs to detect the probe: <b>probe-rs list</b> produces the response <i>No debug probes were found.</i>

**Expected behavior**
On Raspberry Pi 4B, running the same OS, a response like the following is produced <i>[0]: Picoprobe (CMSIS-DAP) (VID: 2e8a, PID: 000c, Serial: E6614C775B1A6D2D, CmsisDap)</i>. A similar response were to be expected on Raspberry Pi 5.

**Desktop**
- Raspberry Pi OS (Bookworm, December 5th 2023)
- Raspberry Pi 5

**Additional context**
Segger J-Link Mini EDU, Segger J-Link OB, Segger J-Link Plus, STLink V2, STLink V2-1, ESP-PROG (RS232-HS) are all detected just fine with probe-rs 0.21.1.

adamgreig, 2023-12-18:
Have you installed the udev rules from https://probe.rs/docs/getting-started/probe-setup/#linux ?

DagBjorndal, 2023-12-18:
No.... Great catch! Thanks

closed 2023-12-18 by DagBjorndal