probe-rs/probe-rs #1971: MIMXRT1170-EVKB | MCU-LINK | no code execution

scalarion, 2023-12-18:
**Describe the bug**
I am quite a beginner with rust and embedded rust. I successfully flashed my program to the MIMXRT1170-EVKB with probe-rs using different methods like probe-rs run, cargo run via probe-rs runner or cargo flash. However, it seems probe-rs does not execute the binary:
[image] 
The integrated MCU-LINK of the MIMXRT1170-EVKB normally flashes the probe led slowly, but after probe-run it starts to flash very quickly, indicating that data transmission is taking place between the probe and the host. However, the application leds do not start blinking which indicates that the binary is not executed. Manually resetting the board after flashing with probe-rs or cargo flash executes the binary and the application leds begin to flash as intended.

Moreover, using VSCode for debugging with probe-rs extension shows similar behavior and manually halting the debugger cannot find the source code for the binary: 
[image]

I am using:
```
$ rustc --version
rustc 1.74.1 (a28077b28 2023-12-04)

$ probe-rs --version
probe-rs 0.21.1
git commit: c252f97
```

VSCode probe-rs extension is v0.21.2 

**To Reproduce**
Steps to reproduce the behavior:

1. clone https://github.com/scalarion/imxrt-rs-1170evkb-led-toggle.git
2. run `probe-rs run target/thumbv7em-none-eabihf/debug/imxrt-rs-1170evkb-led-toggle --chip MIMXRT1170`

**Expected behavior**
- the binary executes on the evk printing rtt messages to the rtthost console respectively to the console window in VSCode
- debugging in VSCode connects to the source code and halts at break points set in source code

**Desktop (please complete the following information):**
- macOS Sonoma 14.2 (Intel)

Yatekii, 2023-12-20:
Does the MIMXRT1170-EVKB have the reset line properly connected?

scalarion, 2023-12-21:
The reset circuit on the MIMXRT1170-EVKB looks like this:

[image]

Among others, TRG_RST can trigger an POR_PIN_RST_B:

[image]

However, TRG_RST seems not to be connected to the on-board debug probe which requires JP5 to be OPEN:

[image]

Connecting an external J-Link or MCU-Link debug probe to J87 and closing JP5 does not work and brings the error Arm(Dap(NoAcknowledge)):

[image] 

However, using J-Link with JLinkGDBServer either via on board probe or external probe and the VSCode Cortex-Debug extension seems to reset the the device correctly and allows step debugging from within VSCode connected to the source:

[image]

Does probe-rs need the "hard-reset" line POR_PIN_RST_B?

Yatekii, 2023-12-22:
Hmm actually, we use a soft reset for the tools you are working with. I had a hunch it might be HW reset.

So it's possible the reset sequence for this target is broken :/

apparentlymart, 2024-01-05:
I don't have experience with this _particular_ target, but while I was working on the support for another model in the same series -- starting over in #1642 -- I found that it required some special handling of reset due to some details of how the boot ROM works.

Specifically, the boot ROM runs in a state where halting debug is disabled and so the typical reset catch doesn't work properly. Instead, there's a documented workaround of placing a watchpoint on a particular address which the boot ROM then accesses just before it passes control to the user application. However, my symptom of having this wrong initially was just being unable to halt-before-boot, which you aren't trying to do here, so this particular thing probably isn't your immediate concern.

The other interesting quirk of the i.MX RT5xx series is that they require a carefully-formatted "header" data structure in the boot ROM, which includes some data interleaved into some architectually-reserved fields in the initial vector table and so it's incompatible with the way the `cortex-m-rt` crate works. However, I see that you are already using `imx_rt::entry` instead of `cortex_m_rt::entry`, so hopefully you've already dealt with this part. I'm mentioning it just because my initial attempts failed with similar symptoms as reported here, and the fix was to correct the header to indicate that my code should run in non-secure mode.

The reference manual for this chip is structured completely differently to the one I was referring to and so I wasn't able to confirm whether similar concerns apply here, but hopefully this comment gives some clues for things that might be worth investigating!

PR #2387 (2024-04-25): Use system reset for MIMXRT1170
Today's MIMXRT1170 debug sequence only intends on supporting the Cortex M7. To simplify that core's support, the PR removes the Cortex M4 from the target. We could later add support for this second core, and the removal commit has notes about how that might work.

The PR updates the reset sequence for the MIMXRT1170. The sequence affects a system reset that passes through the boot ROM. The reset process assumes that the user is resetting into a program available to FlexSPI1.

Similar to #1771, this sequence needs to be aware that the debug port is disabled after [... cut]
closed 2024-04-25 by bugadani