probe-rs/probe-rs #1981: How to attach and read some registers without knowing the exact chip model

HalfSweet, 2023-12-22:
**Is your feature request related to a problem? Please describe.**
I would like to connect and read certain registers through the debugger to get other complete information when I don't know specific information about a chip. For example, IDCODE of the kernel, DBG_IDCODE of the STM32 series, and other specific registers to find out roughly the model of the chip I'm connecting to

**Describe the solution you'd like**
Hopefully some sort of interface can be provided to read it when the exact model is not known

**Describe alternatives you've considered**
I tried using `probe_rs::Session::auto_attach`, where `target` I set to `TargetSelector::Auto`, but this doesn't seem to work and returns `ChipNotFound(ChipAutodetectFailed)`. This is my code:
```rust
use probe_rs::{flashing, Session, Permissions, MemoryInterface, Probe, Lister};
use probe_rs::config::TargetSelector;

fn main() {
    let mut session = Session::auto_attach(TargetSelector::Auto, Permissions::default()).unwrap();

}
```
I also tried using `attach_to_unspecified` in `probe_rs::Probe`, but it returns `Result<(), Error>`, which I am confused about and don't know what to do next

**Additional context**

Yatekii, 2023-12-22:
You can just use `let mut session = Session::auto_attach("<target_name>", Permissions::default()).unwrap();`

elfmimi, 2023-12-23:
In the case of pyOCD , there is `cortex_m` target. which is guaranteed to be able to connect to any variant of the ARM Cortex series cores.
For probe-rs , I suppose specifying `cortex-m0` can connect to other cortex-m variants such as cortex-m4 too.

But it is nice to have some sort of `auto` target and to be able to read/write memory and registers, for ARM and RISC-V target alike.

HalfSweet, 2023-12-24:
Thank you very much @elfmimi  for the solution, after trying it out I found that it works fine on chips with cortex-M0+/M3/M4 cores.
Do I need to disable this issue?

elfmimi, 2023-12-28:
Let me just point out there is an exception. ( there always be. )
That is there exist kind of target which requires TARGETSEL sequence before making a connection.
The well known one is RP2040.
RP2040 cannot be connected with specifying `--chip cortex_m0`.