probe-rs/probe-rs #2001: Xtensa architecture tracking issue

MabezDev, 2024-01-02:
This issue aims to track the progress of adding Xtensa support to probe-rs, in particular the Xtensa esp targets.

- [x] Initial Xtensa support, Xdm communication with halt/resume and memory interface - https://github.com/probe-rs/probe-rs/pull/1928
- [x]  Add detection of on chip flash sizes - https://github.com/probe-rs/probe-rs/pull/1952, 
- [x] Implement Xdm batching - https://github.com/probe-rs/probe-rs/pull/1969
- [x] Implement breakpoint support - https://github.com/probe-rs/probe-rs/pull/1977 
- [x] Support flashing the esp32s3 - https://github.com/probe-rs/probe-rs/pull/1956
- [ ] Support flashing the esp32
  - [ ] Add work around for TDI level reset for 1.8V flash chips 
- [ ] Support flashing the esp32s2 
- [x] Implement compressed transfers - https://github.com/probe-rs/probe-rs/issues/827, https://github.com/probe-rs/probe-rs/pull/1947
- [ ] Add debugger support
  - [ ] probe-rs uses capstone for disassembly which doesn't yet support Xtensa

bugadani, 2024-01-21:
Flashing support status update:

ESP32-S3 should be woring with all supported probes (tested with J-Link, esp-prog, a random FT232H board and espusbjtag).

S2 is hopefully just a matter of debugging the flash loader.

ESP32 is a different story. `probe-rs` is still too limited to support it in at least two ways:

 - There is no (working) way to specify for the flasher where to download the image chunks. This means they will be downloaded next to the flash loader algorithm. This is a problem for complicated device-specific reasons (no suitable single SRAM region). It can be worked around by filling out the SRAM where the flash loader is located but it's one more hack on top of a pile of other hacks already used by the Xtensa loaders.
 - There is currently no way to specify what level TDI should take for states where it is not used. The TDI input is used as a strapping pin which sets the flash chip's voltage level on startup. ESP32 modules come in two variants: with 1.8V flash, or 3.3V. If the input is low level on power up, ESP32 assumes a 3.3V flash chip is connected, and vice versa. This means that currently an **ESP32 connected to probe-rs must not be reset**.

Hecatron, 2024-03-10:
For info using the latest stable version of probe-rs under windows

## ESP32 - ProS3 Board

I had a quick try with the ProS3 board which is an ESP32S3, that appears to use EspJtag
  * https://esp32s3.com/pros3.html
  * Flashing with probe-rs doesn't currently work
    https://github.com/probe-rs/probe-rs/discussions/2272
  * I tried fhashing with the esptool after restoring the original driver, switching back to winusb
    then trying to attach under vscode, this seemed to allow me to attach I think, although no breakpoints and stepping caused errors

## ESP32-Wroom-32

I had more luck with the ESP32-Wroom-32 board.
Although not with some other boards I tried

  * https://github.com/probe-rs/probe-rs/discussions/2275

bugadani, 2024-03-11:
> Flashing with probe-rs doesn't currently work

That's interesting considering I did most of the work while working with an S3. Can you provide more info?

SergioGasquez, 2024-03-18:
Just created a new Xtensa issue: https://github.com/probe-rs/probe-rs/issues/2301

Rot127, 2024-06-05:
Capstone has now a WIP PR for Xtensa support: https://github.com/capstone-engine/capstone/pull/2380. Better track this one, we hope it can be finished before the v6 release in September.