probe-rs/probe-rs #2023: flash fail on esp32-c3

nim65s, 2024-01-04:
**Describe the bug**

I can't flash an esp32-c3 with probe-rs on 2 different setups. I'm not sure if this is the same issue or not.

**To Reproduce, case 1**

In this case, I am using a esp32-c3-devkit-RUST-1 v1.2a, so with only one USB cable I have a working JTAG

1. clone https://github.com/esp-rs/esp-hal
2. set `runner = "probe-rs run --chip esp32c3"` in `esp32c3-hal/.cargo/config.toml`
3. `cargo run --example blinky --release` from `esp32c3-hal` subdir

```
   Compiling esp32c3-hal v0.14.0 (…/esp-hal/esp32c3-hal)
    Finished release [optimized + debuginfo] target(s) in 0.22s
     Running `probe-rs run --chip esp32c3 target/riscv32imc-unknown-none-elf/release/examples/blinky`
      Erasing ⠁ [00:00:00] [-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------]      0 B/     0 B @      0 B/s (eta 0s )Error: The flashing procedure failed for 'target/riscv32imc-unknown-none-elf/release/examples/blinky'.

Caused by:
    0: Something during the interaction with the core went wrong
    1: An error with the usage of the probe occurred
    2: USB Communication Error
    3: Operation timed out
```

(this was initially reported in https://github.com/esp-rs/esp-hal/issues/1060)

**To Reproduce, case 2**

In this case, I am using a esp32-C3-devkitC-02v1.1, and a ST-link v2, and with the same steps I get:

```
   Compiling esp32c3-hal v0.14.0 (/home/nim/local/esp-rs/esp-hal/esp32c3-hal)
    Finished release [optimized + debuginfo] target(s) in 0.22s
     Running `probe-rs run --chip esp32c3 target/riscv32imc-unknown-none-elf/release/examples/blinky`
Error: Connecting to the chip was unsuccessful.

Caused by:
    0: An error with the usage of the probe occurred
    1: An error specific to a probe type occurred
    2: Command failed with status JtagGetIdcodeError
```

(this was initially reported in https://github.com/esp-rs/esp-hal/issues/1061)

**Expected behavior**

esp32-c3 should be flashed

**Desktop (please complete the following information):**

Arch

**Additional context, case 1**

- `dmesg`

```
[ 2087.030094] usb 1-6.4.4.2: new full-speed USB device number 39 using xhci_hcd
[ 2087.154933] usb 1-6.4.4.2: New USB device found, idVendor=303a, idProduct=1001, bcdDevice= 1.01
[ 2087.154948] usb 1-6.4.4.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2087.154955] usb 1-6.4.4.2: Product: USB JTAG/serial debug unit
[ 2087.154960] usb 1-6.4.4.2: Manufacturer: Espressif
[ 2087.154965] usb 1-6.4.4.2: SerialNumber: 60:55:F9:BC:45:2C
[ 2087.168153] cdc_acm 1-6.4.4.2:1.0: ttyACM0: USB ACM device
```

- `probe-rs list`

```
The following debug probes were found:
[0]: ESP JTAG (VID: 303a, PID: 1001, Serial: 60:55:F9:BC:45:2C, EspJtag)
```

- `probe-rs info`

```
Probing target via JTAG

No DAP interface was found on the connected probe. Thus, ARM info cannot be printed.
RISC-V Chip:
	IDCODE: 0000005c25
	Version:      0
	Part:         5
	Manufacturer: 1554 (Espressif Systems (Shanghai)  Co Ltd)

Probing target via SWD

Error identifying target using protocol SWD: Probe does not support SWD
```

**Additional context, case 2**

- Connections

| FTDI | esp32-c3  | st-link v2 |
| --- | --- | --- |
| TMS | 4 | 7 |
| TDI | 5 | 5 |
| TCK | 6 | 9 |
| TDO | 7 | 13 |

ref. https://www.st.com/resource/en/user_manual/um1075-stlinkv2-incircuit-debuggerprogrammer-for-stm8-and-stm32-stmicroelectronics.pdf
ref. https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/jtag-debugging/configure-other-jtag.html

- `dmesg` for esp32-c3

```
[ 3545.410960] usb 1-6.4.4.3: new full-speed USB device number 46 using xhci_hcd
[ 3545.526853] usb 1-6.4.4.3: New USB device found, idVendor=10c4, idProduct=ea60, bcdDevice= 1.00
[ 3545.526868] usb 1-6.4.4.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 3545.526875] usb 1-6.4.4.3: Product: CP2102N USB to UART Bridge Controller
[ 3545.526880] usb 1-6.4.4.3: Manufacturer: Silicon Labs
[ 3545.526885] usb 1-6.4.4.3: SerialNumber: d6d63185e0a0eb11b5a5cdacdf749906
[ 3545.532726] cp210x 1-6.4.4.3:1.0: cp210x converter detected
[ 3545.535637] usb 1-6.4.4.3: cp210x converter now attached to ttyUSB0
```

- `dmesg` for st-link v2

```
[ 3567.154350] usb 1-6.4.4.1: new full-speed USB device number 47 using xhci_hcd
[ 3567.276142] usb 1-6.4.4.1: New USB device found, idVendor=0483, idProduct=3748, bcdDevice= 1.00
[ 3567.276158] usb 1-6.4.4.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 3567.276165] usb 1-6.4.4.1: Product: STM32 STLink
[ 3567.276170] usb 1-6.4.4.1: Manufacturer: STMicroelectronics
[ 3567.276175] usb 1-6.4.4.1: SerialNumber: 6ÿn\x06BF127\x15\x13W
```

- `probe-rs list`

```
The following debug probes were found:
[0]: STLink V2 (VID: 0483, PID: 3748, Serial: 36C3BF6E064246313237151357, StLink)
```

- `probe-rs info`

```
Probing target via JTAG

Error identifying target using protocol JTAG: An error with the usage of the probe occurred

Probing target via SWD

Error identifying target using protocol SWD: An error with the usage of the probe occurred
```

MabezDev, 2024-01-04:
stlink is swd only, so you won't be able to flash a esp with that, so I will ignore case 2 here.

As for using the usb-serial-jtag on the esp32c3, its working for me?

Could you try holding the boot pin down whilst plugging in the USB? If the board is resetting because of the previously flashed application fast enough, the usb-serial-jtag device can reset too. Putting it into download mode should stop that behavior.

nim65s, 2024-01-05:
Thanks for your answer. I already have a blinky flashed (with espflash), and when I hold the BOOT button while plugging the USB (or puhing the RST button), the LED is not blinking, so I guess it is effectively in download mode, but I still have the exact same error message, whether I release BOOT before running`cargo run --example blinky --release` or not  😢 

And thanks for the info on stlink. Could you recommend a device with the ability to flash + RTT + debug for esp32-c3 ? I tried a pickit 4, but `probe-rs` says `No debug probes were found.`, and a hs-probe, but I get `The connected probe does not support the interface 'RISC-V'`

As I'm trying to make USB devices, embedded usb-serial-jtag on newer esp32-c3 chips won't be an option for RTT/debug anyways.

bugadani, 2024-01-06:
It's possible we weren't rigorous enough with some of the latest changes. Could you please try copying either https://github.com/probe-rs/probe-rs/blame/f13397ec4feeb4bb3c3ed044a087224fa6f1c34e/probe-rs/targets/esp32c3.yaml or https://github.com/probe-rs/probe-rs/blame/6df7fbd0f656ad5ff9889d198baee0a6af6fd081/probe-rs/targets/esp32c3.yaml to your project root and changing your runner to `runner = "probe-rs run --chip esp32c3 --chip-description-path esp32c3.yaml"`?

nim65s, 2024-01-06:
Thanks @bugadani, it looks like this is working for me with both files !

And I can reproduce the previous error by using the file from master https://github.com/probe-rs/probe-rs/blob/master/probe-rs/targets/esp32c3.yaml

 I also have RTT output here.

bugadani, 2024-01-06:
I don't have a C3 so I can't test it, but I re-generated the flash loader [here](https://github.com/probe-rs/probe-rs/blob/5544fcf805a46c4f05bf6d65ee6786bf82a0e0ed/probe-rs/targets/esp32c3.yaml). Can I ask one of you to test if this is better or still broken?

nim65s, 2024-01-06:
This one is still broken

MabezDev, 2024-01-08:
I've reproduced the error now, but it seems somewhat random as to when it happens.

Could you try:

1. Flash with espflash instead of probe-rs
2. Once the app is running (check the serial monitor to see that its not resetting repeatedly)
3. Try flashing with probe-rs again

For me this consistently works provided there is an application already running.

nim65s, 2024-01-10:
Flashing blinky with espflash works, and the monitor looks good. After this, trying to flash again blinky with probe-rs still time out.

noppej, 2024-02-02:
@MabezDev and @bugadani I'm still unable to flash my esp32-c3 with `probe-rs` master branch. `espflash` works fine.

The error I get varies between three: 
1. Same as  what @nim65s encountered
2. 
  ```
Connecting to the chip was unsuccessful.
	"A RISC-V specific error occurred."
		"Timeout during DMI access."
  ```
3. `probe-rs` panics (usually if I did the thing ^^^ with holding the boot button while plugging in USB) with 
  ```
thread 'main' panicked at probe-rs/src/probe/espusbjtag/protocol.rs:152:38:
index out of bounds: the len is 0 but the index is 0
```

noppej, 2024-02-06:
@MabezDev or @bugadani ... I hooked up the UART monitor to my board, and it complains about `SHA-256 comparision failed`. Is this significant or a red-herring?

```
(50) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
(50) boot: compile time Jun  7 2023 07:59:10
(51) boot: chip revision: v0.3
(55) boot.esp32c3: SPI Speed      : 40MHz
(60) boot.esp32c3: SPI Mode       : DIO
(65) boot.esp32c3: SPI Flash Size : 4MB
(69) boot: Enabling RNG early entropy source...
(75) boot: Partition Table:
(78) boot: ## Label            Usage          Type ST Offset   Length
(86) boot:  0 nvs              WiFi data        01 02 00009000 00006000
(93) boot:  1 phy_init         RF data          01 01 0000f000 00001000
(101) boot:  2 factory          factory app      00 00 00010000 003f0000
(108) boot: End of partition table
(112) esp_image: segment 0: paddr=00010020 vaddr=3c030020 size=06f14h ( 28436) map
(127) esp_image: segment 1: paddr=00016f3c vaddr=40380000 size=011fch (  4604) load
(130) esp_image: segment 2: paddr=00018140 vaddr=00000000 size=07ed8h ( 32472) 
(144) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=2fd54h (195924) map
(189) boot: Loaded app from partition at offset 0x10000
(189) boot: Disabling RNG early entropy source...
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd5820,len:0x171c
load:0x403cc710,len:0x968
load:0x403ce710,len:0x2f68
SHA-256 comparison failed:
Calculated: 1d06b938c0222bf626e0bdf46178b1b37ab24d03f0360fc8fcf7153c2571deaf
Expected: 68d7bdf643ba446b8ed7ae8423241d442fd052b2bc77091100ba06fd65dcf8d5
Attempting to boot anyway...
entry 0x403cc710
(50) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
(50) boot: compile time Jun  7 2023 07:59:10
(51) boot: chip revision: v0.3
(55) boot.esp32c3: SPI Speed      : 40MHz
(60) boot.esp32c3: SPI Mode       : DIO
(65) boot.esp32c3: SPI Flash Size : 4MB
(69) boot: Enabling RNG early entropy source...
(75) boot: Partition Table:
(78) boot: ## Label            Usage          Type ST Offset   Length
(86) boot:  0 nvs              WiFi data        01 02 00009000 00006000
(93) boot:  1 phy_init         RF data          01 01 0000f000 00001000
(101) boot:  2 factory          factory app      00 00 00010000 003f0000
(108) boot: End of partition table
(112) esp_image: segment 0: paddr=00010020 vaddr=3c030020 size=06f14h ( 28436) map
(127) esp_image: segment 1: paddr=00016f3c vaddr=40380000 size=011fch (  4604) load
(130) esp_image: segment 2: paddr=00018140 vaddr=00000000 size=07ed8h ( 32472) 
(144) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=2fd54h (195924) map
(189) boot: Loaded app from partition at offset 0x10000
(189) boot: Disabling RNG early entropy source...
```

bugadani, 2024-02-06:
> Is this significant or a red-herring?

It's a bug in espflash, nothing to worry about.

bugadani, 2024-02-07:
No, the SHA comparison problem is a purely cosmetic issue caused by espflash. The library modifies prebuild bootloader images, but does not (yet) recalculate its checksum before handing it back to probe-rs.

noppej, 2024-02-07:
Do you think I have a hardware problem, or are other people (besides OP) also having problems with this chip?

MabezDev, 2024-02-07:
@noppej @nim65s Could you try this branch please: https://github.com/probe-rs/probe-rs/pull/2151.

I noticed the core was still running after attach, which could be very bad since we start poking the flash chip in the sequences, meaning cache pulls and the probe are likely to collide and well after that all bets are off.

noppej, 2024-02-07:
WhootWhoot ... that works for me!  5 times in a row (I'm an experienced optimist ... LOL)

That said, first I had to run into ... ```thread 'main' panicked at probe-rs/src/probe/espusbjtag/protocol.rs:152:38:
index out of bounds: the len is 0 but the index is 0``` ... a few times. The good news, it is reproduce-able if you set `speed 24000` on your connection.

bugadani, 2024-02-07:
> The good news, it is reproduce-able if you set speed 24000

Well then, don't 😅. The error message is consistent with the MCU restarting mid-flash, but 24MHz may also be just too fast for it. I'm assuming you're using an external JTAG probe if this makes a consistent difference - the espusbjtag driver just ignores `--speed`.

MabezDev, 2024-02-07:
> That said, first I had to run into ... thread 'main' panicked at probe-rs/src/probe/espusbjtag/protocol.rs:152:38: index out of bounds: the len is 0 but the index is 0 ... a few times. The good news, it is reproduce-able if you set speed 24000 on your connection.

I've run into that too, it looks like an nusb issue/difference - I should be able to solve that too. Glad to hear it is working though!

noppej, 2024-02-07:
> the espusbjtag driver just ignores `--speed`.

That just proves that my test case was a red herring. I am not using external probe ... direct connect to the esp32-c3 devkit.

After unplugging, and the replugging, I get the error consistently. Then I hold down boot while plugging in, then reset, and then I can flash again. Let me know if you need help debugging.

noppej, 2024-02-07:
> Glad to hear it is working though!

Me too. I thought I was going to have to buy a new dev kit ... except DigiKey Canada sells them with a minimum order quantity of 240 units (Cdn$ 6760.89) ... ROTFL

closed 2024-02-10 by MabezDev

MabezDev, 2024-02-11:
FYI https://github.com/probe-rs/probe-rs/pull/2173 has fixed the index out of range issue on a cold start up of the espusbjtag probe.

juliankrieger, 2024-03-25:
> Do you think I have a hardware problem, or are other people (besides OP) also having problems with this chip?

I am unfortunately also having similar issues with the C3 Devkit M1. Seems like the default speed is too high. Setting it to 10k worked for me so far.

PR #2151 (2024-02-10): Halt the core after attaching so sequences can run freely
This PR adds a new method on Session, which currently isn't public which allows blocks of code to be run when all cores are halted. This is useful for running sequences, especially in the case of the esps, as some sequences poke flash. Doing so during operation of the chip can cause all manner of issues, related to cache pulls.

Closes #2023