probe-rs/probe-rs #2045: WCH-LinkE in CMSIS-DAP mode fails to read DPIDR, failing evaluation

Marcuss2, 2024-01-10:
**Describe the bug**
Examination fails when running due to failure to read DPIDR on RP2040, during first access to the multidrop.

**To Reproduce**
Steps to reproduce the behavior:

1. Set WCH LinkE to CMSIS-DAP mode if not so already (can be done using `wlink mode-switch --dap`)
2. Connect RP2040 to WCH LinkE
3. run probe-rs info --speed 100 --chip rp2040 or run on any binary

**Expected behavior**
No problems, info shows info, binary is flashed and runs

**Desktop (please complete the following information):**

- Linux (Arch)

**Additional context**
OpenOCD works fine. pyOCD also works fine.

Also, when PyOCD is ran first in commander mode `pyocd commander --target rp2040` and then exited using `exit`, probe-rs starts to work correctly as expected.

Logs:
```
$ RUST_LOG=probe_rs=trace probe-rs run target/thumbv6m-none-eabi/debug/universe-dap --speed 100 --chip rp2040                      Wed Jan 10 18:49:17 2024
DEBUG probe_rs::config::registry: Searching registry for chip with name rp2040
DEBUG probe_rs::config::registry: Exact match for chip name: RP2040
DEBUG probe_rs::config::registry: Partial match for chip name: RP2040_SELFDEBUG
 INFO probe_rs::config::target: Using sequence Arm(DefaultArmSequence(()))
DEBUG list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Searching for CMSIS-DAP probes using nusb
DEBUG list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: failed to open usb device to check for cmsisdap interfaces: Os { code: 13, kind: PermissionDenied, message: "Permission denied" } DeviceInfo { bus_number: 4, device_address: 5, vendor_id: 0x0BDA, product_id: 0x0411, device_version: 0x0104, class: 9, subclass: 0, protocol: 3, speed: Some(Super), manufacturer_string: Some("Generic"), product_string: Some("4-Port USB 3.0 Hub"), serial_number: None, path: SysfsPath("/sys/bus/usb/devices/4-2") }
DEBUG list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: failed to open usb device to check for cmsisdap interfaces: Os { code: 13, kind: PermissionDenied, message: "Permission denied" } DeviceInfo { bus_number: 3, device_address: 1, vendor_id: 0x1D6B, product_id: 0x0002, device_version: 0x0606, class: 9, subclass: 0, protocol: 1, speed: Some(High), manufacturer_string: Some("Linux 6.6.10-arch1-1 xhci-hcd"), product_string: Some("xHCI Host Controller"), serial_number: Some("0000:11:00.3"), path: SysfsPath("/sys/bus/usb/devices/usb3") }
DEBUG list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: failed to open usb device to check for cmsisdap interfaces: Os { code: 13, kind: PermissionDenied, message: "Permission denied" } DeviceInfo { bus_number: 3, device_address: 11, vendor_id: 0xFC32, product_id: 0x0287, device_version: 0x0001, class: 0, subclass: 0, protocol: 0, speed: Some(Full), manufacturer_string: Some("JosefAdamcik"), product_string: Some("Sofle"), serial_number: None, path: SysfsPath("/sys/bus/usb/devices/3-1.4") }
DEBUG list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: failed to open usb device to check for cmsisdap interfaces: Os { code: 13, kind: PermissionDenied, message: "Permission denied" } DeviceInfo { bus_number: 1, device_address: 1, vendor_id: 0x1D6B, product_id: 0x0002, device_version: 0x0606, class: 9, subclass: 0, protocol: 1, speed: Some(High), manufacturer_string: Some("Linux 6.6.10-arch1-1 xhci-hcd"), product_string: Some("xHCI Host Controller"), serial_number: Some("0000:03:00.0"), path: SysfsPath("/sys/bus/usb/devices/usb1") }
DEBUG list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: failed to open usb device to check for cmsisdap interfaces: Os { code: 13, kind: PermissionDenied, message: "Permission denied" } DeviceInfo { bus_number: 3, device_address: 10, vendor_id: 0x046D, product_id: 0x085B, device_version: 0x0016, class: 239, subclass: 2, protocol: 1, speed: Some(High), manufacturer_string: None, product_string: Some("Logitech Webcam C925e"), serial_number: Some("0E5A2D5F"), path: SysfsPath("/sys/bus/usb/devices/3-1.2") }
DEBUG list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: failed to open usb device to check for cmsisdap interfaces: Os { code: 13, kind: PermissionDenied, message: "Permission denied" } DeviceInfo { bus_number: 1, device_address: 3, vendor_id: 0x17A0, product_id: 0x0304, device_version: 0x1003, class: 0, subclass: 0, protocol: 0, speed: Some(Full), manufacturer_string: Some("Samson Technologies"), product_string: Some("Samson Q2U Microphone"), serial_number: None, path: SysfsPath("/sys/bus/usb/devices/1-10") }
DEBUG list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: failed to open usb device to check for cmsisdap interfaces: Os { code: 13, kind: PermissionDenied, message: "Permission denied" } DeviceInfo { bus_number: 3, device_address: 12, vendor_id: 0x0BDA, product_id: 0x5411, device_version: 0x0104, class: 9, subclass: 0, protocol: 2, speed: Some(High), manufacturer_string: Some("Generic"), product_string: Some("4-Port USB 2.0 Hub"), serial_number: None, path: SysfsPath("/sys/bus/usb/devices/3-2") }
DEBUG list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: failed to open usb device to check for cmsisdap interfaces: Os { code: 13, kind: PermissionDenied, message: "Permission denied" } DeviceInfo { bus_number: 1, device_address: 2, vendor_id: 0x0B05, product_id: 0x17CB, device_version: 0x0112, class: 255, subclass: 1, protocol: 1, speed: Some(Full), manufacturer_string: Some("Broadcom Corp"), product_string: Some("BCM20702A0"), serial_number: Some("5CF370828A60"), path: SysfsPath("/sys/bus/usb/devices/1-3") }
TRACE list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools:   Interface 0: WCH CMSIS-DAP
TRACE list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Could not read string for interface 1, skipping
TRACE list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Could not read string for interface 2, skipping
TRACE list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: WCH-Link: CMSIS-DAP device with 3 interfaces
TRACE list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: No HID interface for CMSIS-DAP found.
DEBUG list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: failed to open usb device to check for cmsisdap interfaces: Os { code: 13, kind: PermissionDenied, message: "Permission denied" } DeviceInfo { bus_number: 4, device_address: 1, vendor_id: 0x1D6B, product_id: 0x0003, device_version: 0x0606, class: 9, subclass: 0, protocol: 3, speed: Some(Super), manufacturer_string: Some("Linux 6.6.10-arch1-1 xhci-hcd"), product_string: Some("xHCI Host Controller"), serial_number: Some("0000:11:00.3"), path: SysfsPath("/sys/bus/usb/devices/usb4") }
DEBUG list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: failed to open usb device to check for cmsisdap interfaces: Os { code: 13, kind: PermissionDenied, message: "Permission denied" } DeviceInfo { bus_number: 4, device_address: 4, vendor_id: 0x05E3, product_id: 0x0626, device_version: 0x0663, class: 9, subclass: 0, protocol: 3, speed: Some(Super), manufacturer_string: Some("GenesysLogic"), product_string: Some("USB3.1 Hub"), serial_number: None, path: SysfsPath("/sys/bus/usb/devices/4-1") }
DEBUG list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: failed to open usb device to check for cmsisdap interfaces: Os { code: 13, kind: PermissionDenied, message: "Permission denied" } DeviceInfo { bus_number: 2, device_address: 1, vendor_id: 0x1D6B, product_id: 0x0003, device_version: 0x0606, class: 9, subclass: 0, protocol: 3, speed: Some(SuperPlus), manufacturer_string: Some("Linux 6.6.10-arch1-1 xhci-hcd"), product_string: Some("xHCI Host Controller"), serial_number: Some("0000:03:00.0"), path: SysfsPath("/sys/bus/usb/devices/usb2") }
DEBUG list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: failed to open usb device to check for cmsisdap interfaces: Os { code: 13, kind: PermissionDenied, message: "Permission denied" } DeviceInfo { bus_number: 3, device_address: 9, vendor_id: 0x046D, product_id: 0xC08B, device_version: 0x2702, class: 0, subclass: 0, protocol: 0, speed: Some(Full), manufacturer_string: Some("Logitech"), product_string: Some("G502 HERO Gaming Mouse"), serial_number: Some("138C38693232"), path: SysfsPath("/sys/bus/usb/devices/3-1.1") }
DEBUG list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: failed to open usb device to check for cmsisdap interfaces: Os { code: 13, kind: PermissionDenied, message: "Permission denied" } DeviceInfo { bus_number: 3, device_address: 8, vendor_id: 0x05E3, product_id: 0x0610, device_version: 0x0663, class: 9, subclass: 0, pro [... cut ...]
TRACE probe_rs::probe::cmsisdap::commands: Transmit buffer: [00, 03]...
TRACE probe_rs::probe::cmsisdap::commands: Receive buffer: [03, 00]...
TRACE probe_rs::probe::cmsisdap::commands: Transmit buffer: [00, 01, 00]...
TRACE probe_rs::probe::cmsisdap::commands: Receive buffer: [01, 00]...
Error: Connecting to the chip was unsuccessful.

Caused by:
    0: An ARM specific error occurred.
    1: An error occurred in the communication with an access port or debug port.
    2: Target device did not respond to request.
```

Tiwalun, 2024-02-01:
Is this still an issue after #2092 ?

Marcuss2, 2024-02-02:
> Is this still an issue after #2092 ?

Yes, now, --info at least shows some chip information and probe-rs run does work if the chip is specified.

closed 2024-02-02 by Marcuss2