probe-rs/probe-rs #2052: STM32H723ZGTx: Register error: No value for Return Address register.

msrd0, 2024-01-13:
**Describe the bug**

My first attempt at programming my new nucleo board didn't went as expected. The real problem is with my embassy configs, but probe_rs was unable to provide a stack trace and asked me to report a bug:

```
     Erasing sectors ✔ [00:00:03] [##################################################################################################################################] 256.00 KiB/256.00 KiB @ 67.09 KiB/s (eta 0s )
 Programming pages   ✔ [00:00:03] [##################################################################################################################################] 130.00 KiB/130.00 KiB @ 35.83 KiB/s (eta 0s )    Finished in 7.455s
 INFO probe_rs::util::rtt: Initializing RTT
 INFO probe_rs::util::rtt: RTT initialized.
ERROR panicked at 'assertion failed: d1cpre_clk <= d1cpre_clk_max'
└─ embassy_stm32::rcc::_version::init @ /usr/local/share/rust/git/checkouts/embassy-9312dcb0ed774b29/3bc6e41/embassy-stm32/src/fmt.rs:15
ERROR panicked at /usr/local/share/rust/registry/src/github.com-1ecc6299db9ec823/defmt-0.3.5/src/lib.rs:368:5:
explicit panic
└─ panic_probe::print_defmt::print @ /usr/local/share/rust/registry/src/github.com-1ecc6299db9ec823/panic-probe-0.3.1/src/lib.rs:104
 WARN probe_rs::debug::debug_info: UNWIND: Error while checking for exception context. The stack trace will not include the calling frames. : Register error: No value for Return Address register. Please report this as a bug.
Frame 0: "HardFault handler. Cause: Escalated UsageFault (Undefined instruction)." @ 0x08018450
Frame 1: __udf @ 0x08016a3e inline
       ./asm/lib.rs:48:1
Frame 2: __udf @ 0x0000000008016a3e
       ./asm/lib.rs:51:17
Frame 3: disable @ 0x080048b4
       /usr/local/share/rust/registry/src/github.com-1ecc6299db9ec823/cortex-m-0.7.7/src/interrupt.rs:37:1
 INFO session_drop:debug_core_stop: probe_rs::architecture::arm::sequences::stm32h7: Disabling STM32H7 debug components
 INFO probe_rs: Wrote log to "/home/msrd0/.local/share/probe-rs/1705162418638.log"
Error: CPU halted unexpectedly.
```

**To Reproduce**
Steps to reproduce the behavior:

1. Write a program that panics
2. Try to run it through probe-rs

**Expected behavior**

See a full stack trace.

**Desktop (please complete the following information):**

ArchLinux

Yatekii, 2024-02-01:
cc @noppej: see the missing Return Address register :)

noppej, 2024-02-01:
@Yatekii ... that `WARN` message is misleading. It is related to #1667. At some point during refactoring, we unwind deeper than we should. I have WIP to clean that up, and possibly improve the unwind to include the `Reset` handler.

@msrd0 ... the embassy code is causing a panic and a hardfault, which is why `probe-rs` is trying to unwind the stack. 

The cause of the hardfault is here 
```
ERROR panicked at 'assertion failed: d1cpre_clk <= d1cpre_clk_max'
└─ embassy_stm32::rcc::_version::init @ /usr/local/share/rust/git/checkouts/embassy-9312dcb0ed774b29/3bc6e41/embassy-stm32/src/fmt.rs:15
ERROR panicked at /usr/local/share/rust/registry/src/github.com-1ecc6299db9ec823/defmt-0.3.5/src/lib.rs:368:5:
explicit panic
└─ panic_probe::print_defmt::print @ /usr/local/share/rust/registry/src/github.com-1ecc6299db9ec823/panic-probe-0.3.1/src/lib.rs:104
```

.. and probe-rs is unwinding the hardfault with the following 'hint' ... 
```
Frame 0: "HardFault handler. Cause: Escalated UsageFault (Undefined instruction)." @ 0x08018450
```

I recommend you seek help at https://github.com/embassy-rs/embassy/issues or their Matrix channel to determine why your clock configuration is causing an explicit panic and hardfault.

closed 2024-02-01 by noppej

msrd0, 2024-02-02:
Don't worry, I already opened an issue with embassy https://github.com/embassy-rs/embassy/issues/2445