probe-rs/probe-rs #2067: `probe-rs info` can only detect Xtensa chips if probe has no ARM support

bugadani, 2024-01-18:
ARM info reading must be intrusive, because if I run `probe-rs info` with a J-Link, it can't detect my ESP32. However, using an FTDI probe (which has no ARM interface currently) the command gives the expected output:

```
Xtensa Chip:
  IDCODE: 00120034e5
    Version:      1
    Part:         8195
    Manufacturer: 626 (Tensilica)
```

I'm not sure what we can do about this and the issue will be more visible as (hopefully) more probes support ARM interface in the future.

Tiwalun, 2024-01-18:
I think the ARM reading should be called `SWD` reading, that would describe it better I think. It probably uses the same pins, so it might put the JTAG interface in a weird state?

Is there some kind of reset sequence to bring the JTAG interface back into a reset state? 

There is also a protocol flag, if you select jtag it should not do any SWD output. Not sure how xtensa and the riscv jtag part plays together.

bugadani, 2024-01-18:
Sorry I should have specified that `probe-rs info --protocol jtag` has the same issue as it also tries to read ARM chip information.

Tiwalun, 2024-01-18:
If there is no way of resetting the the xtensa JTAG interface, then this is probably not solvable, expect for adding a flag to disable the ARM probing.

PR #2069 (2024-01-19): Do not overwrite detected IR length
Closes #2067

Xtensas and some other MCUs use 5 IR bits. `swj_sequence` overwriting this causes problems in `probe-rs info` where the code forces 4 bits, no matter the detected length. In general, we either measure the IR lengths on attach, or use the value configured by target descriptions, so forcing 4 bits in code should not be necessary. Similarly, we configure IR lengths by setting the scan chain, so a method that sets the length after attach *should* not be necessary.
closed 2024-01-19 by Yatekii