probe-rs/probe-rs #2070: An unwrap unexpectedly panics in `raw_read_register`

ithinuel, 2024-01-19:
**Describe the bug**
An unwrap unexpectedly panics in https://github.com/probe-rs/probe-rs/blob/a5d7838929e58a4e77ed642b83ee6de32b018493/probe-rs/src/probe/cmsisdap/mod.rs#L943-L945

**To Reproduce**
Steps to reproduce the behavior:

1. run tests with defmt-tests.
2. the tests should hang
3. let the computer enter sleep (it happenned over night, not sure exactly what triggered it)

**Expected behavior**
The test session to either still be running or stopped with a message about the probe disconnecting (eg when entering usb suspend) 

**Stacktrace**
```
thread 'main' panicked at /home/ithinuel/.cargo/registry/src/index.crates.io-6f17d22bba15001f/probe-rs-0.22.0/src/probe/cmsisdap/mod.rs:937:16:
called `Option::unwrap()` on a `None` value
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

**Desktop (please complete the following information):**

- Windows 11

**Additional context**

Running probe-rs 0.22

ithinuel, 2024-02-15:
thread 'main' panicked at /home/ithinuel/.cargo/registry/src/index.crates.io-6f17d22bba15001f/probe-rs-0.22.0/src/probe/cmsisdap/mod.rs:937:16:
called `Option::unwrap()` on a `None` value
stack backtrace:
   0: rust_begin_unwind
             at /rustc/30dfb9e046aeb878db04332c74de76e52fb7db10/library/std/src/panicking.rs:647:5
   1: core::panicking::panic_fmt
             at /rustc/30dfb9e046aeb878db04332c74de76e52fb7db10/library/core/src/panicking.rs:72:14
   2: core::panicking::panic
             at /rustc/30dfb9e046aeb878db04332c74de76e52fb7db10/library/core/src/panicking.rs:144:5
   3: <probe_rs::probe::cmsisdap::CmsisDap as probe_rs::architecture::arm::traits::RawDapAccess>::raw_read_register
   4: <T as probe_rs::architecture::arm::ap::ApAccess>::read_ap_register
   5: probe_rs::architecture::arm::memory::adi_v5_memory_interface::ADIMemoryInterface<AP>::read_word_32
   6: <probe_rs::architecture::arm::memory::adi_v5_memory_interface::ADIMemoryInterface<AP> as probe_rs::architecture::arm::memory::adi_v5_memory_interface::ArmProbe>::read_32
   7: probe_rs::architecture::arm::memory::adi_v5_memory_interface::ArmProbe::read_word_32
   8: <probe_rs::architecture::arm::core::armv6m::Armv6m as probe_rs::core::CoreInterface>::hw_breakpoints
   9: probe_rs::core::Core::clear_all_hw_breakpoints
  10: <probe_rs::session::Session as core::ops::drop::Drop>::drop
  11: probe_rs::cmd::run::Cmd::run
  12: probe_rs::main
note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.

ithinuel, 2024-02-15:
Last occurence accually occured a mere seconds after the tests "completed" (hangged in the middle of an operation).

The probe is connected to via a usb3 hub and bound to my vmware VM (windows host, debian guest).
Is there a cmd arg to capture the swd/probe com? I’m happy to to it the cmd line I’m using.
I have no clue about the volume of data involved though.

If there’s a way to set a limit to the file size and/or make a circullar buffer and dump it on exit/panic, that’d be great.

Tiwalun, 2024-02-15:
You should be able to specify `--log-file log.txt` to write the log. You'll have to test it to see how big the file gets, but it will get overwritten again by the next run, so it shouldn't become too big.

ithinuel, 2024-02-15:
It turns out that with that enabled, my test pass :( 
It’s failing understress but the adder load/latency on the host relaxes the pressure too much on the i2c-pio bus driver.