probe-rs/probe-rs #2085: Can't flash MIMXRT1060 - fails to verify flash algorithm

cstrahan, 2024-01-22:
**Describe the bug**

Can't flash MIMXRT1060 targets on lastest release (fails to verify flash algorithm).

```
        WARN probe_rs::config::sequences::nxp_armv7m > MIMXRT10xx core type supplied as Armv7m, but the actual core is a Armv7em
       ERROR probe_rs::flashing::flasher             > Failed to verify flash algorithm. Data mismatch at address 0x000000
       ERROR probe_rs::flashing::flasher             > Original instruction: 0xe00abe00
       ERROR probe_rs::flashing::flasher             > Readback instruction: 0x000000
       ERROR probe_rs::flashing::flasher             > Original: [e00abe00, 62d780d, 24084068, d3000040, 1e644058, 1c49d1fa, 2a001e52, 4770d1f, 4770ba40, 4770ba40, 4770ba40, 4770bac0, 4770bac0, 4770bac0, 4855b51c, 48559000, 8498901, 81010049, 890a4953, 520852, 8802810a, 7522304, da022a00, 439a8802, 88088002, d5020740, 43988808, 484c8008, 6041494a, 6081494b, 22806801, 22204391, 60014311, f8a8f000, 68014847, d54503c9, 60014946, 29006801, 6b01dafc, 3122201, 63014391, 29006b01, 4a42dafc, 60114940, 3092103, 69016101, dafc2900, 22016901, 43910412, 4b3b6101, 3b40493b, 493b6319, 610b2301, 4c3a694b, 544023, 2405191b, 43230264, 698b614b, 40234c36, 4a42403, 4c35191b, 618b4323, 4c3469cb, 24034023, 191b0764, 43234c32, 680161cb, 60014391, 43916b01, 69016301, 61014391, 466a492d, 20004449, f9d3f000, d1052800, 44494929, f9b8f000, d0002800, bd1c2001, 47702000, b5104924, 20004449, f9bbf000, d0002800, bd102001, 7492105, 491e1842, 2301b510, 4449041b, f0002000, 2800f9b8, 2001d000, 2105bd10, 7494613, 49161842, 4449b510, f0002000, 2800f995, 2001d000, bd10, c0000006, 400b8000, 400d0000, d928c520, 400bc000, ffff, 400d8000, 12018, 18131818, 400d8100, f1a2323, 400fc000, fff8e0ff, 1ff3fcff, e0000100, 9c7fff80, 3800001, c, 49d42001, 69c00540, 60084449, 4ad14770, 6812444a, 68526992, 49ce4710, 68094449, 68896989, 4acb4708, 6812444a, 68d26992, 49c84710, 68094449, 69096989, 4ac54708, 6812444a, 69526992, 49c24710, 68094449, 69896989, 49bf4708, 68094449, 69c96989, 49bc4708, 68094449, 6a096989, 4bb94708, 681b444b, 6a5b699b, 48b64718, 68004448, 6ac06980, 48b34700, 68004448, 6bc06980, 49b04700, 68094449, 6c096989, 49ad4708, 68094449, 6c496989, 49aa4708, 68094449, 6b096989, 48a74708, 68004448, 6b406980, 49a44700, 68094449, 6b896989, 48a14708, 68004448, 6c806980, 489e4700, 68004448, 6cc06980, 499b4700, 68094449, 6a896989, 4a984708, 6812444a, 32806992, 47106a52, 444a4a94, 69926812, 6b123280, 4a914710, 6812444a, 32806992, 47106a92, 4448488d, 69806800, 6ac03080, 498a4700, 68094449, 6d096989, 48874708, 68004448, 6d406980, 49844700, 68094449, 6d896989, 48814708, 68004448, 6dc06980, 497e4700, 68094449, 6e096989, 487b4708, 68004448, 6e406980, 49784700, 68094449, 6e896989, 48754708, 68004448, 6ec06980, 49724700, 68094449, 6f096989, 486f4708, 68004448, 6f406980, 496c4700, 68094449, 6f896989, 48694708, 68004448, 6fc06980, 49664700, 68094449, 31806989, 47086809, 44484862, 69806800, 68403080, 495f4700, 68094449, 31806989, 47086889, 444a4a5b, 69926812, 68d23280, 49584710, 68094449, 31806989, 47086909, 444a4a54, 69926812, 69523280, 49514710, 68094449, 31806989, 47086989, 4449494d, 69896809, 69c93180, 494a4708, 68094449, 31806989, 47086a09, 444a4a46, 69926812, 6b523280, 48434710, 68004448, 30806980, 47006b80, 444a4a3f, 69d26812, 47106a12, 4449493c, 69c96809, 47086809, 444a4a39, 69d26812, 47106852, 44494936, 69c96809, 47086889, 44494933, 6a096809, 47086809, 444a4a30, 6a126812, 47106852, 4449492d, 6a096809, 47086889, 4449492a, 6a096809, 47086989, 444a4a27, 6a126812, 471069d2, 44494924, 6a096809, 47086ac9, 444a4a21, 69126812, 47106852, 4c1eb430, 6824444c, 68a46924, bc3046a4, 4a1a4760, 6812444a, 68d26912, 4b174710, 681b444b, 6a5b691b, b4304718, 444c4c13, 69246824, 46a46924, 4760bc30, 9c04b538, 4c0e9400, 6824444c, 69646924, bd3847a0, 4c0ab430, 6824444c, 6a246924, bc3046a4, 4a064760, 6812444a, 69d26912, 49034710, 68094449, 69896909, 4708, 4, b5104770, 89014857, 490849, 49568101, 852890a, 810a0052, 23048802, 2a000752, 8802da02, 8002439a, 7408808, 8808d502, 80084398, 494d484e, 494e6041, 68016081, 43912280, 43112220, 484b6001, 7c96901, 6901d003, 490849, f7ff6101, bd10ffd1, 4d46b5f8, 4a466968, 1804946, 69a8d522, 31b2303, d0074018, 31b2301, d00c1ac0, d1084298, e0066b10, 3c06910, d5066910, f800400, 2000d001, 4608e008, 780e006, 2016d501, 2014e000, 43484936, 896969, e03d0f49, 3c06810, d5066810, f800400, 2700d001, 460fe006, 640e004, e404b2d, 8474358, 3c06b10, d5066b10, f800400, 2400d001, 460ce008, 7c0e006, 2416d001, 2414e000, 43444823, 30404821, 69006a06, f976f000, 23004632, f844f000, 190069a9, 40114a1d, 2201d011, 4b1c0492, d0161a89, d0171a89, d1074291, 7416928, 46380f49, f0001c49, e000f8b0, 69692000, f4904c9, f0001c49, 4912f8a8, 60084449, 6819bdf8, e0000c09, 6896819, f0000e89, 2112f89c, e7ea4348, 400b8000, 400d0000, d928c520, 400bc000, ffff, e000e000, 400fc000, 400d8000, 16e3600, c0000, 400d8100, 8, 4605b5fe, 460c4610, d0734318, 468c46ae, 1aad2000, 419c4601, 4666d367, 24012700, 1ab6463d, d302419d, 463a4613, 46652421, 42f4676, 433e0c36, 1ab60c2d, d304419d, 41b0c15, 412432b, 46653410, 62f4676, 433e0a36, 1ab60a2d, d304419d, 21b0e15, 212432b, 46653408, 72f4676, 433e0936, 1ab6092d, d304419d, 11b0f15, 112432b, 46651d24, 7af4676, 433e08b6, 1ab608ad, d304419d, 9b0f95, 92432b, 46651ca4, 7ef4676, 433e0876, 1ab6086d, d31a419d, 415b1892, e0161c64, 46761800, 41494665, 419d1ab7, 90009101, 4660d309, 41981ab1, 4684468e, 99019800, 1c402500, 7dd4169, 432a0852, 1e64085b, 4672d5e6, b0034663, e7ffbdf0, 46012000, 46c046c0, 4623462a, 2200e7f5, 428b0903, a03d32c, d311428b, 469c2300, 4603e04e, d43c430b, 8432200, d331428b, 428b0903, a03d31c, d301428b, e03f4694, 428b09c3, 1cbd301, 41521ac0, 428b0983, 18bd301, 41521ac0, 428b0943, 14bd301, 41521ac0, 428b0903, 10bd301, 41521ac0, 428b08c3, cbd301, 41521ac0, 428b0883, 8bd301, 41521ac0, 428b0843, 4bd301, 41521ac0, d2001a41, 41524601, 47704610, fcae05d, 4249d000, d3001003, 40534240, 469c2200, 428b0903, a03d32d, d312428b, 18922fc, a03ba12, d30c428b, 11920189, d308428b, 11920189, d304428b, d03a0189, e0001192, 9c30989, d301428b, 1ac001cb, 9834152, d301428b, 1ac0018b, 9434152, d301428b, 1ac0014b, 9034152, d301428b, 1ac0010b, 8c34152, d301428b, 1ac000cb, 8834152, d301428b, 1ac0008b, d2d94152, 428b0843, 4bd301, 41521ac0, d2001a41, 46634601, 105b4152, d3014610, 2b004240, 4249d500, 46634770, d300105b, b5014240, 46c02000, bd0246c0, c04b5f8, 4626b28b, 435eb282, 46100c0d, 43580c37, 19800436, 41792100, 436e4616, 4360c37, 41791980, 436e4626, bdf81989, 0, 0, 1f78a400, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
       ERROR probe_rs::flashing::flasher             > Readback: [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, [... cut ...] d624ce2679d96a1e43c4759bd, then it works as I expected.

This is the merge commit that seems to break things: https://github.com/probe-rs/probe-rs/commit/e7f943bc3c0ce7f83f29e34022307e4c1a8d8e6f

Will see if I can sort it out.

**To Reproduce**
Steps to reproduce the behavior:

1. Flash a MIMXRT1060 target via `cargo-embed`

**Expected behavior**

Should flash successfully.

**Stacktrace**

N/A

**Desktop (please complete the following information):**

- macOS Sonoma

**Additional context**

N/A

cstrahan, 2024-01-22:
If I revert the changes made in https://github.com/probe-rs/probe-rs/pull/1676, then this works:

```patch
diff --git a/probe-rs/targets/MIMXRT1060.yaml b/probe-rs/targets/MIMXRT1060.yaml
index 9f2ffe48..93616fc7 100644
--- a/probe-rs/targets/MIMXRT1060.yaml
+++ b/probe-rs/targets/MIMXRT1060.yaml
@@ -20,14 +20,6 @@ variants:
           is_boot_memory: true
           cores:
             - main
-      - !Ram
-          name: ITCM
-          range:
-            start: 0x00000000
-            end:   0x00020000
-          is_boot_memory: false
-          cores:
-            - main
       - !Ram
           name: OCRAM2
           range:
```

```
        WARN probe_rs::config::target > Using custom sequence for MIMXRT10xx
        WARN probe_rs::architecture::arm::sequences::nxp > MIMXRT10xx core type supplied as Armv7m, but the actual core is a Armv7em
     Erasing sectors ✔ [00:00:07] [#############################################################################] 1.12 MiB/1.12 MiB @ 149.95 KiB/s (eta 0s )
 Programming pages   ✔ [00:00:56] [##############################################################################] 1.09 MiB/1.09 MiB @ 19.60 KiB/s (eta 0s )    Finished flashing in 64.667s
        WARN probe_rs::architecture::arm::sequences::nxp > MIMXRT10xx core type supplied as Armv7m, but the actual core is a Armv7em
```

FWIW, I have my flash banks configured as per: https://github.com/mciantyre/teensy4-rs/issues/86#issuecomment-1429052793

Not sure what's wrong / how to fix.

/cc @Hoohaha @mciantyre

bugadani, 2024-01-22:
Maybe the flash loader isn't position independent. You can try specifying the load address to point into `OCRAM27`.

cstrahan, 2024-01-23:
@bugadani thanks for the feedback! Do you know if we have any way to specify a particular address by region name (e.g. `OCRMA2`)? I see `--base-adress` as an option in `cargo flash`, but not sure if it can accept a name (which would be more convenient than specifying a literal address), and maybe I'm missing it, but I don't see where I could configure this for `cargo embed` (per https://github.com/probe-rs/probe-rs/blob/master/probe-rs/src/bin/probe-rs/cmd/cargo_embed/config/default.toml).

bugadani, 2024-01-23:
All I'M suggesting is to replace `load_address: ~` with an actual value in `MIMXRT1060.yaml`. `--base-address` changes where the binary is flashed, so I don't think that would help :)

bugadani, 2024-01-23:
Could you please test https://github.com/probe-rs/probe-rs/pull/2089 for me?

cstrahan, 2024-01-23:
> All I'M suggesting is to replace `load_address: ~` with an actual value in `MIMXRT1060.yaml`. `--base-address` changes where the binary is flashed, so I don't think that would help :)

Ohhh. "load address" here is as in [LMA](https://sourceware.org/binutils/docs/ld/Basic-Script-Concepts.html), right? I must admit that I haven't given much thought (yet) to flash algorithms, and the precise meaning and implications of the settings in the `*.yaml` files.

If I'm following, then is the load address here just needed to know where to read from to verify that what was written to flash is written correctly, by reading it back from the expected Load Memory Address?

(Going to have to wrap my head around [the flash algorithm stuff](https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/flashAlgorithm.html#CreateFPA) sooner or later -- I want to leverage your work in https://github.com/probe-rs/probe-rs/pull/1947 and add a miniz flash algorithm for the MIMXRT1060 😄).

> Could you please test https://github.com/probe-rs/probe-rs/pull/2089 for me?

I'll give that a go once I'm back at my desk in another hour or two -- thanks!

cstrahan, 2024-01-24:
@bugadani I tried copying the `MIMXRT1060.yaml` from your PR, and set my `Embed.toml` like so:

```toml
[default.general]
chip = "MIMXRT1060"
chip_descriptions = ["./MIMXRT1060.yaml"]
```

But I get:

```
     Erasing sectors ⠁ [00:00:00] [###################################]      0 B/     0 B @      0 B/s (eta 0s )       Error failed to flash /Users/cstrahan/src/tmm_flexio_lcd_driver/target/thumbv7em-none-eabihf/release/hello-teensy
             
             Caused by:
                 0: Error while flashing
                 1: Failed to load flash algorithm into RAM at address 201FFFE0. Is there space for the algorithm header?
```

cstrahan, 2024-01-24:
Since the header seems to be 0x20 bytes long (`0x20200000 - 0x201FFFE0`), I tried setting the `load_address` to `0x20200020`, and that flashes successfully.

Is the header *always* 0x20 bytes, or is that dependent on particular flash algo? Offsetting the address by 0x20 _feels_ kinda suspicious, but I dunno. Is that a proper solution?

bugadani, 2024-01-24:
Right, so probe-rs prepends [this](https://github.com/probe-rs/probe-rs/blob/ddb08cb707d6c44d492d94ba899a636dcd28ca22/probe-rs/src/flashing/flash_algorithm.rs#L171) to the flash algo. So you're right that this is going to be somewhat fragile but that's still better than not working. Thanks for the testing!

mciantyre, 2024-01-24:
I don't have a debuggable 1060 target, but I can reproduce something similar on my 1010EVK.

> On a different note, maybe ITCM isn't writeable in the state the CPU is in when the download happens.

That's the gist. The runtime Charles is using will disable ITCM. That runtime state persists up to the download phase. In order to reproduce this, I have to define an equivalent runtime for the program I'm flashing. (This persistence might be considered part of the issue; didn't expect this.)

On the other hand, OCRAM2 is always available. Using it by default should be safe.

Looks like we can change the `!Ram` region ordering in the target file to prefer OCRAM2, without touching the load address. This doesn't work:

```yaml
      - !Ram
          name: ITCM
          # etc...
      - !Ram
          name: OCRAM2
          # etc...
```

But this does work:

```yaml
      - !Ram
          name: OCRAM2
          # etc...
      - !Ram
          name: ITCM
          # etc...
```

PR #2089 (2024-01-25): Fix MIMXRT1060 flash algo placement
Resolves #2085 

Flashing this device seems to have stopped working when a new memory region was added before `OCRAM2`. Let's try and see if forcing the load address helps.

I do understand that a proper solution would include marking ram regions somehow that the flash loader can be placed into them. This is also an issue with Xtensa/ESP32 where the loader must be loaded into IRAM. However, the reasons are different: here, the memory is not writeable (at least at boot), there it's not necessarily executable.
closed 2024-01-25 by MabezDev