probe-rs/probe-rs #2100: commit 06e181e breaks download for a real ST-LINK V2 programmer.

vDorst, 2024-01-26:
**Describe the bug**
Commit: https://github.com/probe-rs/probe-rs/commit/06e181e1742ebf98355e0fea7a3d4349565a1eb8 break the download function for a real stlink v2 programmer

with self compiled version 06e181e1 and above I get this error while try to run my code.
```
# cargo run --release

warning: `stm32_audio` (bin "stm32_audio") generated 13 warnings (run `cargo fix --bin "stm32_audio"` to apply 2 suggestions)
    Finished release [optimized + debuginfo] target(s) in 0.10s
     Running `probe-rs run --chip STM32L4S5ZITx target\thumbv7em-none-eabihf\release\stm32_audio`
Error: Failed to open the debug probe.

Caused by:
    0: Probe could not be created
    1: Probe was not found.
error: process didn't exit successfully: `probe-rs run --chip STM32L4S5ZITx target\thumbv7em-none-eabihf\release\stm32_audio` (exit code: 1)
```

If I downgrade to version before that I works fine.
Funny thing is I was was working with an st nucleo board with a onboard st-link and I had no issues with that commit.

The broken commit still detects the probe with `probe-rs list`.

```
[0]: STLink V2 (VID: 0483, PID: 3748, Serial: 3526333635334245304426302634, StLink
```

**Desktop (please complete the following information):**
- Windows 11 (Rust 1.75)

**Probes used**
* Real one: STLink V2 (VID: 0483, PID: 3748, Serial: 3526333635334245304426302634, StLink) Firmware: V2J38S27
* On a nucleoboard: STLink V2-1 (VID: 0483, PID: 3752, Serial: 066CFF494849887767242126, StLink) Firmware: V2J38M27

bugadani, 2024-01-26:
cc @Dirbaio in case you want to deal with it :)

Dirbaio, 2024-01-26:
can you paste logs with `RUST_LOG=trace`?

closed 2024-01-26 by bugadani

Dirbaio, 2024-01-26:
@vDorst can you confirm whether it works now?

vDorst, 2024-01-27:
@Dirbaio I do that on Monday.

vDorst, 2024-01-29:
@Dirbaio yes this fixed the issue.
Can use both programmers again with the latest commit 153f9092.
Thanks!

Dirbaio, 2024-01-29:
thanks for confirming!

PR #2101 (2024-01-26): stlink: use hex-encoded serial to match selectors.
Regression introduced in #2086

Fixes #2100