probe-rs/probe-rs #2122: Can no longer detect probes on macOS

cstrahan, 2024-01-31:
**Describe the bug**

Probes can't be found on macOS after merging https://github.com/probe-rs/probe-rs/pull/2065 (commit bf909017c2b35e27902fcbd9433e8b3318ebaac0 on master):

```
% uname -a
Darwin Charless-MBP.lan 23.1.0 Darwin Kernel Version 23.1.0: Mon Oct  9 21:27:24 PDT 2023; root:xnu-10002.41.9~6/RELEASE_ARM64_T6000 arm64
```

Before that commit (i.e. 3da458c2ed2a52ec30f2fcdac48a1f41bd67092d):

```
% probe-rs list               
The following debug probes were found:
[0]: Orbtrace (VID: 1209, PID: 3443, Serial: 32A8708163008E06, CmsisDap)
```

After (i.e bf909017c2b35e27902fcbd9433e8b3318ebaac0):

```
% RUST_LOG=trace probe-rs list                         
DEBUG list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Searching for CMSIS-DAP probes using nusb
 INFO list_cmsisdap_devices: nusb::platform::macos_iokit::enumeration: Probing device 4294990396    
 INFO list_cmsisdap_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `bInterfaceNumber`    
 INFO list_cmsisdap_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `USB Interface Name`    
 INFO list_cmsisdap_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `USB Interface Name`    
 INFO list_cmsisdap_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `USB Interface Name`    
 INFO list_cmsisdap_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `USB Interface Name`    
 INFO list_cmsisdap_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `USB Interface Name`    
 INFO list_cmsisdap_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `USB Interface Name`    
 INFO list_cmsisdap_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `USB Interface Name`    
 INFO list_cmsisdap_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `USB Interface Name`    
TRACE list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: interface 0 has no string, skipping
TRACE list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: interface 1 has no string, skipping
TRACE list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: interface 2 has no string, skipping
TRACE list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: interface 3 has no string, skipping
TRACE list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: interface 4 has no string, skipping
TRACE list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: interface 5 has no string, skipping
TRACE list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: interface 6 has no string, skipping
TRACE list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: interface 7 has no string, skipping
DEBUG list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Found 0 CMSIS-DAP probes using nusb, searching HID
DEBUG list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Found 0 CMSIS-DAP probes total
 INFO list_stlink_devices: nusb::platform::macos_iokit::enumeration: Probing device 4294990396    
 INFO list_stlink_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `bInterfaceNumber`    
 INFO list_stlink_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `USB Interface Name`    
 INFO list_stlink_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `USB Interface Name`    
 INFO list_stlink_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `USB Interface Name`    
 INFO list_stlink_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `USB Interface Name`    
 INFO list_stlink_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `USB Interface Name`    
 INFO list_stlink_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `USB Interface Name`    
 INFO list_stlink_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `USB Interface Name`    
 INFO list_stlink_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `USB Interface Name`    
 INFO list_jlink_devices: nusb::platform::macos_iokit::enumeration: Probing device 4294990396    
 INFO list_jlink_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `bInterfaceNumber`    
 INFO list_jlink_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `USB Interface Name`    
 INFO list_jlink_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `USB Interface Name`    
 INFO list_jlink_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `USB Interface Name`    
 INFO list_jlink_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `USB Interface Name`    
 INFO list_jlink_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `USB Interface Name`    
 INFO list_jlink_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `USB Interface Name`    
 INFO list_jlink_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `USB Interface Name`    
 INFO list_jlink_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `USB Interface Name`    
 INFO list_espjtag_devices: nusb::platform::macos_iokit::enumeration: Probing device 4294990396    
 INFO list_espjtag_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `bInterfaceNumber`    
 INFO list_espjtag_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `USB Interface Name`    
 INFO list_espjtag_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `USB Interface Name`    
 INFO list_espjtag_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `USB Interface Name`    
 INFO list_espjtag_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `USB Interface Name`    
 INFO list_espjtag_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `USB Interface Name`    
 INFO list_espjtag_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `USB Interface Name`    
 INFO list_espjtag_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `USB Interface Name`    
 INFO list_espjtag_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `USB Interface Name`    
DEBUG list_wlink_devices: probe_rs::probe::wlink: Searching for WCH-Link(RV) probes
 INFO list_wlink_devices: nusb::platform::macos_iokit::enumeration: Probing device 4294990396    
 INFO list_wlink_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `bInterfaceNumber`    
 INFO list_wlink_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `USB Interface Name`    
 INFO list_wlink_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `USB Interface Name`    
 INFO list_wlink_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `USB Interface Name`    
 INFO list_wlink_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `USB Interface Name`    
 INFO list_wlink_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `USB Interface Name`    
 INFO list_wlink_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `USB Interface Name`    
 INFO list_wlink_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `USB Interface Name`    
 INFO list_wlink_devices: nusb::platform::macos_iokit::enumeration: Device does not have property `USB Interface Name`    
DEBUG list_wlink_devices: probe_rs::probe::wlink: Found 0 WCH-Link probes total
No debug probes were found.
```

**To Reproduce**
Steps to reproduce the behavior:

1. Check out any commit at or after bf909017c2b35e27902fcbd9433e8b3318ebaac0
2. `cargo install --path . --force --features=cli` (from the `probe-rs` dir)
3. `probe-rs list`

**Expected behavior**

Should detect my probe (`Orbtrace (VID: 1209, PID: 3443, Serial: 32A8708163008E06, CmsisDap)`).

**Stacktrace**

N/A

**Desktop (please complete the following information):**

- macOS Sonoma 14.1 (23B74)

**Additional context**

N/A

cstrahan, 2024-01-31:
Strangely, it _can_ find my MCU-LINK probe:

```
% probe-rs list
The following debug probes were found:
[0]: MCU-LINK (r0FF) CMSIS-DAP V3.10 (VID: 1fc9, PID: 0143, Serial: ZEHHARUMJ0XEJ, CmsisDap)
```

IORegistryExplorer:

[image]

vs

[image]

cstrahan, 2024-01-31:
lsusb output for Orbtrace:

```
% lsusb -d 1209:3443 -v
Bus 000 Device 001: ID 1209:3443  
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.10
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1209 
  idProduct          0x3443 
  bcdDevice            1.31
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x00a9
    bNumInterfaces          8
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         2
      bFunctionClass          2 [unknown]
      bFunctionSubClass       2 [unknown]
      bFunctionProtocol       0 
      iFunction               0 
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         2 [unknown]
      bInterfaceSubClass      2 [unknown]
      bInterfaceProtocol      1 
      iInterface              0 
      CDC Header:
        bcdCDC               1.10
      CDC Union:
        bMasterInterface        0
        bSlaveInterface         1 
      CDC Call Management:
        bmCapabilities       0x00
        bDataInterface          1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval              11
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
[... 321 lines cut ...]
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x03  EP 3 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
Device Status:     0x0000
  (Bus Powered)
```

</details>

cstrahan, 2024-01-31:
`probe-rs list` **_can_** find the Orbtrace probe on my Linux machine.

So my guess is that there's some subtle detail of how Orbtrace is handling USB such that macOS fails to surface the interface string via `IORegistryEntrySearchCFProperty`: https://github.com/kevinmehall/nusb/pull/29/files#diff-ea9fc7f2a10ed7e08f2ec212feb1197d4e77bc92dbf961b57a3a1429d6c9b59dL90

cstrahan, 2024-01-31:
I believe this should fix the problem: https://github.com/kevinmehall/nusb/pull/37

As things stand currently, I believe the _only_ way probes will resolve on macOS is if the product name includes CMSIS-DAP (or the vid:pid pair are in our list of known probes) -- searching by interface strings simply doesn't work.

Tiwalun, 2024-02-01:
> As things stand currently, I believe the _only_ way probes will resolve on macOS (edit: found out this seems to only affect either later versions of macOS or maybe specifically Apple Silicon hardware) is if the product name includes CMSIS-DAP (or the vid:pid pair are in our list of known probes) -- searching by interface strings simply doesn't work.

Is this still correct with the fix in nusb? This would be kind of a bad regression, and if it worked with libusb we should be able to get it working with nusb as well.

closed 2024-02-03 by Tiwalun

cstrahan, 2024-02-03:
@Tiwalun this is fixed by that PR I wrote over in the `nusb` repo, so now that @bugadani has bumped the version thereof here, we're all good to go :)

PR #2107 (2024-02-03): update nusb, attempt to detach ftdi drivers
Closes #2096

Implemented as https://github.com/kevinmehall/nusb/pull/35

nusb 0.1.6 is released so this probably fixes #2122, too