probe-rs/probe-rs #2131: Error on every invocation claiming not ability to get local time for timestamps.

kareled, 2024-02-02:
**Describe the bug**

On every (or nearly every, once I was succesfull) invocation of `probe-rs` I always get this error message:

```
PS C:\Users\gardas\src\rust> probe-rs chip list
Error: Failed to determine local time for timestamps

Caused by:
    The system's UTC offset could not be determined
```

**To Reproduce**
Steps to reproduce the behavior:

1. cabal install probe-rs --features cli
2. probe-rs --version
or
3. probe-rs chip list

**Expected behavior**
probe-rs working as documented.

**Desktop (please complete the following information):**

- Windows 11

bugadani, 2024-02-02:
Not sure what happened but this was fixed in #2129

kareled, 2024-02-02:
As my cargo install claim I have just 0.22.0 version installed:
```
C:\Users\gardas>cargo install probe-rs --features cli
    Updating crates.io index
     Ignored package `probe-rs v0.22.0` is already installed, use --force to override
```
I guess this is a matter of time of bug fix propagating to real release. Isn't it?

kareled, 2024-02-02:
Indeed, following command is a workaround till the release happen:

```
>cargo install --git https://github.com/probe-rs/probe-rs.git probe-rs --features cli
```
tested here.

closed 2024-02-02 by Yatekii