probe-rs/probe-rs #2140: Debugger: Cannot resolve variables with DW_AT_accessibility attributes

noppej, 2024-02-05:
Rust LLVM [recently introduced the use of `DW_AT_accessibility` attributes on variables](https://github.com/rust-lang/rust/pull/115165). 

This prevents the ability to unwind values of variables which previously worked. 

`probe-rs` debug should be updated to interpret these attributes.

bugadani, 2024-02-05:
I'm hoping this is as simple as adding one new qualifier to where the C `const` and `volatile` is handled (and ignored currently). At least right now while we don't do anything fancy?

noppej, 2024-02-05:
This is actually a problem for RUST variables also. I have not started trying to fix it yet, so not sure what the extent of the fix required is. If you want to tackle it, I'd be happy to review/test. If not, I'll wait until we merge your cfa PR, so that I can avoid crazy rebase issues.

noppej, 2024-02-06:
> I'm hoping this is as simple as adding one new qualifier to where the C `const` and `volatile` is handled (and ignored currently). At least right now while we don't do anything fancy?

The [DWARF 5 Spec](https://dwarfstd.org/doc/DWARF5.pdf) and current Rust implementation, uses this attribute to specify 'public/private/protected' visibility, so I'm not sure it is related to the C `const` and `volatile` handling - I suspect looking at the  "Section 5.3 : Type Modifier Entries" will set you off on the right path.

closed 2024-02-07 by noppej