probe-rs/probe-rs #2148: `probe-rs` SIGSEGVs when compiled with musl

CuriouslyCurious, 2024-02-06:
**Describe the bug**
Running `probe-rs` with any command after having it compiled using musl instead of glibc results in a SIGSEGV.

**To Reproduce**
Steps to reproduce the behavior:

1. Compile using musl.
2. `cargo run --features=cli --bin probe-rs -- list`
3. `SIGSEGV`
4. Sadness.

**Expected behavior**
Normal CLI behaviour.

**Stacktrace**
```sh
fish: Job 1, 'cargo run --bin probe-rs --feat…' terminated by signal SIGSEGV (Address boundary error)
```

**Desktop (please complete the following information):**

- Alpine Linux 3.20.0_alpha20231219

Tiwalun, 2024-02-14:
How exactly did you compile probe-rs?  I haven't yet gotten it to actually compile with musl, I usually give up when it tries to get libudev.

CuriouslyCurious, 2024-02-28:
> How exactly did you compile probe-rs? I haven't yet gotten it to actually compile with musl, I usually give up when it tries to get libudev.

You can build `probe-rs` with [`eudev-dev`](https://pkgs.alpinelinux.org/package/v3.18/main/x86_64/eudev-dev) installed. 

I just tried this in a VM (actually an LXC container using Alpine Linux 3.18) and then on my local machine (using Alpine Linux edge 3.20) and `probe-rs` seemingly functioned in the VM but not on my actual machine. So, my guess would be there has been some kind of regression in `eudev` between version 3.2.11-r8 and 3.2.14-r1 which sounds delightfully fun to debug.

I have narrowed it down to occur whenever the [Rust libudev bindings](https://github.com/dcuddeback/libudev-rs) are trying to create a new device a stack overflow occurs but unsure as to why.

CuriouslyCurious, 2024-02-28:
I've figured it out and feel deflated that it wasn't a regression or something equally exciting, but instead was just linking shenaniganry.

Quoting the [Rustlang reference](https://doc.rust-lang.org/reference/linkage.html#static-and-dynamic-c-runtimes): 
> Typically targets are linked dynamically by default, but there are exceptions which are static by default such as:
> ...
> x86_64-unknown-linux-musl

So, from what I understand, what was happening was `libudev-rs` wants a link to the `libudev` library file via `libudev-sys` but, since Alpine is a musl-based target, cargo defaulted to statically try to compile `libudev-rs` (and `libudev-sys` into the resulting `probe-rs` binary which then expects to be able to dynamically link `libudev` in at runtime which then fails since `libudev-sys` doesn't actually properly grab the underlying library.

This is a bug in `libudev-sys`, since it doesn't properly link against the `libudev.so` library file in its `build.rs` and thus this subtly can get introduced into a successfully compiled binary. There is a [standing PR](https://github.com/dcuddeback/libudev-sys/pull/16) that fixes this that has not been merged in yet.

However, seeing as the `libudev-sys` crate has not been updated in a bit and `libudev-rs` has a [standing issue](https://github.com/dcuddeback/libudev-rs/issues/25) about whether it is maintained or not, I think it would probably be good to have a note in the README.md for people using musl on how to compile this. (Also adding a musl build to the CI would be cool). At least until the maintainer uploads the fix to the registry.

closed 2024-02-28 by CuriouslyCurious