probe-rs/probe-rs #2170: Crates release failes to compile with --feauters=cli

perlindgren, 2024-02-11:
Crates release seems broken.

```shell
cargo install probe-rs --features=cli

error[E0603]: struct `Formatter` is private
   --> /home/pln/.cargo/registry/src/index.crates.io-6f17d22bba15001f/probe-rs-0.22.0/src/bin/probe-rs/util/rtt.rs:389:36
    |
389 |     formatter: defmt_decoder::log::Formatter,
    |                                    ^^^^^^^^^ private struct
...
```

Tested with 1.76.0 and 1.75 under Windows 10, 1.76 under latest arch (on two different systems).

MabezDev, 2024-02-11:
Duplicate of https://github.com/probe-rs/probe-rs/issues/2168 and https://github.com/probe-rs/probe-rs/issues/2155

closed 2024-02-11 by MabezDev

perlindgren, 2024-02-11:
Seems that it is fixed in the git. I have students doing labs starting tomorrow where they start by installing probe and friends. Should I just point them to the git, or are you (@MabezDev ) planning to make a new release before tomorrow?

MabezDev, 2024-02-11:
Git or `--locked` for the last released version.