probe-rs/probe-rs #2180: Disabled breakpoints

SergioGasquez, 2024-02-13:
**Describe the bug**
When using probe-rs (`main` rev `ff2370fb80e89a6a0fea23d08b767dcba67cf9c9`) in VS Code, I  cant set up breakpoints a few levels deep from `main.rs`.
**To Reproduce**
Steps to reproduce the behavior:

1. See https://github.com/SergioGasquez/probers-bp readme

**Expected behavior**
I can setup breakpoints and they are not disabled

**Stacktrace**

**Desktop (please complete the following information):**
- Arch Linux 

**Additional context**
Reproduced in ESP32C3 with builtin serial jtag and with ESP32C2 using ESP-PROG

SergioGasquez, 2024-02-13:
cc: @noppej

bugadani, 2024-02-24:
Got closed prematurely I believe, sorry about that :)

noppej, 2024-03-09:
@SergioGasquez Are you open to updating your test case to use the new `esp-hal`?  [migration notes here](https://github.com/esp-rs/esp-hal/releases/tag/v0.16.0)

I should have something for you to test your breakpoints soon.

SergioGasquez, 2024-03-09:
> @SergioGasquez Are you open to updating your test case to use the new `esp-hal`? [migration notes here](https://github.com/esp-rs/esp-hal/releases/tag/v0.16.0)
> 
> I should have something for you to test your breakpoints soon.

Just updated it! I haven't tested it on HW nor reproduced the issue again, but I can do it on Monday or whenever you want me to test something! Thanks for all your work into this issue!