probe-rs/probe-rs #2182: struct 'Formatter' is private during install

kkingsbe, 2024-02-13:
When following the install instructions from the official documentation [found here](https://probe.rs/docs/getting-started/installation/) to do a fresh install on a new computer, the `cargo install probe-rs --features cli` command somehow throws a compiler error and fails during the compilation step after downloading all subdependencies. We've tried also installing an older version without success. I was able to install this on my machine without issues last week, but it seems that either the latest build, or compatibility between the latest build and the latest stable version of `rustc` or `cargo`, may have issues.  I found a github issue reported by someone else for the same issue (but in the wrong repo sadly), where they shared more info. [It can be found here.](https://github.com/rust-lang/cargo/issues/13426)

ajwahab, 2024-02-14:
Try `cargo install probe-rs --locked --features cli`

kkingsbe, 2024-02-14:
Perfect, looks like that worked! Thanks

closed 2024-02-14 by kkingsbe