probe-rs/probe-rs #2183: Can't attach RTT to running RP2040

adsnaider, 2024-02-14:
**Describe the bug**
I want to attach to the RTT on my RP2040 chip that is currently running (flashed with probe-rs). When I use the command
`probe-rs attach [my binary] --chip RP2040`, I get the following error message

```
Error: An ARM specific error occurred.

Caused by:
    The core needs to be halted for this operation but was not.
```

I don't believe halting should be necessary in this case. It's also unclear to me how to halt the CPU from the `probe-rs` CLI. 

**To Reproduce**

1. Flash a binary into an RP2040 chip using `DEFMT_LOG=info probe-rs run --chip RP2040`
2. Ctrl-C out of the rtt log once its running
3. Run `probe-rs attach [my binary] --chip RP2040`

**Expected behavior**
`probe-rs attach` should show work and display the RTT log

**Stacktrace**
If applicable, add a stacktrace to help explain your problem.

```                                                                                                                                                                            02/14/24 12:07:35 PM
 INFO probe_rs::config::target: Using sequence Arm(DefaultArmSequence(()))
 INFO attach: probe_rs::probe::cmsisdap: Using protocol SWD
Error: An ARM specific error occurred.

Caused by:
    The core needs to be halted for this operation but was not.

Stack backtrace:
   0: probe_rs::cmd::run::Cmd::run
   1: probe_rs::main
   2: std::sys_common::backtrace::__rust_begin_short_backtrace
   3: std::rt::lang_start::{{closure}}
   4: core::ops::function::impls::<impl core::ops::function::FnOnce<A> for &F>::call_once
             at /rustc/3d0e6bed600c0175628e96f1118293cf44fb97bd/library/core/src/ops/function.rs:284:13
   5: std::panicking::try::do_call
             at /rustc/3d0e6bed600c0175628e96f1118293cf44fb97bd/library/std/src/panicking.rs:552:40
   6: std::panicking::try
             at /rustc/3d0e6bed600c0175628e96f1118293cf44fb97bd/library/std/src/panicking.rs:516:19
   7: std::panic::catch_unwind
             at /rustc/3d0e6bed600c0175628e96f1118293cf44fb97bd/library/std/src/panic.rs:142:14
   8: std::rt::lang_start_internal::{{closure}}
             at /rustc/3d0e6bed600c0175628e96f1118293cf44fb97bd/library/std/src/rt.rs:148:48
   9: std::panicking::try::do_call
             at /rustc/3d0e6bed600c0175628e96f1118293cf44fb97bd/library/std/src/panicking.rs:552:40
  10: std::panicking::try
             at /rustc/3d0e6bed600c0175628e96f1118293cf44fb97bd/library/std/src/panicking.rs:516:19
  11: std::panic::catch_unwind
             at /rustc/3d0e6bed600c0175628e96f1118293cf44fb97bd/library/std/src/panic.rs:142:14
  12: std::rt::lang_start_internal
             at /rustc/3d0e6bed600c0175628e96f1118293cf44fb97bd/library/std/src/rt.rs:148:20
  13: main
  14: __libc_start_call_main
             at ./csu/../sysdeps/nptl/libc_start_call_main.h:58:16
  15: __libc_start_main_impl
             at ./csu/../csu/libc-start.c:392:3
  16: _start
```

**Desktop (please complete the following information):**

Ubuntu 22.04, probe-rs version 0.22.0

**Additional context**
Add any other context about the problem here.

CBJamo, 2024-02-18:
This bug only appears to happen on 0.22, on 0.21 attaching to a running 2040 works.

Yatekii, 2024-02-18:
Does this happen with `master` too? :)

CBJamo, 2024-02-19:
The error is different, but it still doesn't work.

```
probe-rs --version
probe-rs 0.22.0
git commit: v0.22.0-120-gebe2f08c
```

```
probe-rs run --chip rp2040 ./target/thumbv6m-none-eabi/debug/wifi_blinky
      Erasing ✔ [00:00:04] [####################################################################################################################] 404.00 KiB/404.00 KiB @ 84.92 KiB/s (eta 0s )
  Programming ✔ [00:00:09] [####################################################################################################################] 404.00 KiB/404.00 KiB @ 40.97 KiB/s (eta 0s )    Finished in 14.643s
 WARN defmt_decoder::log::format: logger format contains timestamp but no timestamp implementation was provided; consider removing the timestamp (`{t}` or `{T}`) from the logger format or provide a `defmt::timestamp!` implementation
1.567433 WARN  BDC event, incomplete header
└─ cyw43::runner::{impl#1}::rx @ /home/cbjamo/projects/embassy/cyw43/src/fmt.rs:156
1.568663 WARN  BDC event, incomplete header
└─ cyw43::runner::{impl#1}::rx @ /home/cbjamo/projects/embassy/cyw43/src/fmt.rs:156
1.571696 WARN  IRQ DATA_UNAVAILABLE, clearing...
└─ cyw43::runner::{impl#1}::handle_irq::{async_fn#0} @ /home/cbjamo/projects/embassy/cyw43/src/fmt.rs:156
2.204756 WARN  IRQ DATA_UNAVAILABLE, clearing...
└─ cyw43::runner::{impl#1}::handle_irq::{async_fn#0} @ /home/cbjamo/projects/embassy/cyw43/src/fmt.rs:156
2.456896 INFO  led on!
└─ wifi_blinky::____embassy_main_task::{async_fn#0} @ src/bin/wifi_blinky.rs:58
3.464448 INFO  led off!
└─ wifi_blinky::____embassy_main_task::{async_fn#0} @ src/bin/wifi_blinky.rs:62
```

```
probe-rs attach --chip rp2040 ./target/thumbv6m-none-eabi/debug/wifi_blinky
Error: An ARM specific error occurred.

Caused by:
    Timeout occurred during operation.
```

Since that error is distinctly unhelpful, [here it is with RUST_LOG=trace](https://gist.githubusercontent.com/CBJamo/1500f02d2036e06952d7acdd7e5f49fd/raw/bb5babb926f78754a42ae4f890116bc1004bd421/gistfile2.txt).

adsnaider, 2024-02-19:
I was actually seeing the `Error: An ARM specific error occurred.` with probe-rs@0.21

adsnaider, 2024-02-19:
Fascinating, I'm fairly confident I was seeing this error on 0.21.1 when I installed using cargo. However, using the prebuilt binary I see no such issue.

Yatekii, 2024-02-23:
Hmm ok, I guess we have to have a look. Maybe nusb is involved @Dirbaio?

bugadani, 2024-03-30:
I see that #2259 was merged with the intent of fixing this issue. Can someone please confirm that this issue can be closed?

jenshnielsen, 2024-03-30:
@bugadani I can confirm that I was able to do

1. cargo run
2. detach using Ctrl-C
3. Reattach using `probe-rs attach --chip RP2040 --protocol swd 'target\thumbv6m-none-eabi\debug\rp2040-project-template'`

Seeing the expected output after step 3.

Building the basic RP2040 template project, using a Raspberry Pi debug probe. 
probe-rs build locally from a65812021baee279d4df3ca2fec4f6326dfbd17b

closed 2024-03-30 by bugadani