probe-rs/probe-rs #2184: probe-rs cli: Enable non-blocking RTT mode after attaching

Tiwalun, 2024-02-14:
Like in probe-run, it would be nice to have a feature where `probe-rs run` would change the RTT mode to be blocking after attaching.

This enables the usage where RTT is configured to be non-blocking in the firmware, so that it still works without debugger attached, but no data is lost when a debugger is attached.

Ideally, this should be configurable by a flag.

jamesmunns, 2024-02-14:
This is how probe-run does/did the attach dance: https://github.com/knurling-rs/probe-run/blob/b4d99b621fc1de71cb2743fb10e015ff2905915b/src/main.rs#L311-L340

bugadani, 2024-03-30:
Bit of a status update: Starting with #2326 we enable blocking RTT for defmt channels. The CLI does not, but cargo-embed and the debugger has options to configure other channels.

jamesmunns, 2024-09-24:
Sorry, but @bugadani @Tiwalun I'm not sure how to interpret:

> The CLI does not

Does this mean that `probe-rs run` still does not set block if full for defmt channels by default?

jamesmunns, 2024-09-24:
Ah, @9names cleared this up for me. It was a parser failure for me, the answer is "probe-rs run uses the default (set to block if full), but does not provide a way to override this setting. cargo-embed and the debugger ALSO use the default value of 'block if full', and they DO have a way to override this"

closed 2026-01-19 by bugadani