probe-rs/probe-rs #2185: probe-rs run: Consider setting a breakpoint on main to synchronize RTT attach

Tiwalun, 2024-02-14:
probe-run does set a breakpoint on main, and attaches to RTT after that breakpoint is hit. probe-rs-cli currently tries to attach repeatedly.

I'm not sure if we consciously decided to take a different approach, so if anybody knows feel free to explain here and close this.

See https://github.com/knurling-rs/probe-run/blob/b4d99b621fc1de71cb2743fb10e015ff2905915b/src/main.rs#L311-L340

noppej, 2024-02-14:
@Tiwalun If we break on main, before the user code executes, will it not be a case of the RTT control structures have not yet been configured?

noppej, 2024-02-14:
That is a valuable distinction. Our handling and timing of attach should follow different paths then for the two.

Tiwalun, 2024-02-14:
Now I'm just wondering why `rtt-target` actually need an explicit call :smile:

MabezDev, 2024-02-14:
> Now I'm just wondering why `rtt-target` actually need an explicit call 😄

rtt-target works around the possibility of the debugger polling the rtt control block before its fully constructed. It does this by writing the segger string last. However, in practice, I'm not sure this workaround is entirely relevant because I don't think anyone has ever run into it. It might not be worth implementing this workaround, and falling back to how defmt initializes the control block.

Yatekii, 2024-02-15:
It's actually a real world issue that we run into sometimes. But maybe if we do the breakpoint procedure we can mitigate it and get something more out of it even :)

bugadani, 2025-05-10:
Don't forget that SEGGER RTT is predominantly a C-based technology, and probe-rs is not limited to Rust firmware. If we overfit on defmt-rtt, we'll have issues working with the original C library.

jamesmunns, 2025-05-11:
probe-rs could definitely LOOK to see if it finds an entrypoint it recognizes, and then fall back to the current behavior if it doesn't.

This doesn't have to break C code, to make rust code still more convenient.