probe-rs/probe-rs #2207: Latest git: reset not supported by WinUSB.

vDorst, 2024-02-19:
I was converting my own SWO reader tool to the latest probe-rs git.

When connecting to a onboard ST-LINK V2-1 nucleo board which has not closed proberby by the previous program.
I get this error with the latest git commit ebe2f08.
```
    Finished dev [unoptimized + debuginfo] target(s) in 0.18s
     Running `target\debug\swo_robot.exe -s --probe 0483:374b`
Error: USB Communication Error

Caused by:
    reset not supported by WinUSB
error: process didn't exit successfully: `target\debug\swo_robot.exe -s --probe 0483:374b` (exit code: 1)
```

When using the latest stable release v0.22 I get this error
```
    Finished dev [unoptimized + debuginfo] target(s) in 1.26s
     Running `target\debug\swo_robot.exe -s --probe 0483:374b`
Error: An error specific to a probe type occurred

Caused by:
    0: USB
    1: Operation timed out
error: process didn't exit successfully: `target\debug\swo_robot.exe -s --probe 0483:374b` (exit code: 1)
```
**Expected behavior**

Don't know what to except.
By replugging the st-link it works fine with both probe-rs versions.
So this is more of a seens thing then a real bug.

**Desktop (please complete the following information):**
- Windows 11, Rust 1.76

bugadani, 2024-02-19:
cc @Dirbaio 

nusb errors when calling [`reset`](https://github.com/kevinmehall/nusb/blob/6668fa1f212d8c4aa83173194261d3baab22159f/src/platform/windows_winusb/device.rs#L100) [here](https://github.com/probe-rs/probe-rs/blob/ebe2f08ca245c532ff23e6bb02655ae63ed89b91/probe-rs/src/probe/stlink/usb_interface.rs#L283). I'm not sure what we can do here except not trying to reset the probe and hoping it all works out.

Dirbaio, 2024-02-19:
this seems to be a fallback to get the stlink "unstuck", see https://github.com/probe-rs/probe-rs/blob/ebe2f08ca245c532ff23e6bb02655ae63ed89b91/probe-rs/src/probe/stlink/mod.rs#L522-L523

in normal operation stlink shouldn't need a reset, and it works fine for me on Windows. I'd guess there's something else wrong with that stlink, that not even a reset can fix. the fact that it doesn't work on 0.22 either (which *can* do the reset) fits in with that.

it's a bug that we show that error, we should ignore if reset fails with "not supported" errors.

kevinmehall, 2024-02-22:
FWIW, this is a WinUSB limitation, not just nusb.

libusb's WinUSB backend [just cancels transfers and clears the data toggles](https://github.com/libusb/libusb/blob/5c64e52554daa0a7d297154cbc07fee039f8727f/libusb/os/windows_winusb.c#L3372) but that's not at all the same as a full reset of the port. nusb's design philosophy is that it should tell you when you when it can't do something instead of poorly trying to emulate it.

Dirbaio, 2024-02-25:
> When connecting to a onboard ST-LINK V2-1 nucleo board which has not closed proberby by the previous program.

could you provide exact steps to reproduce this? Maybe there's another way to get the probe unstuck.