probe-rs/probe-rs #2212: windows, cmsisdap: incorrect "Probe was not found" error if wrong driver bound with zadig.

Dirbaio, 2024-02-19:
- Use windows, and a cmsisdap probe (example is the Raspberry Pi Debug Probe)
- Use zadig, bind the libusb0 driver to it.
- Try to use probe-rs. You get "probe was not found" errors that make no sense. [Full logs](https://gist.github.com/JomerDev/95f9bb6fdcdc252c0a6a0d64fc069b7c)

Real issue is

```
DEBUG nusb::platform::windows_winusb::enumeration: Probing interface `USB\VID_2E8A&PID_000C&MI_00\7&1503623E&0&0000` of composite device                                                                                                                                                                                       
DEBUG nusb::platform::windows_winusb::enumeration: Driver is Some("libusb0"), not usable.                                                                                                                                                                                                                                      
```

Maybe we can somehow detect this and tell the user to use Zadig again to bind the winusb driver instead.

cc @JomerDev

kevinmehall, 2024-02-25:
This would be detected by an error returned by `claim_interface`. probe-rs is currently silently ignoring errors from that call, but by that point you've decided that it's the interface you want, so you should probably display a message if claiming it fails instead of saying "Probe was not found".

https://github.com/probe-rs/probe-rs/blob/6b60cdbb5d614b995cc554a2622147ee5990f2d0/probe-rs/src/probe/cmsisdap/tools.rs#L204-L216

https://github.com/kevinmehall/nusb/pull/41 makes the error from `claim_interface` more specific in this case. `ErrorKind::Unsupported` roughly maps to "You should use Zadig to bind WinUSB".

Claiming an interface would also fail if another process has it claimed -- `ErrorKind::ResourceBusy` on Linux and macOS, currently `ErrorKind::PermissionDenied` on Windows, not sure we can tell it apart from a permissions issue.

rgrr, 2024-03-11:
Solution could be to not use a probe requiring Zadig.  Zadig has TMO too many compatibility issues.

Question: does the most recent picoprobe still require Zadig?  If yes, I'ld suggest https://github.com/rgrr/yapicoprobe ;-)

rgrr, 2024-03-11:
Thanks for clarification.

I saw the keyword "Zadig" which creates immediate rash on my side.  Now that I have read the last sentence of the description I understand the issue. I'ld write some kind of configuration FAQ.  But perhaps not a bad idea to implement this FAQ in code if the problem is clearly detectable.

Dirbaio, 2024-03-11:
exactly. this issue is about "probe-rs says `probe not found`, but it should say `probe found but you installed the wrong driver with zadig, please use zadig again to install winusb`"