probe-rs/probe-rs #2222: STM32G431KBT erase fails

evilGitHub, 2024-02-23:
**Describe the bug**
When running the command
`probe-rs erase --chip STM32G431KBTx --speed 1000`
the command fails with
`Error: The execution of 'init' failed with code 134217728. This might indicate a problem with the flash algorithm.`
The Chip is not erased. It only gets reset and halts. When I push reset afterwards the factory blink example will still run. 

**To Reproduce**
Steps to reproduce the behaviour:

1. Grab a new Nucleo-32 STM32G431 board from ST
2. Update the Firmware of the included STLINK-V3E to the newest version "V3.J14.M5". Choose "STM32 Debug+Mass storage+VCP" and "HSI/2 8MHz" when using the ST-Link Upgrade tool.
3. run the command "probe-rs erase --chip STM32G431KBTx --speed 1000"

**Expected behavior**
The command finishes with no output an the Chip is erased. The preloaded program will not run any more after a reset.

**Stacktrace**
[erase.log](https://github.com/probe-rs/probe-rs/files/14389127/erase.log)
The attached log file was produced with "probe-rs erase --chip STM32G431KBTx --speed 1000 --log-file erase.log" and resulted in the specified error.

**Desktop (please complete the following information):**
- Windows 10 64bit Enterprise

**Additional context**
- "probe-rs list" gives "[0]: STLink V3 (VID: 0483, PID: 374e, Serial: 003C00173137510639383538, StLink)"
- I confirmed that the hardware is not defect: I flashed a c example with OpenOCD which worked and I tried a second brand new STM32G431 Nucleo32 board which had the same results
- I confirmed "probe-rs erase" command works with another Nucleo32 Board: I tried erasing a STM32L432 Nucleo32 board (different chip). The erase worked.
- I confirmed BOOT0 is tied to ground: 10k
- flashing and running also does not work with probe-rs for the STM32G431KBT: I discovered the error by trying to flash an example from https://github.com/David-OConnor/stm32-hal 
- I confirmed with the owner of that repository that other STM32G431 Chips/Boards work: https://github.com/David-OConnor/stm32-hal/issues/99  he is able to use probe-rs for a STM32G431CBU without errors.

bugadani, 2024-02-23:
Can you try the latest code from the `master` branch? We've had https://github.com/probe-rs/probe-rs/pull/2061 which might be a relevant change since the latest release came out.

evilGitHub, 2024-02-24:
thank you for your quick answer. I used the Head of the master branch. Now erasing and flashing works.