probe-rs/probe-rs #2231: is `probe-rs list` supposed to take 83 seconds?

Machatron, 2024-02-25:
- Windows 10 x86_64
- probe-rs 0.23.0 git commit: crates.io
Although I am using the latest 0.23 that came out today i had the same issue with 0.22. This isn't something newly introduced to this version.

`cargo embed` takes forever to run. It seems the call to probe-rs is the cause. I timed it and it took exactly 83 seconds both times i ran `probe-rs list`. This happens with or without a probe actually plugged in. Even without a probe it takes this long to find out nothing is connected. I compared on an intel mac and it works instantly.

Did the obvious stuff like restart my pc and close all programs that might potentially be hogging a COM port.

Tiwalun, 2024-02-25:
This probably means that an USB device which you have connected behaves weirdly. Try to disconnect other devices, and see if that helps.

bugadani, 2024-02-25:
It's definitely an issue specific to your system. `probe-rs list` takes maybe up to a second on everything I have. Can you try booting a linux live image and trying there, to see if something is going wrong with your Windows installation or with your hardware?

Machatron, 2024-02-25:
works fine in WSL2. and yes the culprit seems to be my corsair m65 pro mouse. weird.

closed 2024-02-25 by Machatron

Machatron, 2024-02-25:
im going to use wsl for now, but is it known why this could potentially be happening? any way something can be done within probe-rs to avoid picking up that corsair mouse?