probe-rs/probe-rs #2236: Confusing `probe_type` on `ProbeFactory`

Tiwalun, 2024-02-26:
I find this field on the `DebugProbeInfo` very confusing now:

https://github.com/probe-rs/probe-rs/blob/6b60cdbb5d614b995cc554a2622147ee5990f2d0/probe-rs/src/probe.rs#L678

I've used this to filter on the `ProbeType`, but after #2027 this now is a reference to a `ProbeFactory`, and it's not obvious at all anymore how it should be used.

It's still possible to filter probes now with something like this:

```rust
    probe_info.probe_type
                    == (&probe_rs::probe::cmsisdap::CmsisDapFactory
                        as &dyn probe_rs::probe::ProbeFactory)
```

but this feels very unergonomic.

bugadani, 2024-02-26:
If trait_upcasting ever gets stabilized, we'll be able to do `probe_info.probe_type.is::<CmsisDapFactory>()`. There was no indication at all what this was used for except to print the probe family in `probe-rs list` 🤷‍♂️

Tiwalun, 2024-02-27:
The sample code I posted was actually wrong, it needs to be something like this:

```rust
       (&*probe_info.probe_type).type_id()
                    == TypeId::of::<probe_rs::probe::cmsisdap::CmsisDapFactory>()
```

PR #2238 (2024-02-27): Add shorthand to filter probe type
Closes #2236
closed 2024-02-27 by bugadani