probe-rs/probe-rs #2237: NXP MCU-Link Probe was not found

Logiase, 2024-02-26:
**Describe the bug**
`probe-rs info` can not found the probe but `probe-rs list` could with LPC55S69 based NXP MCU-Link onboard.

**To Reproduce**

1. `probe-rs info`

**Expected behavior**
`probe-rs` show the information.

**Stacktrace**
If applicable, add a stacktrace to help explain your problem.

**Desktop (please complete the following information):**

- Windows 11 23H2 22631.3155

**Additional context**
```text
# probe-rs list
DEBUG list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Searching for CMSIS-DAP probes using nusb
DEBUG list_cmsisdap_devices: nusb::platform::windows_winusb::enumeration: Probing device "USB\\VID_1FC9&PID_0143\\R3TXVRKA5XWHZ"
TRACE list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools:   Interface 0: MCU-LINK on-board (r0E7) CMSIS-DAP V3.133
TRACE list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: MCU-LINK on-board (r0E7) CMSIS-DAP V3.133: CMSIS-DAP device with 4 interfaces
TRACE list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: No HID interface for CMSIS-DAP found.
DEBUG list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Found 1 CMSIS-DAP probes using nusb, searching HID
DEBUG list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Found 1 CMSIS-DAP probes total
DEBUG list_ftdi_devices: nusb::platform::windows_winusb::enumeration: Probing device "USB\\VID_1FC9&PID_0143\\R3TXVRKA5XWHZ"
DEBUG list_stlink_devices: nusb::platform::windows_winusb::enumeration: Probing device "USB\\VID_1FC9&PID_0143\\R3TXVRKA5XWHZ"
DEBUG list_jlink_devices: nusb::platform::windows_winusb::enumeration: Probing device "USB\\VID_1FC9&PID_0143\\R3TXVRKA5XWHZ"
DEBUG list_espjtag_devices: nusb::platform::windows_winusb::enumeration: Probing device "USB\\VID_1FC9&PID_0143\\R3TXVRKA5XWHZ"
DEBUG list_wlink_devices: probe_rs::probe::wlink: Searching for WCH-Link(RV) probes
DEBUG list_wlink_devices: nusb::platform::windows_winusb::enumeration: Probing device "USB\\VID_1FC9&PID_0143\\R3TXVRKA5XWHZ"
DEBUG list_wlink_devices: probe_rs::probe::wlink: Found 0 WCH-Link probes total
The following debug probes were found:
[0]: MCU-LINK on-board (r0E7) CMSIS-DAP V3.133 (VID: 1fc9, PID: 0143, Serial: R3TXVRKA5XWHZ, CmsisDap)
```

```text
# probe-rs info
DEBUG list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Searching for CMSIS-DAP probes using nusb
DEBUG list_cmsisdap_devices: nusb::platform::windows_winusb::enumeration: Probing device "USB\\VID_1FC9&PID_0143\\R3TXVRKA5XWHZ"
TRACE list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools:   Interface 0: MCU-LINK on-board (r0E7) CMSIS-DAP V3.133
TRACE list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: MCU-LINK on-board (r0E7) CMSIS-DAP V3.133: CMSIS-DAP device with 4 interfaces
TRACE list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: No HID interface for CMSIS-DAP found.
DEBUG list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Found 1 CMSIS-DAP probes using nusb, searching HID
DEBUG list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Found 1 CMSIS-DAP probes total
DEBUG list_ftdi_devices: nusb::platform::windows_winusb::enumeration: Probing device "USB\\VID_1FC9&PID_0143\\R3TXVRKA5XWHZ"
DEBUG list_stlink_devices: nusb::platform::windows_winusb::enumeration: Probing device "USB\\VID_1FC9&PID_0143\\R3TXVRKA5XWHZ"
DEBUG list_jlink_devices: nusb::platform::windows_winusb::enumeration: Probing device "USB\\VID_1FC9&PID_0143\\R3TXVRKA5XWHZ"
DEBUG list_espjtag_devices: nusb::platform::windows_winusb::enumeration: Probing device "USB\\VID_1FC9&PID_0143\\R3TXVRKA5XWHZ"
DEBUG list_wlink_devices: probe_rs::probe::wlink: Searching for WCH-Link(RV) probes
DEBUG list_wlink_devices: nusb::platform::windows_winusb::enumeration: Probing device "USB\\VID_1FC9&PID_0143\\R3TXVRKA5XWHZ"
DEBUG list_wlink_devices: probe_rs::probe::wlink: Found 0 WCH-Link probes total
TRACE probe_rs::probe::cmsisdap::tools: Attempting to open device matching 1fc9:0143:R3TXVRKA5XWHZ
DEBUG nusb::platform::windows_winusb::enumeration: Probing device "USB\\VID_1FC9&PID_0143\\R3TXVRKA5XWHZ"\
TRACE probe_rs::probe::cmsisdap::tools: Trying device DeviceInfo { bus_number: 0, device_address: 8, vendor_id: 0x1FC9, product_id: 0x0143, device_version: 0x0313, class: 0xEF, subclass: 0x02, protocol: 0x01, speed: Some(High), manufacturer_string: None, product_string: Some("MCU-LINK on-board (r0E7) CMSIS-DAP V3.133"), serial_number: Some("R3TXVRKA5XWHZ"), instance_id: "USB\\VID_1FC9&PID_0143\\R3TXVRKA5XWHZ", parent_instance_id: "USB\\ROOT_HUB30\\7&dacba&0&0", port_number: 6, driver: Some("usbccgp"), interfaces: [InterfaceInfo { interface_number: 0, class: 0xFF, subclass: 0x00, protocol: 0x00, interface_string: Some("MCU-LINK on-board (r0E7) CMSIS-DAP V3.133") }, InterfaceInfo { interface_number: 1, class: 0x03, subclass: 0x00, protocol: 0x00, interface_string: Some("MCU-LINK NXP TRACE") }, InterfaceInfo { interface_number: 2, class: 0x02, subclass: 0x02, protocol: 0x00, interface_string: Some("VCOM") }, InterfaceInfo { interface_number: 4, class: 0x03, subclass: 0x00, protocol: 0x00, interface_string: Some("LPCSIO") }] }
TRACE probe_rs::probe::cmsisdap::tools:   Interface 0: MCU-LINK on-board (r0E7) CMSIS-DAP V3.133
TRACE probe_rs::probe::cmsisdap::tools: MCU-LINK on-board (r0E7) CMSIS-DAP V3.133: CMSIS-DAP device with 4 interfaces
TRACE probe_rs::probe::cmsisdap::tools: No HID interface for CMSIS-DAP found.
DEBUG nusb::platform::windows_winusb::device: Creating device for "USB\\VID_1FC9&PID_0143\\R3TXVRKA5XWHZ"
DEBUG nusb::platform::windows_winusb::enumeration: Driver is "usbccgp"
DEBUG nusb::platform::windows_winusb::enumeration: Probing interface `USB\VID_1FC9&PID_0143&MI_00\9&209E5D94&0&0000` of composite device
ERROR nusb::platform::windows_winusb::enumeration: Failed to get DeviceInterfaceGUIDs from registry: The system cannot find the file specified. (os error 2)
DEBUG nusb::platform::windows_winusb::enumeration: Probing interface `USB\VID_1FC9&PID_0143&MI_01\9&209E5D94&0&0001` of composite device
DEBUG nusb::platform::windows_winusb::enumeration: Driver is Some("HidUsb"), not usable.
DEBUG nusb::platform::windows_winusb::enumeration: Probing interface `USB\VID_1FC9&PID_0143&MI_02\9&209E5D94&0&0002` of composite device
DEBUG nusb::platform::windows_winusb::enumeration: Driver is Some("usbser"), not usable.
DEBUG nusb::platform::windows_winusb::enumeration: Probing interface `USB\VID_1FC9&PID_0143&MI_04\9&209E5D94&0&0004` of composite device
DEBUG nusb::platform::windows_winusb::enumeration: Driver is Some("HidUsb"), not usable.
DEBUG probe_rs::probe::cmsisdap::tools: Could not open 1fc9:0143 in CMSIS-DAP v2 mode
DEBUG probe_rs::probe::cmsisdap::tools: Attempting to open 1fc9:0143 in CMSIS-DAP v1 mode
DEBUG nusb::platform::windows_winusb::enumeration: Probing device "USB\\VID_1FC9&PID_0143\\R3TXVRKA5XWHZ"
DEBUG nusb::platform::windows_winusb::enumeration: Probing device "USB\\VID_1FC9&PID_0143\\R3TXVRKA5XWHZ"
DEBUG nusb::platform::windows_winusb::enumeration: Probing device "USB\\VID_1FC9&PID_0143\\R3TXVRKA5XWHZ"
DEBUG nusb::platform::windows_winusb::enumeration: Probing device "USB\\VID_1FC9&PID_0143\\R3TXVRKA5XWHZ"
Error: Failed to open the debug probe.

Caused by:
    0: Probe could not be created
    1: Probe was not found.
```

kevinmehall, 2024-02-27:
```
ERROR nusb::platform::windows_winusb::enumeration: Failed to get DeviceInterfaceGUIDs from registry: The system cannot find the file specified. (os error 2)
```

Might be fixed by https://github.com/kevinmehall/nusb/pull/39 in case it has `DeviceInterfaceGUID` instead of `DeviceInterfaceGUIDs`.

Reinstalling the WinUSB driver for interface 0 of the device with [Zadig](https://zadig.akeo.ie/) would probably work around that if it is the issue, or would fix it if the problem is that neither of those registry keys are set.

The incorrect "Probe was not found" error message is https://github.com/probe-rs/probe-rs/issues/2212.

Logiase, 2024-02-28:
Thanks to @kevinmehall 's advice, replace the original driver with Zadig's WinUSB *non-WCID* driver fix this issue.

> But I encountered other problems. XD