probe-rs/probe-rs #2250: STLink V2 do not work with Raspberry Pi Pico (PR2040)

sartor, 2024-03-02:
When I tried to flash simple blinky project (https://github.com/aptlyundecided/super-blank-rust-pico-project/tree/master)
It works via USB connection and elf file upload, but didn't work via probe-rs with STLInk connected

There is erro whena I am running `cargo run --bin blinky`:
```
warning: unused manifest key: target.thumbv6m-none-eabi.runner
warning: unused import: `panic_halt as _`
  --> src/main.rs:17:5
   |
17 | use panic_halt as _;
   |     ^^^^^^^^^^^^^^^
   |
   = note: `#[warn(unused_imports)]` on by default

warning: `super-blank-project` (bin "blinky") generated 1 warning (run `cargo fix --bin "blinky"` to apply 1 suggestion)
    Finished dev [unoptimized + debuginfo] target(s) in 0.05s
     Running `probe-rs run --chip RP2040 target\thumbv6m-none-eabi\debug\blinky`
 WARN probe_rs::probe::stlink: send_jtag_command 242 failed: JtagGetIdcodeError
Error: Connecting to the chip was unsuccessful.

Caused by:
    0: An error with the usage of the probe occurred
    1: An error specific to a probe type occurred
    2: Command failed with status JtagGetIdcodeError
error: process didn't exit successfully: `probe-rs run --chip RP2040 target\thumbv6m-none-eabi\debug\blinky` (exit code: 1)
```

`probe-rs list` returns:
```
The following debug probes were found:
[0]: STLink V2 (VID: 0483, PID: 3748, Serial: 37263236464643303126302634, StLink)
```

but `probe-rs info`:
```
Probing target via JTAG

 WARN probe_rs::probe::stlink: send_jtag_command 242 failed: JtagGetIdcodeError
Error identifying target using protocol JTAG: An error with the usage of the probe occurred

Probing target via SWD

 WARN probe_rs::probe::stlink: send_jtag_command 242 failed: JtagGetIdcodeError
Error identifying target using protocol SWD: An error with the usage of the probe occurred
```

I've checked voltage, update STLink drivers and firmware - no changes

vDorst, 2024-03-06:
I reed that the st-firmware is missing some features that are needed to connect to a dual-core cpu.
See https://electronics.stackexchange.com/questions/592979/programming-rp2040-with-st-link

Yatekii, 2024-03-10:
ST-Links cannot debug RP2040 unfortunately :(

closed 2024-03-10 by Yatekii

sartor, 2024-03-11:
Thank you. May be this info will help other people. Can you add it to manual?

Yatekii, 2024-03-11:
We will happily accept PRs :)

sartor, 2024-03-11:
I can't find repository with documentations on site:
https://probe.rs/docs/getting-started/probe-setup/